v4l2loopback/v4l2loopback #185: Cannot build on kernel 4.15.0-24

chuthagoras, 2018-07-13:
When doing a sudo apt install v4l2loopback-dkms, the install fails.

`$ cat /var/crash/v4l2loopback-dkms.0.crash
ProblemType: Package
DKMSBuildLog:
 DKMS make.log for v4l2loopback-0.9.1 for kernel 4.15.0-24-generic (x86_64)
 Fri Jul 13 08:10:26 EDT 2018
 Building v4l2-loopback driver...
 make -C /lib/modules/4.15.0-24-generic/build M=/var/lib/dkms/v4l2loopback/0.9.1/build modules
 make[1]: Entering directory '/usr/src/linux-headers-4.15.0-24-generic'
 Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"                                                                                    
   CC [M]  /var/lib/dkms/v4l2loopback/0.9.1/build/v4l2loopback.o
 /var/lib/dkms/v4l2loopback/0.9.1/build/v4l2loopback.c: In function ‘v4l2_loopback_init’:
 /var/lib/dkms/v4l2loopback/0.9.1/build/v4l2loopback.c:2181:2: error: implicit declaration of function ‘setup_timer’ [-Werror=implicit-function-declaration]                                               
   setup_timer(&dev->sustain_timer, sustain_timer_clb, nr);
   ^
 cc1: some warnings being treated as errors
 scripts/Makefile.build:339: recipe for target '/var/lib/dkms/v4l2loopback/0.9.1/build/v4l2loopback.o' failed                                                                                              
 make[2]: *** [/var/lib/dkms/v4l2loopback/0.9.1/build/v4l2loopback.o] Error 1
 Makefile:1552: recipe for target '_module_/var/lib/dkms/v4l2loopback/0.9.1/build' failed
 make[1]: *** [_module_/var/lib/dkms/v4l2loopback/0.9.1/build] Error 2
 make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-24-generic'
 Makefile:42: recipe for target 'v4l2loopback.ko' failed
 make: *** [v4l2loopback.ko] Error 2
DKMSKernelVersion: 4.15.0-24-generic
Date: Fri Jul 13 08:10:30 2018
DuplicateSignature: dkms:v4l2loopback-dkms:0.9.1-4:/var/lib/dkms/v4l2loopback/0.9.1/build/v4l2loopback.c:2181:2: error: implicit declaration of function ‘setup_timer’ [-Werror=implicit-function-declaration]
Package: v4l2loopback-dkms 0.9.1-4
PackageVersion: 0.9.1-4
SourcePackage: v4l2loopback
Title: v4l2loopback-dkms 0.9.1-4: v4l2loopback kernel module failed to build
`

I found another discussion on this from a different kernel module: http://patchwork.dpdk.org/patch/31739/

As the error mentions, the timer APIs changed from 4.15.

chuthagoras, 2018-07-13:
Sorry, I just noticed master already has the fix on commit 00eb0df20248d7a37594905d64030d0bd0b72eb8

It'll be nice to update the version in the apt-repo though.

closed 2018-07-13 by chuthagoras

umlaeute, 2018-07-16:
which apt-repo are you talking about?
- [Debian](https://packages.debian.org/search?keywords=v4l2loopback) has `0.12.0`d
- [Ubuntu](https://packages.ubuntu.com/search?keywords=v4l2loopback) has `0.12.0`

chuthagoras, 2018-07-18:
$ apt-cache policy v4l2loopback-dkms
v4l2loopback-dkms:
  Installed: 0.9.1-4
  Candidate: 0.9.1-4
  Version table:
 *** 0.9.1-4 500
        500 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 Packages
        500 http://us.archive.ubuntu.com/ubuntu xenial/universe i386 Packages
        100 /var/lib/dpkg/status

umlaeute, 2018-07-18:
well, if you want updated packages, you have to upgrade your system.  The Ubuntu-link I gave aboce, should show you which Ubuntu versions include which versions of v4l2loopback.

high400, 2019-11-20:
"Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"

I got this error when i tried to install my wifi driver on Linux. 
For more details, my Linux is Cinnamon Mint 19.2, 64 bit
I'm trying to fix this, if i am able to fix this, then i will be able to install wifi driver

umlaeute, 2019-11-20:
@high400: how about doing what the error says?

high400, 2019-12-18:
> @high400: how about doing what the error says?

I did. It was so "persistent" that no matter what i did, it DID NOT receive the wifi.
I changed to install Ubuntu, the same thing repeated itself

high400, 2019-12-24:
> I did. It was so "persistent" that
>
> so you installed libelf-dev and related packages?
>
> no matter what i did, it DID NOT receive the wifi.
>
> sorry, I don't understand.
> what "WiFi" did you "not receive"?

The Wifi Driver. And yes, i did install libelf-dev and related packages, if
that is what you mean

umlaeute, 2019-12-24:
then you are barking up the wrong tree.
WIFI is totally unrelated to `v4l2loopback`.