v4l2loopback/v4l2loopback #186: Missing formats when cloning v4l2 video device

Supereg, 2018-08-12:
### Environment

  * `v4l2loopback` version: 0.12.0

  * kernel version: Linux 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux

  * Distribution (+version): Raspbian GNU/Linux 9.4 (stretch)
  
### Problem:

Hello, my goal is to produce multiple virtual copies of the raspberry pi camera (uses v4l2 with interface `/dev/video0`) with v4l2loopback in order to stream the copy via rtp with ffmpeg to multiple receivers simultaneously. When I run `v4l2-ctl --list-formats -d /dev/video0` it lists all available formats for the rpi cam including h.264 and mjpg. For `v4l2-ctl --list-formats -d /dev/video1` it only lists 'Planar YUV 4:2:0' as the only available format. As a result when I try to stream the content of video1 with ffmpeg via rtp it tells me that no proper format could be found for respective codec (h264/mjpg).

I already looked at the `v4l2loopback_formats.h` file and of course both codecs are supported. Is it maybe my ffmpeg command which copies the data from video0 to video1? 
`ffmpeg -f video4linux2 -s 1280x720 -i /dev/video0 -codec copy -f v4l2 /dev/video1`

#### Steps to reproduce:

  1. Have an video device like the pi camera on `/dev/video0`
  2. Make and install v4l2loopback and create video device at `/dev/video1`
  3. Run ffmpeg command to copy content from video0 to video1: `ffmpeg -f video4linux2 -s 1280x720 -i /dev/video0 -codec copy -f v4l2 /dev/video1`
  
#### Observed Results:

Result of `v4l2-ctl --list-formats -d /dev/video1` (not containing h264 nor mjpg):
```
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YU12'
        Name        : Planar YUV 4:2:0
```

Result of `v4l2-ctl --list-formats -d /dev/video0`
```
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YU12'
        Name        : Planar YUV 4:2:0

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUYV 4:2:2

        Index       : 2
        Type        : Video Capture
        Pixel Format: 'RGB3'
        Name        : 24-bit RGB 8-8-8

        Index       : 3
        Type        : Video Capture
        Pixel Format: 'JPEG' (compressed)
        Name        : JFIF JPEG

        Index       : 4
        Type        : Video Capture
        Pixel Format: 'H264' (compressed)
        Name        : H.264

        Index       : 5
        Type        : Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name        : Motion-JPEG

        Index       : 6
        Type        : Video Capture
        Pixel Format: 'YVYU'
        Name        : YVYU 4:2:2

        Index       : 7
        Type        : Video Capture
        Pixel Format: 'VYUY'
        Name        : VYUY 4:2:2

        Index       : 8
        Type        : Video Capture
        Pixel Format: 'UYVY'
        Name        : UYVY 4:2:2

        Index       : 9
        Type        : Video Capture
        Pixel Format: 'NV12'
        Name        : Y/CbCr 4:2:0

        Index       : 10
        Type        : Video Capture
        Pixel Format: 'BGR3'
        Name        : 24-bit BGR 8-8-8

        Index       : 11
        Type        : Video Capture
        Pixel Format: 'YV12'
        Name        : Planar YVU 4:2:0

        Index       : 12
        Type        : Video Capture
        Pixel Format: 'NV21'
        Name        : Y/CrCb 4:2:0

        Index       : 13
        Type        : Video Capture
        Pixel Format: 'BGR4'
        Name        : 32-bit BGRA/X 8-8-8-8
```
 
I'm pretty new to the whole v4l2 thing so it is possible that I had overlooked something.

**Some notice:** I discovered that the list-formats command is empty for video1 when the ffmpg copy command isn't running. What would be the right command to copy every format type to the loopback device? I did indeed fully read the wiki articles however this scenario isn't really documented.

Big thanks in advance for any help 
Greets Andi

Supereg, 2018-08-12:
Oh and `ffmpeg -version`:
```
ffmpeg version N-91579-g243ecadad5 Copyright (c) 2000-2018 the FFmpeg developers
built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
configuration: --enable-libfreetype --enable-gpl --enable-nonfree --enable-libx264 --enable-libass --enable-libmp3lame --bindir=/usr/local/bin --enable-mmal --enable-omx --enable-omx-rpi
libavutil      56. 18.102 / 56. 18.102
libavcodec     58. 22.100 / 58. 22.100
libavformat    58. 17.101 / 58. 17.101
libavdevice    58.  4.101 / 58.  4.101
libavfilter     7. 26.100 /  7. 26.100
libswscale      5.  2.100 /  5.  2.100
libswresample   3.  2.100 /  3.  2.100
libpostproc    55.  2.100 / 55.  2.100
```

umlaeute, 2018-08-12:
i'm not sure what you are expecting (and what you want to achieve)

the loopback device doesn't know anything about the source of the pixel data (e.g. whether it was copied from a real video device or just generated on the fly).
it only knows what the producer is sending it (in your case `ffmpeg` sets up the loopback device to use `YU12` as the format; it doesn't pass any information about other possible formats to the loopback).

since the format of the loopback device is entirely determined by the producer, the list of formats is empty if you haven't connected a producer to the loopback device.

also, there's no way for a v4l2 module (in kernel-space) to signal back to the producer application (in user-space), that a consumer (in user-space) wants a different format.

probably the easiest solution to your problem is to tell `ffmpeg` to output the data in a format that your RTP streamer can handle (contact the manuals of `ffmpeg` and your RTP streamer on how to do that)

closed 2018-08-12 by umlaeute

Supereg, 2018-08-13:
My current setup is as follows. My program uses ffmpeg to take the h264 stream and copying to the rtp stream. Now I would like to utilize v4l2loopback to be able to read the camera stream with more than one stream.
The thing is that according to ffmpeg the v4l2loopback device expects raw video input. Thus I would need to encode the raw video stream from /dev/video1 to an h264 stream and than stream it over rtp. This creates cpu usage the raspberry pi cannot really handle and destroys the purpose of an live rtp stream.
I did understand, that the loopback device doesn't know about anything and really cannot doe anything in kernel-space. However I assumed that I would be able to send h264 to the loopback devices.
Is there anything I'm not aware of?

I appreciate the time you are spending trying to help me, I hope I'm not that frustrating.
Many thanks!

mark-hahn, 2019-01-04:
I am in a similar situation with the same OS, v4l2loopback, raspi, etc.   I want to stream one /dev/video* with mjpg-streamer and occasionally use opencv on another /dev/video*.

I have a working raspi camera feed on video0.  I ran `modprobe v4l2loopback` to get video1.  `--list-formats` on video0 shows the normal long list of formats.  `--list-formats` on video1 just shows `ioctl: VIDIOC_ENUM_FMT`.

I am a v4l2loopback newbie so I don't completely understand how to use it.  The OP above said *my ffmpeg command which copies the data from video0 to video1*, but I don't understand why you would send a stream to video1.  I thought it was supposed to be "tee" of video0.

If I do have to run something to copy video0 to video 1 what should I use?  Is it gstream?  I don't know what gstream is but everyone seems to be using it on this forum.  I looked up gstream and it seems a lot heavier and more complex that a "tee".

Any help would be appreciated.