v4l2loopback/v4l2loopback #187: kernel warning on Raspberry Pi about HEVC not found

marktheunissen, 2018-08-13:
Hi there, the following appears in the kernel logs on v4l2loopback version 0.12 on Raspberry Pi:

```
Aug 13 12:01:28 pi200 kernel: WARNING: CPU: 0 PID: 8297 at drivers/media/v4l2-core/v4l2-ioctl.c:1288 v4l_enum_fmt+0x1434/0x1798 [videodev]
Aug 13 12:01:28 pi200 kernel: Unknown pixelformat 0x43564548
Aug 13 12:01:28 pi200 kernel: Modules linked in: v4l2loopback(O) videodev media cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq uio fixed ip_tables x_tables ipv6 [last unloaded: media]
Aug 13 12:01:28 pi200 kernel: CPU: 0 PID: 8297 Comm: gst-launch-1.0 Tainted: G        WC O    4.14.52-v7+ #1123
Aug 13 12:01:28 pi200 kernel: Hardware name: BCM2835
Aug 13 12:01:28 pi200 kernel: [<8010ffd8>] (unwind_backtrace) from [<8010c240>] (show_stack+0x20/0x24)
Aug 13 12:01:28 pi200 kernel: [<8010c240>] (show_stack) from [<80785e84>] (dump_stack+0xd4/0x118)
Aug 13 12:01:28 pi200 kernel: [<80785e84>] (dump_stack) from [<8011da4c>] (__warn+0xf8/0x110)
Aug 13 12:01:28 pi200 kernel: [<8011da4c>] (__warn) from [<8011daac>] (warn_slowpath_fmt+0x48/0x50)
Aug 13 12:01:28 pi200 kernel: [<8011daac>] (warn_slowpath_fmt) from [<7f21b9b4>] (v4l_enum_fmt+0x1434/0x1798 [videodev])
Aug 13 12:01:28 pi200 kernel: [<7f21b9b4>] (v4l_enum_fmt [videodev]) from [<7f218974>] (__video_do_ioctl+0x2a0/0x314 [videodev])
Aug 13 12:01:28 pi200 kernel: [<7f218974>] (__video_do_ioctl [videodev]) from [<7f218318>] (video_usercopy+0x224/0x5bc [videodev])
Aug 13 12:01:28 pi200 kernel: [<7f218318>] (video_usercopy [videodev]) from [<7f2186d0>] (video_ioctl2+0x20/0x24 [videodev])
Aug 13 12:01:28 pi200 kernel: [<7f2186d0>] (video_ioctl2 [videodev]) from [<7f214704>] (v4l2_ioctl+0xb8/0xe8 [videodev])
Aug 13 12:01:28 pi200 kernel: [<7f214704>] (v4l2_ioctl [videodev]) from [<8029e02c>] (do_vfs_ioctl+0xac/0x7c4)
Aug 13 12:01:28 pi200 kernel: [<8029e02c>] (do_vfs_ioctl) from [<8029e788>] (SyS_ioctl+0x44/0x6c)
Aug 13 12:01:28 pi200 kernel: [<8029e788>] (SyS_ioctl) from [<80108060>] (ret_fast_syscall+0x0/0x28)
Aug 13 12:01:28 pi200 kernel: ---[ end trace be16edb2d0e14173 ]---
```

Looks to be `Unknown pixelformat 0x43564548`, which by Googling I discovered is HEVC. I see this was added to v4l2loopback here: https://github.com/umlaeute/v4l2loopback/commit/79a71bf7308be5faf8d97c72107688f8c6da1733

I reverted to 0.10 and now the warning does not appear in the kernel logs. I don't think this is related to the  particular crash I'm trying to debug, but wanted to point it out.

```
uname -a
Linux pi200 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux

Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch
```

Thanks!

umlaeute, 2018-10-01:
how are you feeding the loopback device?

umlaeute, 2018-11-15:
your running kernel doesn't support the `HEVC` pixel format, although it's headers properly define it.

try removing the the `HEVC` section near the end of [`v4l2loopback_formats.h`](https://github.com/umlaeute/v4l2loopback/blob/b2b33ee/v4l2loopback_formats.h#LL413-L418), recompile and report back whether this fixes the problem.