v4l2loopback/v4l2loopback #189: Warning: Error Pipe: Unable to link videotestsrc0 to v4l2sink0

linuxkun, 2018-09-21:
hi there,a warning appeared when I typed v4l2loopback-ctl set-caps "video/x-raw-yuv, width=640, height=480" /dev/video0.

(gst-launch-1.0:4440): GStreamer-WARNING **: 0.10-style raw video caps are being created. Should be video/x-raw,format=(string).. now.
Warning: Error Pipe: Unable to link videotestsrc0 to v4l2sink0
output to /dev/video0 failed

the kernel version :
uname -a
Linux kun 4.4.0-133-generic #159-Ubuntu SMP Fri Aug 10 07:31:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

Can you tell me why this happened? 
Thanks

umlaeute, 2018-10-01:
as the warning says, you should use GStreamer-1.0 caps, rather than GStreamer-0.10.

try:
~~~sh
$ v4l2loopback-ctl set-caps  "video/x-raw,format=UYVY,width=320,height=240" /dev/video0
~~~

(the docs are out-of-date)

closed 2018-10-01 by umlaeute

Niek-Okido, 2020-06-26:
still out-dated documents , really irritating me (this gives me hard time the last few days :| )

umlaeute, 2020-06-30:
@Niek-Okido PRs welcome.