v4l2loopback/v4l2loopback #191: Half-written frames using mmaped output at high frame rates

ochilan, 2018-10-15:
Hello,

I'm trying to write a relatively simple program to forward one V4L2 device to one or more other (virtual) V4L2 devices. The main reason is that ffmpeg only has simple write() V4L2 output and thus does not support forwarding timestamps at all, and gstreamer has other problems with certain buffer sizes as described in issue #190. Anyway, I want to have full control over what happens, and maybe others would also like to have a program which simply forwards an input to multiple outputs efficiently without much processing going on whatsoever.

However, I'm having a problem and I don't know if it's because I have never used the V4L2 API before, or because there is some kind of race condition going on with the buffers outside of my code.

Please consider the code attached at the end of this issue. What it is supposed to do (see main.cpp) is to initialize a capturing VideoDevice and one or more output VideoDevices. Then it should go into a "forward_loop" and read from the first device and write to the output devices. The setup of the devices is done in the VideoDevice constructor and is inpired by the "capture.c" example of V4L2, adjusted to take into account the type of the device (capture or output).

However, for this example, I have simplified the setting to only an output device. It initializes the VideoDevice (should be a v4l2loopback device) and then goes into "test_loop" which simply writes some colored frames out to mmaped buffers. The colors are set rudimentally using memset. While it does that, it outputs the index of the dequeued buffer, and the number of the current frame.

Now the results are weird in two ways. First, if I use a sufficiently high frame rate (60 FPS) and capture from the same v4l2 device using e.g. ffmpeg while my tool is running, I sometimes get frames that are only half-written. For example, such a frame looks like this:

[image]

This also happens when I'm using input from another, real V4L2 device instead of using the test_loop. It looks to me like frames are already output to ffmpeg while they are still being written in my program.

Another weirdness is that I always get buffer index 3 dequeued in my application. I'm not sure how it is supposed to work, but shouldn't the buffers (in this case 4 buffers) be iterated in some way?

After all, like I said, this is the first time I'm really digging into V4L2 and maybe I'm just mixing up something with the buffers or their flags. Any assistance would be really appreciated. The program is not very complex – it's mainly the device setup in the VideoDevice constructor and then the test_loop. Maybe I'm just overlooking something.

Best regards and thank you in advance
Ochi

v4l2loopback version: 0.12.0
Kernel: 4.18.12
Distribution: Arch Linux

[v4l2forward.zip](https://github.com/umlaeute/v4l2loopback/files/2479862/v4l2forward.zip)

ochilan, 2018-10-17:
I have to admit that I still don't understand how buffer queueing/dequeueing is _supposed_ to work. :(

To reduce my current problem to a simple question: If I'm using mmap to write data out to the v4l2loopback device, is the *application* or is the *driver* responsible for rotating through the allocated buffers?

From what I'm seeing in examples, my understanding is that to write out a new image, I would dequeue a buffer from the driver, write my image data to it (or rather to the respective mmapped memory), and enqueue it again. Then the driver would wait for some other application to fetch this buffer for reading and in the meantime, the writer would get the _next_ buffer when dequeuing, for writing the next image to it.

Is this correct so far? If so, I still don't understand why I'm always getting the same buffer (i.e. the buffer with the same index) over and over when dequeuing.

umlaeute, 2018-10-18:
i guess this is a bug in v4l2loopback.

J0hannB, 2020-06-19:
@ochilan, did you ever solve your half-written frames problem? 

I'm using v4l2 to stream frames from a camera and I think I may be experiencing the same issue. 

Streaming works flawlessly until I start requesting frames at >60 fps (the camera is capturing at 230 fps). At this point I periodically get an image that is one part the current frame and one part the previous frame (divided much like the image you included above). 

I am also using mmaped buffers to grab frames from the driver, but I have verified that the problem also occurs when I get frames via the read()/write() api as well. 

I would love to know if anyone has any insight into why this might be!

tachang, 2020-11-29:
I am running into an issue like this as well. Not sure if it is the same though. Here are some images:

[image]

tachang, 2020-11-29:
However the issue I am seeing is a relatively low framerate (about 30fps).

tachang, 2020-11-29:
Also I am not using mmaping.

stephematician, 2023-06-23:
> Is this correct so far? If so, I still don't understand why I'm always getting the same buffer (i.e. the buffer with the same index) over and over when dequeuing.

I seem to get the same kind of behaviour when attempting to use mmap. Always get the same buffer returned. Also select() and poll() do not function as expected.

stephematician, 2025-02-06:
@ochilan - if you're still around, this is likely closed as of 24cc1e8.