v4l2loopback/v4l2loopback #193: Chrome Not Seeing Video Device -- Exclusive Caps Wrong

benlemond, 2018-11-05:
All,

I have just pulled down the latest and am trying to create virtual cameras for consumption by Chrome on a linux-ubuntu arm machine.  I believe it worked at first, but I tried to remove it and instantiate more and it seems I am in a pickle.  Chrome no longer sees the cameras.
When I run "cat /sys/module/v4l2loopback/parameters/exclusive_caps" I get "Y,Y,Y,Y,N,N,N,N".  I believe the issue has something to do with the "N" there, but I am not sure.

Current Procedure:
sudo su
cd /usr/src/linux-headers-4.4.38-tegra
make modules_prepare
mkdir v4l2loopback && cd v4l2loopback
git clone https://github.com/umlaeute/v4l2loopback.git
cd v4l2loopback
make && make install
depmod -a 

sudo modeprobe v4l2loopback devices=4 video_nr=2,3,4,5 card_label="camera 0","camera 1","camera 2","camera 3" exclusive_caps=1,1,1,1
 
sudo echo options v4l2loopback devices=4 video_nr=2,3,4,5 card_label="camera 0","camera 1","camera 2","camera 3" exclusive_caps=1,1,1,1 > /etc/modprobe.d/v4l2loopback.conf
 
#add the kernel module to the list of modules to be loaded at boot
echo v4l2loopback > /etc/modules
 
update-initramfs -u

Shouldn't I expect to see 4 (or 8) "Y" in the exclusive_caps field?  Is there any reason I wouldn't be seeing the device in a browser?

Is there a way to completely wipe a module and start over from scratch?

umlaeute, 2018-11-06:
what do you mean with "completely wipe module and start from scratch"?

~~~sh
rm /lib/modules/$(uname -r)/extra/v4l2loopback.ko
~~~

umlaeute, 2018-11-06:
to answer the actual question: it seems like you *do* see 4 "Y", so the 1st 4 devices are set to announce exclusive caps. You will see 8 "Y" if you pass `exclusive_caps=1,1,1,1,1,1,1,1` as module parameters.

i agree that you shouldn't get 8 values if you only have 4 devices, but i think that this is merely a minor glitch (chrome doesn't (shouldn't) use `/sys/module/v4l2loopback/` at all, so the values there don't matter).

also note, that [some of the tests i did (see comments in #183)](https://github.com/umlaeute/v4l2loopback/issues/183#issuecomment-425920125), required a proper audio setup to make chrome see the loopback devices. this probably was only a glitch in the webpage used for testing the video-input in the browser (but i don't know how you do your tests)

umlaeute, 2018-11-06:
yes, that's most likely an issue.

ROBERT-MCDOWELL, 2018-11-15:
I can reproduce the same issue on Chromium, even when I send something to /dev/videoX, with a sound card this time ;)
others browsers like Firefox, Seamonkey etc. works like a charm.

closed 2018-11-15 by benlemond

hutber, 2020-07-25:
I can confirm I am still having this issue too.