v4l2loopback/v4l2loopback #195: Kernel Errors with gstreamer on Tegra with v4l2loopback versions higher than 0.10

HoneyPatouceul, 2018-11-14:
I'm using a NVIDIA Jetson TX2 with L4T R28.2.1 having kernel 4.4.38-tegra.

I am seeing kernel faults when I use v4l2loopback with gstreamer-1.0 (1.8.3) with v4l2loopback versions 0.11.0 and later. With previous versions 0.9.1 or 0.10.0 it doesn't make any fault.

When I launch a gstreamer pipeline with v4l2sink:
```
gst-launch-1.0 -e videotestsrc is-live=true ! 'video/x-raw, format=BGR, width=640, height=480, framerate=30/1' ! tee ! v4l2sink device=/dev/video2
Setting pipeline to PAUSED ...
**libv4l2: error getting pixformat: Invalid argument**
Pipeline is live and does not need PREROLL ...
```

I see these errors with dmesg (The symptoms are very similar to issue #187) :
```
[ 6762.706707] **Unknown pixelformat 0x30395056**
[ 6762.711073] ------------[ cut here ]------------
[ 6762.715714] WARNING: at ffffffc0007a7240 [verbose debug info unavailable]
[ 6762.722511] Modules linked in: v4l2loopback(O) fuse ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 xt_addrtype iptable_filter ip_tables xt_conntrack nf_nat br_netfilter overlay bcmdhd bluedroid_pm pci_tegra [last unloaded: v4l2loopback]

[ 6762.746586] CPU: 3 PID: 16042 Comm: videotestsrc0:s Tainted: G        W  O    4.4.38-tegra #1
[ 6762.755109] Hardware name: quill (DT)
[ 6762.758776] task: ffffffc1ecf08000 ti: ffffffc192294000 task.ti: ffffffc192294000
[ 6762.766271] PC is at v4l_enum_fmt+0x1270/0x1608
[ 6762.770803] LR is at v4l_enum_fmt+0x1270/0x1608
[ 6762.775332] pc : [<ffffffc0007a7240>] lr : [<ffffffc0007a7240>] pstate: 80000045
[ 6762.782717] sp : ffffffc192297bf0
[ 6762.786029] x29: ffffffc192297bf0 x28: ffffffc001348730 
[ 6762.791353] x27: 0000000000000000 x26: ffffffc192297d20 
[ 6762.796676] x25: 0000007f8c005260 x24: ffffffbffc748aa8 
[ 6762.801994] x23: ffffffc0412e9680 x22: ffffffc157b53e00 
[ 6762.807314] x21: ffffffbffc748aa8 x20: 0000000000000000 
[ 6762.812632] x19: ffffffc192297d20 x18: 0000000000000000 
[ 6762.817952] x17: 0000007f924f3b20 x16: ffffffc000b88a60 
[ 6762.823272] x15: ffffffc000b88a60 x14: 0000000000000001 
[ 6762.828591] x13: 0000012ffcbfc857 x12: 00000000000025ae 
[ 6762.833910] x11: 0000000000007531 x10: 00000000000008b0 
[ 6762.839229] x9 : ffffffc1922979b0 x8 : ffffffc1ecf08910 
[ 6762.844547] x7 : 00000000000000ad x6 : 00000000000a20c0 
[ 6762.849866] x5 : 0000000000000000 x4 : ffffffc1ecf08918 
[ 6762.855184] x3 : ffffffc07b198a0c x2 : ffffffc07021cb28 
[ 6762.860502] x1 : 0000000000000000 x0 : 000000000000001e 

[ 6762.867627] ---[ end trace ef74d47dcb7fabe8 ]---
[ 6762.872233] Call trace:
[ 6762.874676] [<ffffffc0007a7240>] v4l_enum_fmt+0x1270/0x1608
[ 6762.880237] [<ffffffc0007a7dcc>] __video_do_ioctl+0x22c/0x2a0
[ 6762.885971] [<ffffffc0007a782c>] video_usercopy+0x254/0x5ac
[ 6762.891532] [<ffffffc0007a7b98>] video_ioctl2+0x14/0x1c
[ 6762.896748] [<ffffffc0007a2b8c>] v4l2_ioctl+0xbc/0xcc
[ 6762.901792] [<ffffffc0001e6350>] do_vfs_ioctl+0x324/0x5e4
[ 6762.907179] [<ffffffc0001e6694>] SyS_ioctl+0x84/0x98
[ 6762.912135] [<ffffffc000084ff0>] el0_svc_naked+0x24/0x28
[ 6762.917644] **Unknown pixelformat 0x43564548**
[ 6762.921765] ------------[ cut here ]------------
[ 6762.926373] WARNING: at ffffffc0007a7240 [verbose debug info unavailable]
[ 6762.933146] Modules linked in: v4l2loopback(O) fuse ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 xt_addrtype iptable_filter ip_tables xt_conntrack nf_nat br_netfilter overlay bcmdhd bluedroid_pm pci_tegra [last unloaded: v4l2loopback]

[ 6762.957022] CPU: 3 PID: 16042 Comm: videotestsrc0:s Tainted: G        W  O    4.4.38-tegra #1
[ 6762.965528] Hardware name: quill (DT)
[ 6762.969182] task: ffffffc1ecf08000 ti: ffffffc192294000 task.ti: ffffffc192294000
[ 6762.976653] PC is at v4l_enum_fmt+0x1270/0x1608
[ 6762.981174] LR is at v4l_enum_fmt+0x1270/0x1608
[ 6762.985694] pc : [<ffffffc0007a7240>] lr : [<ffffffc0007a7240>] pstate: 60000145
[ 6762.993072] sp : ffffffc192297bf0
[ 6762.996379] x29: ffffffc192297bf0 x28: ffffffc001348730 
[ 6763.001700] x27: 0000000000000000 x26: ffffffc192297d20 
[ 6763.007018] x25: 0000007f8c0052b0 x24: ffffffbffc748aa8 
[ 6763.012340] x23: ffffffc0412e9680 x22: ffffffc157b53e00 
[ 6763.017658] x21: ffffffbffc748aa8 x20: 0000000000000000 
[ 6763.022978] x19: ffffffc192297d20 x18: 0000000000000000 
[ 6763.028297] x17: 0000007f924f3b20 x16: ffffffc0001e6610 
[ 6763.033618] x15: 0000007f9286e000 x14: 0ffffffffffffffe 
[ 6763.038937] x13: 0000000000000020 x12: ffffffc001299000 
[ 6763.044256] x11: 0000000000000002 x10: 0000000000000000 
[ 6763.049574] x9 : 0000000000000b7c x8 : 0000000000000001 
[ 6763.054891] x7 : ffffffc192294000 x6 : 0000000000000000 
[ 6763.060209] x5 : 0000000000000000 x4 : 0000000000000000 
[ 6763.065528] x3 : 0000000000000000 x2 : 0000000000000000 
[ 6763.070848] x1 : 0000000000000000 x0 : 000000000000001e 

[ 6763.077962] ---[ end trace ef74d47dcb7fabe9 ]---
[ 6763.082570] Call trace:
[ 6763.085013] [<ffffffc0007a7240>] v4l_enum_fmt+0x1270/0x1608
[ 6763.090574] [<ffffffc0007a7dcc>] __video_do_ioctl+0x22c/0x2a0
[ 6763.096309] [<ffffffc0007a782c>] video_usercopy+0x254/0x5ac
[ 6763.101870] [<ffffffc0007a7b98>] video_ioctl2+0x14/0x1c
[ 6763.107086] [<ffffffc0007a2b8c>] v4l2_ioctl+0xbc/0xcc
[ 6763.112129] [<ffffffc0001e6350>] do_vfs_ioctl+0x324/0x5e4
[ 6763.117517] [<ffffffc0001e6694>] SyS_ioctl+0x84/0x98
[ 6763.122474] [<ffffffc000084ff0>] el0_svc_naked+0x24/0x28
```
The bad pixel format codes are always the same, whatever format I use.
It is the same with other sources than videotestsrc. The problem only happens when I add v4lsink in my pipeline (others sinks are ok). 

Not sure this is related, but reading the virtual device with v4l2src only works with RGB or BGR, it fails with I420, YV12, YUY2. With v0.10.0 all of these work without error.

umlaeute, 2018-11-15:
your running kernel doesn't support the `VP9` pixel format, although it's headers properly define it.

try removing the the `VP9` section near the end of [`v4l2loopback_formats.h`](https://github.com/umlaeute/v4l2loopback/blob/b2b33ee/v4l2loopback_formats.h#L407-L412), recompile and report back whether this fixes the problem.

HoneyPatouceul, 2018-11-15:
Indeed, commenting both VP9 and HEVC sections fixed the problem.
libv4l2 error message is still there when launching gstreamer pipeline, but this doesn't look harmful.
Thanks for this great module.

closed 2018-11-15 by HoneyPatouceul