v4l2loopback/v4l2loopback #196: Chromium 70.x does not show v4l2loopback

ROBERT-MCDOWELL, 2018-11-16:
### Describe your environment
version 0.12.0
Linux 4.19.1-rt3 #1 SMP PREEMPT RT Thu Nov 15 14:31:47 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
LSB Version:    :core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID: Fedora
Description:    Fedora release 27 (Twenty Seven)
Release:        27
Codename:       TwentySeven

### Step 3: Describe the problem:
v4l2loopback does not show in flash/html5 list of video devices, despite of sending data to the video device

#### Steps to reproduce:
start a stream to /dev/videoX and launch chromium
Note: I have a sound card with or without audio data

#### Observed Results:
Video device is not listed

Tested with other browsers mozillas like working nicely

ROBERT-MCDOWELL, 2018-11-16:
### Update: tested also on Ubuntu 14.05 and 18.04 32 and 64bit

umlaeute, 2018-11-16:
unable to reproduce with `v4l2loopback-v0.12.0` and `exclusive_caps` enabled.

closed 2018-11-16 by umlaeute

ROBERT-MCDOWELL, 2018-11-16:
ok so the issue is elsewhere...