v4l2loopback/v4l2loopback #197: Change device settings on the fly

ROBERT-MCDOWELL, 2018-11-24:
* `v4l2loopback` version:  0.12.0

  * kernel version: 4.19.1-rt3

  * Distribution (+version):  Fedora 27 64bits

### Step 3: Describe the problem:
When I select V4l2loopback from Flash/html5 video input devices list
and set it let's say to 15fps, 720x480 everything is ok. but if I change it on the fly (even if I detach, change settings and reattach the video object in actionscript/javascript so the image is distorted and skwed. I have to quit the browser and launch it again to make the new settings work

#### Steps to reproduce:

  1. Select v4l2loopback as video input device from Flash/html5 and set fps,width,height then publish
  2. Stop publish, change fps,widh,height then publish again

#### Observed Results:
Image skewed and distorted as resolution from v4l2loopback and Flash/html5 doesn't match
  
#### Expected Results:
When video settings are changed on the fly from Flash/html5 so v4l2loopback instantly take the new
settings active

luke-jr, 2020-09-20:
But existing V4L2 consumers *do* change the resolution/size of webcam video... So surely they have a way to request it?

umlaeute, 2020-09-21:
@luke-jr that's not really a valid comparision.

if you have a typical hardware-based `/dev/video`, than - whenever the userspace requests a different resolution/... the driver can negotiate this change with the hardware via their internal protocol.

but in the case of `v4l2loopback`, we don't have an internal protocol, only the v4l2-API.

with the v4l2-API there simply is no way (that i am aware of) to signal the requested change to the "other" userspace application (the *producer* that generates the frame data). even less so in a synchronous way (with the application that requested the change  stalling until the other end agreed on the change).

just as a thought experiment: if a (traditional, hardware-backed) `/dev/video` device allows multiple applications to simultaneously stream frames from the device, and one application decides to change the resolution from 1920x1080 to 320x240, how would this work for the other applications? afaict: it won't and it doesn't.

luke-jr, 2020-09-21:
Yes, my point was simply that the consumer side *does* have a settings interface. The producer side is admittedly a problem still.

Do any V4L2 devices allow multiple applications?

(FWIW I found this bug because I simply want to enable "sniffing" webcam video while another application is using it, without disrupting that application's control over the hardware)

umlaeute, 2020-09-21:
> Yes, my point was simply that the consumer side does have a settings interface. 

well, it *has* a settings interface. you can change it all you like, as long as you are the only one accessing the device.

but while another application is using the device, the settings are locked.

> FWIW I found this bug

which bug?
you might be missing what you consider a feature, but this is not a bug.