v4l2loopback/v4l2loopback #198: set-caps question

ROBERT-MCDOWELL, 2018-11-25:
I just created a video device with
sudo modprobe v4l2loopback devices=1 exclusive_caps=1 video_nr=1card_label=mysink

tried to set it like this
sudo v4l2loopback-ctl set-caps "video/x-raw,format=yuv, width=640, height=480" /dev/video1

but I get
WARNING: erroneous pipeline: could not link videotestsrc0 to v4l2sink0, neither element can handle caps video/x-raw, format=(string)yuv, width=(int)640, height=(int)480
output to /dev/video1 failed

any idea? thanks

umlaeute, 2018-11-25:
that doesn't seem to be a valid gstreamer-1.0 caps-string, the format `yuv` does not exist.

try something like `UYVY` instead.

ROBERT-MCDOWELL, 2018-11-25:
ok thanks now I get
v4l2loopback-ctl set-caps "video/x-raw,format=UYVY, width=640, height=480" /dev/video1
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Device '/dev/video1' is not a output device.
Additional debug info:
v4l2_calls.c(642): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Capabilities: 0x85208000
Setting pipeline to NULL ...
Freeing pipeline ...
output to /dev/video1 failed

ROBERT-MCDOWELL, 2018-11-25:
FYI I use /dev/video1 with ffmpeg like this
ffmpeg -re -framerate 12 -loop 1 -i image.png -f v4l2 /dev/video1

ensisoft, 2019-04-17:
I have a similar issue. I've tried both the kernel module from source and the one from package (I'm on Ubuntu 18)
I have a device at /dev/device0.

$ v4l2-ctl --list-devices
Dummy video device (0x0000) (platform:v4l2loopback-000):
	/dev/video0

Trying to use this device is not working at all. For example

$ gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video0
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Device '/dev/video0' is not a output device.
Additional debug info:
v4l2_calls.c(646): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Capabilities: 0x85208000
Setting pipeline to NULL ...
Freeing pipeline ...

$ gst-launch-1.0 v4l2src device=/dev/video0 ! xvimagesink
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Device '/dev/video0' is not a capture device.
Additional debug info:
v4l2_calls.c(638): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Capabilities: 0x85208000
Setting pipeline to NULL ...
Freeing pipeline ...

Honestly I'm not sure if this is just broken or if there's some undocumented magic step that should be done. Thanks!

umlaeute, 2019-04-23:
@enisoft i fail to see how your problem is related to `set-caps` (which is what *this* ticket is about).
please remove your post from this thread (if you don't have permissions, i can do it for you), and add it to a proper issue (or a new one if there is none)

lynn901213, 2020-06-03:
Is it possible for v4l2loopback device to accept mjpeg as input? I tried to set caps like
`v4l2loopback-ctl set-caps "image/jpeg, width=1280, height=720, fps=30/1" /dev/video1`
where /dev/video1 is my v4l2loopback device. It seems like it is using gstreamer videotestsrc as the pipeline video src for setting the caps, and 
`gst-inspect-1.0 videotestsrc` only accepts paddings as video/x-raw.
My question here would be, can we set mjpeg as the format for v4l2loopback device?

umlaeute, 2020-06-03:
sure. you can. the current behaviour simply a limitation of the `v4l2loopback-ctl` script (or rather, as you noted, the underlying `videotestsrc` element).

you can manually change the output-format  by running a pipeline like:

~~~
gst-launch-1.0 videotestsrc num-buffers=1 ! "video/x-raw, width=1280, height=720, fps=30/1" ! avenc_mjpeg ! v4l2sink device=/dev/video1
~~~

nicola-sorace, 2020-11-06:
I'm getting the same error as @ROBERT-MCDOWELL above:
```
Setting pipeline to PAUSED ...
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Device '/dev/video0' is not a output device.
Additional debug info:
../gst-plugins-good/sys/v4l2/v4l2_calls.c(636): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Capabilities: 0x5208000
ERROR: pipeline doesn't want to preroll.
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
output to /dev/video0 failed
```

sbeve72, 2024-01-30:
> I'm getting the same error as @ROBERT-MCDOWELL above:
> 
> ```
> $ sudo v4l2loopback-ctl set-caps "video/x-raw,format=UYVY, width=1920, height=1200" /dev/video0
> Setting pipeline to PAUSED ...
> ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Device '/dev/video0' is not a output device.
> Additional debug info:
> ../gst-plugins-good/sys/v4l2/v4l2_calls.c(636): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
> Capabilities: 0x5208000
> ERROR: pipeline doesn't want to preroll.
> Failed to set pipeline to PAUSED.
> Setting pipeline to NULL ...
> Freeing pipeline ...
> output to /dev/video0 failed
> ```
> 
> Edit: Looks like this was solved by removing the `exclusive_caps=1` option when starting the `v4l2loopback` kernel module.

Is there a way to prevent this behavior without removing that option? I need that option for Discord and Chrome.