v4l2loopback/v4l2loopback #202: emulate video capture capabilities

lexusburn, 2018-12-26:
* `v4l2loopback` version: 0.12.0
* kernel version: Linux raspberrypi 4.9.80+ #1092 Fri Feb 9 13:36:39 GMT 2018 armv6l GNU/Linux
* Distribution (+version): 
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.3 (stretch)
Release:	9.3
Codename:	stretch

I would like to use v4l2loopback to get a loopback device for the raspberry pi cam (v2).
I installed kernel module and created new loopback device with this command:

> sudo modprobe v4l2loopback video_nr=10 exclusive_caps=1 max_width=1280 max_height=720

then i piped output from pi cam to loopback device with ffmpeg:

> ffmpeg -f video4linux2 -input_format h264 -video_size 1280x720 -framerate 30 -i /dev/video0 -vcodec copy -f v4l2 /dev/video10

I can access the video with some software like motion without problems.

But if i want to use linphone (http://www.linphone.org/, https://wiki.linphone.org/xwiki/wiki/public/view/Linphone/Linphone%20and%20Raspberry%20Pi/) it detects /dev/video0 as only camera device. Is it possible to get v4l2loopback to emulate a v4l2 cam more accurate?

It seems to me that some software like linphone checks the capabilities and don't get "Video Capture" on v4l2loopback and therefore don't accept this as a camera device.

umlaeute, 2022-04-06:
@wmailn it would be good if you could add your `+1` as a :+1: as a comment to the message that you are upvoting...otherwise it's unclear what you are referring to.

umlaeute, 2022-04-06:
also: closing this due to inactivity (or put otherwise: I consider this a bug in `linphone` until there is further evidence)

closed 2022-04-06 by umlaeute

wmailn, 2022-04-06:
> @wmailn it would be good if you could add your `+1` as a 👍 as a comment to the message that you are upvoting...otherwise it's unclear what you are referring to.
Sorry, I mean I have the same problem.