v4l2loopback/v4l2loopback #203: issue piping to loopback dummy device

nygma13, 2019-01-11:
Hello,

I have an issue with using stdout pipes streaming to dummy videodevice created via V4l2loopback

environment :
  * `v4l2loopback` version: 0.12.0
  * kernel version: 4.15.0-43-generic
  * Distribution (+version): Ubuntu 18.04.1

So far, I have no issue with simple commands like : 
`gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video1`
or
`ffmpeg -re -i c.mp4 -f v4l2 /dev/video1`

I can see the result without any error on VLC via /dev/video1

So far I am piping the output of my raspicams to my main computer, via this command, that works fine : 
`ssh user@mypi  "raspivid  -fps 25   -w 1920 -h 1080   -t 0 -o -" | tee test.h264 | mplayer -fps 200  -xy 800 -geometry 1920:0  -cache 1024 -`
records the file and displays the stream fine.

I now wanted to replace mplayer by gstreamer or ffmpeg to pipe the output to /dev/video1, and process it via opencv.

I first tried with ffmpeg :
`ssh user@mypi "raspivid  -fps 25   -w 1920 -h 1080   -t 0 -o -" |  ffmpeg -i pipe: v4l2  /dev/video1`

but i get (whatever the various attributes i tried):
Unable to find a suitable output format for 'v4l2'
v4l2: Invalid argument
Looks like V4l2 would need an option to specify an output, but i did not find anything working.

I then tried with gstreamer : 
ssh user@mypi "raspivid  -fps 25   -w 1920 -h 1080   -t 0 -o -" |  gst-launch-1.0 -e -v --gst-debug-level=2  fdsrc fd=0  !  v4l2sink  device=/dev/video1
but again, an error keeps coming : 
0:00:00.015036590  3120 0x5587e90bb4d0 WARN                GST_POLL gstpoll.c:951:gst_poll_remove_fd: 0x5587e8f52e30: couldn't find fd !
Le pipeline est en phase de PREROLL…
0:00:00.850736038  3120 0x5587e8f52e80 ERROR               v4l2sink gstv4l2sink.c:622:gst_v4l2sink_show_frame:<v4l2sink0> not negotiated
0:00:00.850771352  3120 0x5587e8f52e80 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<fdsrc0> error: Internal data stream error.
0:00:00.850778736  3120 0x5587e8f52e80 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<fdsrc0> error: streaming stopped, reason not-negotiated (-4)

this time, looks like the pipe is not found at all. 

So am i just doing something wrong, am i missing some arguments, or is there an issue piping stdout to the dummy video device ?

thanks

G.A.

umlaeute, 2019-01-11:
this seems to be a problem with the used producers (ffmpeg and gstreamer), and unrelated to `v4l2loopback`.

at least for GStreamer i can say that `fdsrc` provides a low-level interface to file-input (in this case, stdin). GStreamer doesn't magically know what the raw bytes it receives "mean".
However, in order to construct a valid plpeline, GStreamer must know how to interpret the data, so you need to somehow parse that data. at least you will need to manually set a caps-string.

closed 2019-01-11 by umlaeute

nygma13, 2019-01-12:
Hello,

I have set a caps string, and it is not far from working, but i get a new error.
When i google that error, 90% of the results point to a bug in V4l2loopback and your name appears on the replies... ;=)

May I kindly ask you to check ?

`ssh user@mypi "raspivid  -fps 25   -w 1920 -h 1080  -rf rgb -t 0 -o -" | tee test.h264 | gst-launch-1.0 --gst-debug-level=2 -v fdsrc ! h264parse   ! avdec_h264  !  video/x-raw,width=1920,height=1080,framerate=25/1  !  v4l2sink  device=/dev/video1`
```

Définition du pipeline à PAUSED...
Le pipeline est en phase de PREROLL…
0:00:00.650468114 29411 0x558f97905280 WARN                    v4l2 gstv4l2object.c:2508:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0:sink> Unknown frame interval type at YUYV@48x32: 0
0:00:00.650520932 29411 0x558f97905280 WARN                    v4l2 gstv4l2object.c:2508:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0:sink> Unknown frame interval type at UYVY@48x32: 0
0:00:00.650573374 29411 0x558f97905280 WARN                    v4l2 gstv4l2object.c:2508:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0:sink> Unknown frame interval type at 422P@48x32: 0
0:00:00.650599437 29411 0x558f97905280 WARN                    v4l2 gstv4l2object.c:2508:gst_v4l2_object_probe_caps_for_format_and_size:<v4l2sink0:sink> Unknown frame interval type at YU12@48x32: 0
...........
...........
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, width=(int)1920, height=(int)1080, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high-10, level=(string)4, codec_data=(buffer)01640028ffe1001027640028ac2b403c0113f2c03c489a8001000528ee025cb0
/GstPipeline:pipeline0/avdec_h264:avdec_h264-0.GstPad:sink: caps = video/x-h264, width=(int)1920, height=(int)1080, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high-10, level=(string)4, codec_data=(buffer)01640028ffe1001027640028ac2b403c0113f2c03c489a8001000528ee025cb0
0:00:00.809954893 29411 0x558f97905280 WARN                GST_PADS gstpad.c:4224:gst_pad_peer_query:<avdec_h264-0:src> could not send sticky events
/GstPipeline:pipeline0/avdec_h264:avdec_h264-0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)bt709, framerate=(fraction)0/1
0:00:00.811675013 29411 0x558f97905280 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<fdsrc0> error: Internal data stream error.
0:00:00.811690991 29411 0x558f97905280 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<fdsrc0> error: streaming stopped, reason not-negotiated (-4)
ERREUR : de l’élément /GstPipeline:pipeline0/GstFdSrc:fdsrc0 : Internal data stream error.
Information de débogage supplémentaire :
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstFdSrc:fdsrc0:
streaming stopped, reason not-negotiated (-4)
ERREUR : le pipeline refuse de passer en phase de préparation.
Définition du pipeline à NULL…
0:00:00.829511792 29411 0x558f97922870 ERROR                  libav :0:: error while decoding MB 69 10, bytestream -14
Libération du pipeline…
```

umlaeute, 2019-01-12:
does it work, if you replace `v4l2sink`  with `xvimagesink` (or `autovideosink`)

nygma13, 2019-01-12:
hello,

thanks for helping and replying fast.

with xvimagesink :

```
Définition du pipeline à PAUSED...
Le pipeline est en phase de PREROLL…
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, width=(int)1920, height=(int)1080, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high-10, level=(string)4, codec_data=(buffer)01640028ffe1001027640028ac2b403c0113f2c03c489a8001000528ee025cb0
/GstPipeline:pipeline0/avdec_h264:avdec_h264-0.GstPad:sink: caps = video/x-h264, width=(int)1920, height=(int)1080, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high-10, level=(string)4, codec_data=(buffer)01640028ffe1001027640028ac2b403c0113f2c03c489a8001000528ee025cb0
0:00:00.817199813  4865 0x561463468450 WARN                GST_PADS gstpad.c:4224:gst_pad_peer_query:<avdec_h264-0:src> could not send sticky events
/GstPipeline:pipeline0/avdec_h264:avdec_h264-0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)bt709, framerate=(fraction)0/1
0:00:00.818868017  4865 0x561463468450 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<fdsrc0> error: Internal data stream error.
0:00:00.818891979  4865 0x561463468450 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<fdsrc0> error: streaming stopped, reason not-negotiated (-4)
ERREUR : de l’élément /GstPipeline:pipeline0/GstFdSrc:fdsrc0 : Internal data stream error.
Information de débogage supplémentaire :
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstFdSrc:fdsrc0:
streaming stopped, reason not-negotiated (-4)
ERREUR : le pipeline refuse de passer en phase de préparation.
Définition du pipeline à NULL…
0:00:00.821985797  4865 0x7f908c1d5e40 ERROR                  libav :0:: error while decoding MB 119 9, bytestream -6
Libération du pipeline…
```

with autovideosink :

```
Définition du pipeline à PAUSED...
0:00:00.099042984  4878 0x55e9140e0b20 WARN                   vaapi gstvaapiutils.c:77:gst_vaapi_warning: va_getDriverName() failed with unknown libva error,driver_name=(null)
0:00:00.125785267  4878 0x55e9140e0b20 WARN                   vaapi gstvaapiutils.c:77:gst_vaapi_warning: va_getDriverName() failed with unknown libva error,driver_name=(null)
Le pipeline est en phase de PREROLL…
Contexte de l’élément « autovideosink0-actual-sink-vaapi » : gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayDRM\)\ vaapidisplaydrm2";
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, width=(int)1920, height=(int)1080, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high-10, level=(string)4, codec_data=(buffer)01640028ffe1001027640028ac2b403c0113f2c03c489a8001000528ee025cb0
/GstPipeline:pipeline0/avdec_h264:avdec_h264-0.GstPad:sink: caps = video/x-h264, width=(int)1920, height=(int)1080, framerate=(fraction)0/1, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, stream-format=(string)avc, alignment=(string)au, profile=(string)high-10, level=(string)4, codec_data=(buffer)01640028ffe1001027640028ac2b403c0113f2c03c489a8001000528ee025cb0
0:00:00.833559457  4878 0x55e9140ce0f0 WARN                GST_PADS gstpad.c:4224:gst_pad_peer_query:<avdec_h264-0:src> could not send sticky events
0:00:00.838045113  4878 0x55e9140ce0f0 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<fdsrc0> error: Internal data stream error.
0:00:00.838125340  4878 0x55e9140ce0f0 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<fdsrc0> error: streaming stopped, reason not-negotiated (-4)
/GstPipeline:pipeline0/avdec_h264:avdec_h264-0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)bt709, framerate=(fraction)0/1
ERREUR : de l’élément /GstPipeline:pipeline0/GstFdSrc:fdsrc0 : Internal data stream error.
Information de débogage supplémentaire :
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstFdSrc:fdsrc0:
streaming stopped, reason not-negotiated (-4)
ERREUR : le pipeline refuse de passer en phase de préparation.
Définition du pipeline à NULL…
0:00:00.858682791  4878 0x7fd38001c380 ERROR                  libav :0:: error while decoding MB 76 12, bytestream -5
Libération du pipeline…
```

umlaeute, 2019-01-12:
sure.
the thing is, that there *are* issues with `v4l2loopback`. you just have to first eliminate the problems that are unrelated to the loopback module, before we can even start assessing the problem with the module....

nygma13, 2019-01-12:
hehe...

found a way to make it work....

I pipe raspivid to my main computer, then gstream via udp, and read it from the same computer send it to /dev/video1, while saving the file on the main computer.

a bit of a crazy solution, but I am somehow fed up with searching.

sender  (main computer accessing pi remotely)
`ssh user@mypi "raspivid   -w 1920 -h 1080   -t 0 -o -" | gst-launch-1.0 -v fdsrc !  h264parse ! tee name=splitter ! queue ! rtph264pay config-interval=10 pt=96 ! udpsink host=192.168.1.22 port=9000 splitter. ! queue ! filesink location="videofile.h264"`

receiver : (still the main computer, receiving itself !)
`gst-launch-1.0 -v udpsrc port=9000 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' ! rtph264depay ! avdec_h264 ! videoconvert !  v4l2sink  device=/dev/video1`

umlaeute, 2019-01-12:
great you found a way (albeit quirky)

nygma13, 2019-01-12:
Strangely, the stream is every few seconds inverting the colors... so It won't work...

BUT I found the missing info for ffmpeg.  so it works !  
`ssh user@mypi "raspivid -fps 25 -w 1920 -h 1080 -t 0 -o -" | ffmpeg -i pipe: -s 1920x1080 -vcodec rawvideo -f v4l2 /dev/video1`

I got 4 pi cameras streamed to 4 local dummy devices.  (with reasonable processor loads)
And opencv reads it nicely.

I can now record in full resolution the pi camera, and process the stream for movement detection, and shape identification with tensor flow.

it works !

thanks again for your nice software.