v4l2loopback/v4l2loopback #204: V4L2Loopback Support for NV12 Stream in Chromium?

benlemond, 2019-01-17:
I am ingesting data from cameras, doing some processing, and outputting that output out to a v4l2loopback device which I am showing in a web browser in Chromium. Currently, the camera streams are coming in as NV12 input. We have to color convert those to RGB24(I believe) before we can output to the v4l2loopback device and have it shown in Chromium.

My question is, DO we have to convert to RGB24 or something other than NV12? When we convert, all is well and the camera feeds show up appropriately in Chromium via WebRTC (meaning the exclusive_caps are being recognized by the browser). However, when I switch the output to maintain NV12 format, all I get out is a gray box in the browser. Does anyone have an idea what is going on or have some suggestions for utilizing NV12 OUTPUT to v4l2loopback and chromium?
`uname -a`
Linux tegra-ubuntu 4.4.38-tegra #1 SMP PREEMPT Thu May 17 00:15:19 PDT 2018 aarch64 aarch64 aarch64 GNU/Linux

`dmesg | grep v4l2loopback`
[    6.890522] v4l2loopback driver version 0.12.0 loaded

`cat /etc/lsb-release `
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.5 LTS"

`head -n 1 /etc/nv_tegra_release `
#R28 (release), REVISION: 2.1, GCID: 11272647, BOARD: t186ref, EABI: aarch64, DATE: Thu May 17 07:29:06 UTC 2018

#This fails in Chromium:
`gst-launch-1.0 videotestsrc is-live=true ! 'video/x-raw, format=NV12, framerate=30/1' ! tee ! v4l2sink device=/dev/video2`

#This works in Chromium:
`gst-launch-1.0 videotestsrc is-live=true ! 'video/x-raw, format=I420, framerate=30/1' ! tee ! v4l2sink device=/dev/video2`

umlaeute, 2019-01-17:
- can you read the `NV12` stream with some other software (e.g. `gstreamer`)?
- can you read native `NV12` streams with chromium?
- in general, choosing another `YUV`-format will be much faster than converting to `RGB`

umlaeute, 2019-01-17:
here's another working pipeline:
~~~sh
gst-launch-1.0 videotestsrc is-live=true ! 'video/x-raw, format=UYVY, framerate=30/1' ! tee ! v4l2sink device=/dev/video2
~~~

so given that both `UYVY` and `I420` work fine with chromium, i guess that they simply do not support `NV12`.
a direct conversion from `NV12` to `I420` should be pretty cheap, so i'd go with that.

closed 2019-01-17 by umlaeute

benlemond, 2019-01-17:
Hello Umlaeute,

- can you read the NV12 stream with some other software (e.g. gstreamer)?
**Yes.  Once I execute the streams above I can open the /dev/video2 device with `guvcview -d /dev/video2` and I see the appropriate output.**
- can you read native NV12 streams with chromium?
**I am unsure about how to go about testing this?  Suggestions here would be helpful.**

I will try to suggest using another native YUV format.

umlaeute, 2019-01-17:
>> can you read native NV12 streams with chromium?
>>
>I  am unsure about how to go about testing this? Suggestions here would be helpful.

plug in a real camera that provides `NV12`, and see whether it works.