v4l2loopback/v4l2loopback #206: ops on 4.9.0-8-amd64 #1 Debian 4.9.130-2

jackfreed, 2019-01-23:
we are using this commit:

commit f998233aafa92a268a91929705e1b453b98f1b9d
Date:   Thu Nov 15 21:52:38 2018 +0100
protect VP9 and HEVC by #ifdef guards

(with commented V4L2_PIX_FMT_VP9 and V4L2_PIX_FMT_HEVC in v4l2loopback_formats.h)

and are getting kernel ops with the following output:

[2242250.913442] ------------[ cut here ]------------
[2242250.913446] WARNING: CPU: 20 PID: 32641 at /build/linux-EbeuWA/linux-4.9.130/lib/list_debug.c:36 __list_add+0xa6/0xb0
[2242250.913448] list_add double add: new=ffff984dd80ce240, prev=ffff984dd80ce240, next=ffff984dd80cef88.
[2242250.913450] Modules linked in: v4l2loopback(O) nls_utf8 isofs loop veth btrfs xor raid6_pq ufs qnx4 hfsplus hfs minix ntfs vfat msdos fat jfs xfs libcrc32c fuse binfmt_misc videodev media xt_nat nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype br_netfilter xt_CHECKSUM iptable_mangle xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp tun ip6table_filter ip6_tables iptable_filter aufs(O) pci_stub vboxpci(O) overlay vboxnetadp(O) vboxnetflt(O) vboxdrv(O) ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack bridge stp llc intel_powerclamp joydev coretemp kvm_intel kvm ipmi_devintf irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel ipmi_si sg iTCO_wdt iTCO_vendor_support lpc_ich mgag200 dcdbas ttm serio_raw evdev pcspkr
[2242250.913525]  shpchp ipmi_msghandler mfd_core wmi drm_kms_helper drm intel_cstate button i7core_edac i2c_algo_bit intel_uncore acpi_power_meter ioatdma edac_core ip_tables x_tables autofs4 ext4 crc16 jbd2 crc32c_generic fscrypto ecb mbcache dm_mod hid_generic usbhid hid ses enclosure ata_generic scsi_transport_sas sd_mod crc32c_intel ata_piix ehci_pci uhci_hcd aesni_intel aes_x86_64 glue_helper lrw ehci_hcd gf128mul ablk_helper ixgbe cryptd dca psmouse libata megaraid_sas ptp usbcore pps_core usb_common mdio scsi_mod bnx2 [last unloaded: v4l2loopback]
[2242250.913585] CPU: 20 PID: 32641 Comm: imagefreeze0:sr Tainted: G        W IO    4.9.0-8-amd64 #1 Debian 4.9.130-2
[2242250.913586] Hardware name: Dell Inc. PowerEdge R710/07THW3, BIOS 6.4.0 07/23/2013
[2242250.913589]  0000000000000000 ffffffff84d33d74 ffffbe747d3f3c10 0000000000000000
[2242250.913594]  ffffffff84a7a59e ffff984dd80ce240 ffffbe747d3f3c68 ffff984dd80cef88
[2242250.913599]  ffff984dd80ce240 ffff98315f5c71a0 ffff98277c83fa00 ffffffff84a7a61f
[2242250.913604] Call Trace:
[2242250.913607]  [<ffffffff84d33d74>] ? dump_stack+0x5c/0x78
[2242250.913611]  [<ffffffff84a7a59e>] ? __warn+0xbe/0xe0
[2242250.913614]  [<ffffffff84a7a61f>] ? warn_slowpath_fmt+0x5f/0x80
[2242250.913618]  [<ffffffff84d52b76>] ? __list_add+0xa6/0xb0
[2242250.913623]  [<ffffffffc093ddb9>] ? vidioc_dqbuf+0x69/0x290 [v4l2loopback]
[2242250.913635]  [<ffffffffc09153fc>] ? __video_do_ioctl+0x2ac/0x320 [videodev]
[2242250.913640]  [<ffffffff84afad4d>] ? get_futex_key+0x32d/0x3c0
[2242250.913650]  [<ffffffffc0914dfd>] ? video_usercopy+0x25d/0x590 [videodev]
[2242250.913661]  [<ffffffffc0915150>] ? video_ioctl2+0x20/0x20 [videodev]
[2242250.913666]  [<ffffffff850190b0>] ? __switch_to_asm+0x40/0x70
[2242250.913669]  [<ffffffff850190a4>] ? __switch_to_asm+0x34/0x70
[2242250.913672]  [<ffffffff850190b0>] ? __switch_to_asm+0x40/0x70
[2242250.913675]  [<ffffffff850190a4>] ? __switch_to_asm+0x34/0x70
[2242250.913677]  [<ffffffff850190b0>] ? __switch_to_asm+0x40/0x70
[2242250.913688]  [<ffffffffc091062f>] ? v4l2_ioctl+0xdf/0xf0 [videodev]
[2242250.913692]  [<ffffffff84c1f442>] ? do_vfs_ioctl+0xa2/0x620
[2242250.913695]  [<ffffffff84c21e16>] ? SyS_ppoll+0xf6/0x190
[2242250.913699]  [<ffffffff84c1fa34>] ? SyS_ioctl+0x74/0x80
[2242250.913702]  [<ffffffff84a03b7d>] ? do_syscall_64+0x8d/0xf0
[2242250.913707]  [<ffffffff85018f8e>] ? entry_SYSCALL_64_after_swapgs+0x58/0xc6
[2242250.913709] ---[ end trace fabcf5ed005ee886 ]---

any hint?

thanks in advance!

jackfreed, 2019-01-24:
we commented it because we were getting unknown pixel format errors in dmesg.

what we been doing in the first place, we grabbed the example pipe from here 

https://gstreamer.freedesktop.org/data/doc/gstreamer/head/gstreamer-plugins/html/gstreamer-plugins-dataurisrc.html

and wanted to sink it into video0 like this:

gst-launch-1.0 -v dataurisrc uri="data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAABAAAAAQCAYAAAAf8/9hAAAAfElEQVQ4je2MwQnAIAxFgziA4EnczIsO4MEROo/gzZWc4xdTbe1R6LGRR74heYS7iKElzfcMiRnt4hf8gk8EayB6luefue/HzlJfCA50XsNjYRxprZmenXNIKSGEsC+QUqK1hhgj521BzhnWWiilUGvdF5RS4L2HMQZCCJy8sHMm2TYdJAAAAABJRU5ErkJggg==" ! pngdec ! videoconvert ! imagefreeze ! videoconvert ! **video/x-raw, format=I420** ! v4l2sink device=/dev/video0

we added the format caps, otherwise the pipeline would not preroll.

jackfreed, 2019-01-24:
sorry didnt have it and hand, it was **Unknown pixelformat 0x30395056**

umlaeute, 2020-09-01:
could you please try again with the latest and greated `v4l2loopback` release, and report whether this is still an issue?

GeniusOnline, 2021-11-19:
hi, i have been using a specific pipeline to achieve the same goal. in testing with imagefreeze i noticed that png makes imagefreeze error out, so using jpg fixed it. though i cannot stream it to the v4l2loopback device.

```
pi@raspberrypi:~ $ gst-launch-1.0 -v filesrc location=img0000.jpg ! jpegdec ! imagefreeze ! videoconvert ! "video/x-raw,width=1920,height=1080,framerate=25/1,format=YUY2" ! v4l2sink device=/dev/video90
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstJpegDec:jpegdec0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstImageFreeze:imagefreeze0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)25/1
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)25/1, format=(string)YUY2, colorimetry=(string)2:4:7:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)25/1, format=(string)YUY2, colorimetry=(string)2:4:7:1
/GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0.GstPad:sink: caps = video/x-raw, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)25/1, format=(string)YUY2, colorimetry=(string)2:4:7:1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)25/1, format=(string)YUY2, colorimetry=(string)2:4:7:1
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)25/1
/GstPipeline:pipeline0/GstImageFreeze:imagefreeze0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)0/1
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstImageFreeze:imagefreeze0: Internal data stream error.
Additional debug info:
gstimagefreeze.c(851): gst_image_freeze_src_loop (): /GstPipeline:pipeline0/GstImageFreeze:imagefreeze0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.040240009
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
pi@raspberrypi:~ $
```

please help :)