v4l2loopback/v4l2loopback #208: ERROR: could not insert 'v4l2loopback': Exec format error

nygma13, 2019-02-08:
Hello,

I had an update to linux-headers-4.15.0-45-generic this evening, and got the following issues : 
`modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/4.15.0-45-generic`
Obviously, i recompiled V4l2loopback.

```
make -C /lib/modules/`uname -r`/build M=/home/nygma/Bureau/v4l2loopback modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-45-generic »
  INSTALL /home/nygma/Bureau/v4l2loopback/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-45-generic
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-45-generic »

```
I think I had the SSL error the last time already, so i did not get too worried about it.

then

`sudo depmod -a`

then 

`sudo modprobe v4l2loopback`
and got : 
`modprobe: ERROR: could not insert 'v4l2loopback': Exec format error`
i did 
`sudo dmesg  | grep -i v4l2loopback`
and got
`v4l2loopback: version magic '4.15.0-43-generic SMP mod_unload ' should be '4.15.0-45-generic SMP mod_unload '`

I googled the error SMP mod_unload, and did not find any applicable solution.

Any clue ?

thanks

umlaeute, 2019-02-10:
the error basically says that `4.15.0-43-generic` is not the same as `4.15.0-45-generic` which is, well... true. the two strings look like kernel versions. so i think you are trying to load the module compiled for one kernel version, into a different kernel version.

you probably just need to run `make clean` (to clean up the previous build against the old kernel version) before doing a `make`

closed 2019-02-10 by umlaeute

nygma13, 2019-02-10:
Hello,

make clean did not improve anything.
(I had anyway recompiled fresh from the latest version from github)

Let me say it differently:

What is a V4l2loopback user supposed to do when there is a kernel update ?

thanks

umlaeute, 2019-02-10:
recompile the module. (like any other kernel module)

nygma13, 2019-02-11:
this is what i did.

I am not that stupid ;=)

the issue is that the module is inside the new kernel, with the vermagic of the old kernel.

sudo modinfo v4l2loopback
filename:       /lib/modules/4.15.0-45-generic/extra/home2/AI/v4l2/v4l2loopback-master/v4l2loopback.ko
license:        GPL
description:    V4L2 loopback video device
srcversion:     F970FD08DB80D838C18FFD7
depends:        videodev
retpoline:      Y
name:           v4l2loopback
vermagic:       4.15.0-43-generic SMP mod_unload 

I tried to force the modprobe without success.

nygma13, 2019-02-11:
Got it.

I compiled it from a certain directory, some weeks ago. 
I deleted that directory in the mean time.

So the recompiling was just creating a second instance of V4l2loopback  in /lib/modules/4.15.0-45-generic/extra/, but not deleting the first one. 
calling the module was in fact calling the first one, (obsolete)

So i deleted everything related to V4L2loopback  in the /extra directory, and recompiled fresh  ;=)

umlaeute, 2019-02-11:
great!

so just to recap: was there anything on the `v4l2loopback` side that could have prevented the problem? PRs are welcome (e.g. for clarifying documentation)

nygma13, 2019-02-11:
The key issue was that my first compile was made from this directory:
 /home2/AI/v4l2/v4l2loopback-master/
this my secondary harddrive.

So somehow, the module was inserted in the kernel not in 
/lib/modules/4.15.0-45-generic/extra/
(like the new compiled one)
but in this 
 /lib/modules/4.15.0-45-generic/extra**/home2/AI/v4l2/v4l2loopback-master/**v4l2loopback.ko

this copied path was what generated the issue. 

When i recompiled, from another directory in my main hard drive  (home/me/Desktop/v4l2loopback), the .ko file was copied straight to /lib/modules/4.15.0-45-generic/extra/, but without the path to it .

So the real question not to have it happen again is:
Who writes the .ko file to the kernel, and why was the original path copied too ?