v4l2loopback/v4l2loopback #209: unable to compile to RPi 3

janeksz, 2019-02-25:
### Step 1: Read this

Thanks for your feedback. It is invaluable for making `v4l2loopback` a better
software.

To help us making the most of your feedback (so we can e.g. fix bugs more
quickly), please make sure to provide the following information.

#### Accessibility
Sometimes pictures say more.
However, mostly they prevent the use of advanced
tools (like "search" or "copy&paste").
And always they prevent people who don't use graphical browser to access the
tracker from reading your content.
So, to make the web a better place, we ask you to post *text* rather than
*screenshots of text* whenever feasible (pretty much always).

#### Cruft

Please exclude these instructions (and other non-relevant information) from your
report.

### Step 2: Describe your environment

  * `v4l2loopback` version: _____

          sudo dmesg  | grep -i v4l2loopback

  * kernel version: _____

          uname -a

  * Distribution (+version): _____

          lsb_release -a
  
### Step 3: Describe the problem:

#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

  ```
  // TODO(you): code here to reproduce the problem
  ```

janeksz, 2019-02-25:
RPi 3 B+
root@RPi3AP:/temp/v4l2loopback# uname -a
Linux RPi3AP 4.14.97-v7+ #1197 SMP Mon Feb 4 21:13:29 GMT 2019 armv7l GNU/Linux
root@RPi3AP:/temp/v4l2loopback#

root@RPi3AP:/temp/v4l2loopback# make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/temp/v4l2loopback modules
make[1]: *** /lib/modules/4.14.97-v7+/build: No such file or directory.  Stop.
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2
root@RPi3AP:/temp/v4l2loopback#

closed 2019-02-25 by umlaeute

janeksz, 2019-02-26:
@umlaeute 

I would like to compile and install the v4lloopback to view the thermal map from MLX90640.
I have uv4l running following the instruction here http://www.linux-projects.org/uv4l/installation/ 
I need your help in installing v4l2loopback for the purpose of streaming thermal map from MLX90640 sensor. I'm new to that and therefore forgive me if I'm doing something stupid. 

**1.** I'm using the latest version Released v4l2loopback v0.12.1 

**2.** My Raspbian is: root@RPi3AP:/temp/v4l2loopback# uname -a
Linux RPi3AP 4.14.97-v7+ #1197 SMP Mon Feb 4 21:13:29 GMT 2019 armv7l GNU/Linux
root@RPi3AP:/temp/v4l2loopback#
root@RPi3AP:/temp/v4l2loopback#  lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.6 (stretch)
Release:        9.6
Codename:       stretch
root@RPi3AP:/temp/v4l2loopback#

**3.** When trying to compile I have the following error:
root@RPi3AP:/temp/v4l2loopback# make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/temp/v4l2loopback modules
**make[1]: *** /lib/modules/4.14.97-v7+/build: No such file or directory.  Stop.**
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2
root@RPi3AP:/temp/v4l2loopback#

Please help

janeksz, 2019-02-26:
I follow the instruction from here https://raspberrypi.stackexchange.com/questions/42197/getting-kernel-headers-for-raspberry-pi-2 
sudo apt-get install build-essential linux-headers-`uname -r` dkms 

root@RPi3AP:/temp/v4l2loopback# sudo apt-get install build-essential linux-headers-`uname -r` dkms
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-headers-4.14.97-v7
E: Couldn't find any package by glob 'linux-headers-4.14.97-v7'
E: Couldn't find any package by regex 'linux-headers-4.14.97-v7'
root@RPi3AP:/temp/v4l2loopback#

umlaeute, 2019-02-26:
your issue is not with `v4l2loopback` but with compiling kernel modules in general.
i really want to keep issues in *this* tracker *on-topic*.

so please use other (raspberry related) forums to research your problem.

janeksz, 2019-02-26:
@umlaeute 
fair enough, thanks