v4l2loopback/v4l2loopback #210: videotestsrc doesnt work on v4l2sink for me

synonymousz, 2019-03-01:
* `v4l2loopback` version: 0.10.0
  * kernel version: `Linux xxx-desktop 4.18.0-15-generic #16~18.04.1-Ubuntu SMP Thu Feb 7 14:06:04 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`
  * Distribution (+version): 
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic
```

I want to use my Canon DSLR-Camera as Webcam and come across with a problem already before even use the feed. When I use the command to show the videotestsrc on xvimagesink it works, but when I try to push it to /dev/video1 it doesn't.

`gst-launch-1.0 videotestsrc ! xvimagesink` - works and gives me a small window with the testvideo
`gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video1` - doesn't work and gives me following errors:
```
Leitung wird auf PAUSIERT gesetzt ...
Leitung läuft vor …
FEHLER: Von Element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Zusätzliche Fehlerdiagnoseinformation:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason not-negotiated (-4)
FEHLER: Leitung läuft nicht vor.
Leitung wird auf NULL gesetzt ...
Leitung wird geleert ...
```
Let me know if you need further information (and how i can gather it)

Thanks in advance

nekorug, 2019-03-02:
I'm having the same or a very similar issue.

Running `Arch Linux 4.20.13-arch1-1-ARCH`, `v4l2loopback-dkms` [package from the AUR](https://aur.archlinux.org/packages/v4l2loopback-dkms/) version `0.11.0-1`. 

From my limited testing of `obs-v4l2sink` which requires a loaded v4l2loopback device, [this is the line](https://github.com/CatxFish/obs-v4l2sink/blob/master/src/v4l2sink.cpp#L158) that returns an error in their code:

```
v4l2_fmt.type = V4L2_BUF_TYPE_VIDEO_OUTPUT;
ret = ioctl(out_data->v4l2_fd, VIDIOC_G_FMT, &v4l2_fmt);
```

From my limited understanding it seems like the video device doesn't report supporting that type, a video output? But I don't really know enough about v4l2 works to fully understand it. If I can help debug it or post additional information I'd love to.

Here's the error message above in English if it helps:

```
> gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video0
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
```

Any way I can help out debugging this, provide additional info, or testing patches is welcome.

synonymousz, 2019-03-03:
What do I need to do exactly? @nekorug

nekorug, 2019-03-05:
> What do I need to do exactly? @nekorug

@synonymousz You need to install the latest version, v.0.12.1. If it's not in the repo you're using you should compile and install it from source.

closed 2019-03-05 by umlaeute