v4l2loopback/v4l2loopback #211: yocto layer

mmar2228, 2019-03-12:
### Step 1: Read this

Thanks for your feedback. It is invaluable for making `v4l2loopback` a better
software.

To help us making the most of your feedback (so we can e.g. fix bugs more
quickly), please make sure to provide the following information.

#### Accessibility
Sometimes pictures say more.
However, mostly they prevent the use of advanced
tools (like "search" or "copy&paste").
And always they prevent people who don't use graphical browser to access the
tracker from reading your content.
So, to make the web a better place, we ask you to post *text* rather than
*screenshots of text* whenever feasible (pretty much always).

#### Cruft

Please exclude these instructions (and other non-relevant information) from your
report.

### Step 2: Describe your environment

  * `v4l2loopback` version: _____

          sudo dmesg  | grep -i v4l2loopback

  * kernel version: _____

          uname -a

  * Distribution (+version): _____

          lsb_release -a
  
### Step 3: Describe the problem:

#### Steps to reproduce:

  1. _____
  2. _____
  3. _____
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

  ```
  // TODO(you): code here to reproduce the problem
  ```

umlaeute, 2019-03-13:
i don't know what a "yocto layer" is and i doubt that your problem is related to `v4l2loopback` - it sounds more like a system integration problem on how to add arbitrary parts to your image.

I therefore suggest to consult the support/manual/... or your "yocto layer" provider. (feel free to convince me that this is indeed a `v4l2loopback` problem)

closed 2019-03-13 by umlaeute

mmar2228, 2019-03-13:
Sure, it is not a problem of your v4l2loopback module. It is a way to install it in a embedded system when you buid the file system. I need v4l2loopback in my target board and I would like to know if somebody caan help me.

TopperBG, 2022-02-10:
> Sure, it is not a problem of your v4l2loopback module. It is a way to install it in a embedded system when you buid the file system. I need v4l2loopback in my target board and I would like to know if somebody can help me.

DId you manage to install it within bitbake recipe ?