v4l2loopback/v4l2loopback #212: modprobe: ERROR: could not insert 'v4l2loopback': Invalid argument

mmar2228, 2019-03-13:
Dear all,
I get the following message when I run "modprobe v4l2loopback":

**_v4l2loopback: disagrees about version of symbol device_create_file
v4l2loopback: Unknown symbol device_create_file (err -22)
v4l2loopback: disagrees about version of symbol dev_err
v4l2loopback: Unknown symbol dev_err (err -22)
v4l2loopback: disagrees about version of symbol device_remove_file
v4l2loopback: Unknown symbol device_remove_file (err -22)
modprobe: ERROR: could not insert 'v4l2loopback': Invalid argument__**

Any idea how could I fix it?

Thanks in advance!

umlaeute, 2019-03-14:
you could start by reading the `README`.
it says:

~~~
make install
depmod -a
~~~

closed 2019-03-14 by umlaeute

mmar2228, 2019-03-14:
I double and triple checked that before writting here. And it still does not work.
Don't worry, I will try solve it myself.

umlaeute, 2019-03-15:
i actually missed parts of your error-message, which indicate the actual problem.

see https://unix.stackexchange.com/questions/125624/ on how to solve it.

mmar2228, 2019-03-15:
Yes, it was the solution. I run "make modules" inside my kernel source directory /usr/src/kernel.

Thanks you for your reply!