v4l2loopback/v4l2loopback #213: gstreamer copy from webcam to dummy device

Maobuff, 2019-03-16:
v4l2loopback driver version 0.12.1
kernel 4.14.98-v7+
raspbian stretch

cloning webcam using gstreamer 
gst-launch-1.0 v4l2src ! v4l2sink device=/dev/video1
provide error : libv4l2 : error getting pixformat: Invalid argumen

but when inserting videorate with caps in pipeline (videorate ! image/jpeg,framerate=30/1) im getting same error, but its actually working (with some image random artifacts)

umlaeute, 2019-03-16:
works for me (on my laptop with `linux-4.19.0-3-amd64`).

Maobuff, 2019-03-16:
well:) what is yours gstreamer version?

umlaeute, 2019-03-16:
what is yours?

~~~
$ gst-launch-1.0 --gst-version
GStreamer Core Library version 1.14.4
~~~

Maobuff, 2019-03-18:
1.10.4
I think gst is incapable of putting coded stream to v4l2sink (only raw pixel format)
Its looks like not implement feature in gst or its just impossible.

umlaeute, 2019-03-18:
this could well be. my camera doesn't produce any compressed video stream, so i cannot test this.
so you probably have to put a decoder into your pipeline.

closing this, as the problem seems to be related to gstreamer.

closed 2019-03-18 by umlaeute