v4l2loopback/v4l2loopback #216: Raspberry Pi -modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/4.19.34-v7+

ernestgwilsonii, 2019-04-14:
### Step 2: Describe your environment

```
uname -a
Linux raspberrypi 4.19.34-v7+ #1211 SMP Mon Apr 8 22:56:37 BST 2019 armv7l GNU/Linux
```
### Step 3: Describe the problem:

Trying to compile and use v4l2loopback on Raspberry Pi Raspbian.
Getting error during:
```
make install
```
Error:
```
make install
make -C /lib/modules/`uname -r`/build M=/tmp/v4l2loopback modules_install
make[1]: Entering directory '/root/linux-6f84467f66ba847e643489a3d18b1cf81bad87d0'
  INSTALL /tmp/v4l2loopback/v4l2loopback.ko
  DEPMOD  4.19.34-v7+
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/root/linux-6f84467f66ba847e643489a3d18b1cf81bad87d0'
```
and then getting error while attempting to use:
```
modprobe v4l2loopback video_nr=100
```
Error:
```
modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/4.19.34-v7+
```

#### Steps to reproduce:

 ```
rpi-update && reboot
rpi-source -v
apt-get install raspberrypi-kernel-headers build-essential linux-libc-dev-armhf-cross patch patchutils bison flex
apt autoremove -y
cd /lib/modules/4.19.34-v7+/build
cat /proc/config.gz > /lib/modules/4.19.34-v7+/build/.config
zcat /proc/config.gz > /usr/src/linux-headers-4.14.98-v7+/.config
make oldconfig && make prepare && make scripts

cd /tmp
git clone https://github.com/umlaeute/v4l2loopback.git
cd v4l2loopback
make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/tmp/v4l2loopback modules
make[1]: Entering directory '/root/linux-6f84467f66ba847e643489a3d18b1cf81bad87d0'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/root/linux-6f84467f66ba847e643489a3d18b1cf81bad87d0'

make install
make -C /lib/modules/`uname -r`/build M=/tmp/v4l2loopback modules_install
make[1]: Entering directory '/root/linux-6f84467f66ba847e643489a3d18b1cf81bad87d0'
  INSTALL /tmp/v4l2loopback/v4l2loopback.ko
  DEPMOD  4.19.34-v7+
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/root/linux-6f84467f66ba847e643489a3d18b1cf81bad87d0'

modprobe v4l2loopback video_nr=100
modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/4.19.34-v7+
```
  
#### Observed Results:

  * See above - got an error during: make install
  * See above - got an error attempting to execute

umlaeute, 2019-04-22:
- what is the "weird error" you are mentioning (i don't see any)?
- raise the verbosity by (always) adding `V=1` to the make arguments.
- can you use `make modprobe` to load the module (from the build directory),

please also see go through *all* the other issues where rasbpi-users have not been able to compile and found that they just generic problems with raspbian.

ernestgwilsonii, 2019-04-23:
The "weird error" (shown previously above is a warning) is during
make install
```
Warning: modules_install: missing 'System.map' file. Skipping depmod.
```
And it never installs anything.

umlaeute, 2019-04-23:
the "warning" is a warning, not an error. it skips the calculation of module dependencies. it is unrelated to installation problems.

raising the verbosity should give more information.

ernestgwilsonii, 2019-04-24:
This was my missing step:
```
depmod -a
```
Then this worked:
```
make modprobe # Although likely not needed
modprobe v4l2loopback video_nr=100
```
Now I see: /dev/video100

Sorry for the false alarm here!
Thanks for responding!
And thanks for making cool stuff and sharing it with the world!

closed 2019-04-24 by ernestgwilsonii