v4l2loopback/v4l2loopback #222: v4l2loopback and mpegts

MichaelDenczek, 2019-05-15:
v4l2loopback fails for communicating from GStreamer to FFMPEG using video/mpegts, MPEG Transport Stream.  GStreamer version 1.8.3, FFMPEG version 4.1.3.

v4l2loopback version: 0.9.1
kernel version: Linux tegra-ubuntu 4.4.38-tegra #1 SMP PREEMPT Wed Feb 21 12:05:07 CST 2018 aarch64 aarch64 aarch64 GNU/Linux
distribution: Ubuntu 16.04.5 LTS

Using GStreamer as the producer, the pipeline takes the output of the mpegtsmux element to the v4l2sink after a tee element.

The GStreamer pipeline runs, but FFMPEG fails using the virtual device indicating it cannot find a proper format for codec 'none'.

To reproduce:

Start GStreamer with virtual device and testvideosrc:

_gst-launch-1.0 videotestsrc ! "video/x-raw, width=1280, height=720, format=I420, framerate=30/1" ! omxh264enc ! 'video/x-h264, stream-format=byte-stream' ! mpegtsmux ! v4l2sink device="/dev/video1"_

And attempt to capture stream using FFMPEG:

_ffmpeg -hide_banner -re -f video4linux2 -i /dev/video6 -f null /dev/null_

Which results in FFMPEG indicating the following error:

_[video4linux2,v4l2 @ 0x666160] Cannot find a proper format for codec 'none' (id 0), pixel format 'none' (id -1)
Assertion *codec_id != AV_CODEC_ID_NONE failed at libavdevice/v4l2.c:811
Aborted_

I was really hoping I would be able to use v4l2loopback to transfer MPEG Transport Stream between GStreamer and FFMPEG.

umlaeute, 2019-05-15:
unfortunately i don't have an `omxh264enc` GStreamer element, so I cannot reproduce.

apart from that: why are you writing to `/dev/video1` but reading from `/dev/video6`?

MichaelDenczek, 2019-05-16:
The x264enc element should work in place of the omxh264enc element.  /dev/video6 was a type which is now fixed as /dev/video1 (output from the GStreamer pipeline.