v4l2loopback/v4l2loopback #226: Cannot Make 5.0.0-15-generic

Capitalmind, 2019-07-04:
### Step 1: Read this

Thanks for your feedback. It is invaluable for making `v4l2loopback` a better
software.

To help us making the most of your feedback (so we can e.g. fix bugs more
quickly), please make sure to provide the following information.

#### Accessibility
Sometimes pictures say more.
However, mostly they prevent the use of advanced
tools (like "search" or "copy&paste").
And always they prevent people who don't use graphical browser to access the
tracker from reading your content.
So, to make the web a better place, we ask you to post *text* rather than
*screenshots of text* whenever feasible (pretty much always).

#### Cruft

Please exclude these instructions (and other non-relevant information) from your
report.

### Step 2: Describe your environment

  * `v4l2loopback` version: [15603.240792] v4l2loopback driver version 0.12.2 loaded

          sudo dmesg  | grep -i v4l2loopback

  * kernel version: Linux zuultux 5.0.0-15-generic #16~18.04.1-Ubuntu SMP Tue May 7 14:17:37 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

          uname -a

  * Distribution (+version): Linux Mint 19.1 Tessa

          lsb_release -a

### Step 3: Describe the problem:

# make install
make -C /lib/modules/`uname -r`/build M=/home/zuultux/Downloads/GIT/v4l2loopback modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-15-generic'
  INSTALL /home/zuultux/Downloads/GIT/v4l2loopback/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.0.0-15-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-15-generic'

#### Steps to reproduce:

  1. git clone latest
  2. $ make
  3. # make install
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.
  
#### Expected Results:

  * What did you expect to happen?
  
#### Relevant Code:

  ```
  // TODO(you): code here to reproduce the problem
  ```

umlaeute, 2019-07-07:
1. please remove the all the "Cruft" from the template.
2. i don't see any error.

closed 2019-07-07 by umlaeute

JaceyPenny, 2020-02-13:
@umlaeute it is bad form to say "I don't see any error" when there are multiple lines of output containing the term "error". You ought to at least link to where you have already answered the question if you do not feel like answering a question.

@Capitalmind your question was answered [here](https://github.com/umlaeute/v4l2loopback/issues/122#issuecomment-262725719). Basically, these errors are printed, but it is not the case that installation failed (despite that there is no indication of success in the output).