v4l2loopback/v4l2loopback #227: use v4l2loopback on alpine linux ?

vrx, 2019-07-08:
I have been using v4l2loopback on debian based systems for a couple of years now, thanks !!
I recently tried to try to compile the module for alpine based systems but a couple of dependencies like vmalloc.h and others seem out of reach (at least to my eyes), is there any chance that a port to these systems is foreseeable ?

Best regards,
V

umlaeute, 2019-07-09:
`v4l2loopback` is distribution agnostic, but it depends on a functional build system for kernel modules, which includes the kernel headers and a proper compiler.

if alpine does not provide such out of the box, this must be fixed in alpine.

closed 2019-07-09 by umlaeute

vrx, 2019-07-10:
thanks for your quick reply !
alpine has kernel headers and compilers, but being based on musl libc, it seems that some headers are missing (again as far as my limited understanding of kernel modules compilation is concerned).

umlaeute, 2019-07-10:
i don't think that the (userspace) libc-implementation has anything to do with the kernel headers.

umlaeute, 2019-07-10:
just to confirm:

~~~
$ vagrant init alpine310
[...]
$ vagrant up
[...]
$ vagrant ssh
[...]
$ uname -a
Linux localhost 4.19.52-0-virt #1-Alpine SMP Tue Jun 18 13:00:27 UTC 2019 x86_64 Linux
$ sudo bash
# apk add git linux-virt-dev gcc make
[...]
# exit
$ git clone https://github.com/umlaeute/v4l2loopback
[...]
$ make 
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/vagrant/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.52-0-virt'
  CC [M]  /home/vagrant/v4l2loopback/v4l2loopback.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: "video_ioctl2" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "__video_register_device" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "v4l2_ctrl_new_custom" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "v4l2_ctrl_handler_init_class" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "video_device_release" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "video_device_alloc" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "v4l2_device_register" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "v4l2_ctrl_handler_free" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "v4l2_device_unregister" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "video_unregister_device" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
WARNING: "video_devdata" [/home/vagrant/v4l2loopback/v4l2loopback.ko] undefined!
  CC      /home/vagrant/v4l2loopback/v4l2loopback.mod.o
  LD [M]  /home/vagrant/v4l2loopback/v4l2loopback.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.19.52-0-virt'
$ make modprobe
[...]
modprobe: FATAL: Module videodev not found in directory /lib/modules/4.19.52-0-virt
$ 
~~~

so the module builds without problems, but fails to load.
it fails to load, alpine's `4.19.52-0-virt` kernel lacks `videodev`.
That's probably due to the `virt`ual nature of the kernel (for use in VMs), and shouldn't be a deal when running the `vanilla` kernel.

umlaeute, 2019-07-10:
indeed, running a `vanilla` kernel and installing the `linux-vanilla-dev` package (instead of the `linux-virt-dev` package), i can compile and load the module just fine.

vrx, 2019-11-24:
great, thank you !