v4l2loopback/v4l2loopback #228: modprobe: ERROR: could not insert 'v4l2loopback': Unknown symbol in module, or unknown parameter

code-mage, 2019-07-29:
* `v4l2loopback` version: _____
```
[  574.063152] v4l2loopback: loading out-of-tree module taints kernel.
[  574.075624] v4l2loopback: Unknown symbol __sanitizer_cov_trace_switch (err -2)
[  574.075917] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp4 (err -2)
[  574.076036] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp8 (err -2)
[  574.076238] v4l2loopback: Unknown symbol __sanitizer_cov_trace_pc (err -2)
[  574.076576] v4l2loopback: Unknown symbol __sanitizer_cov_trace_cmp4 (err -2)
[  750.047811] v4l2loopback: Unknown symbol __sanitizer_cov_trace_switch (err -2)
[  750.048100] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp4 (err -2)
[  750.048219] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp8 (err -2)
[  750.048419] v4l2loopback: Unknown symbol __sanitizer_cov_trace_pc (err -2)
[  750.048757] v4l2loopback: Unknown symbol __sanitizer_cov_trace_cmp4 (err -2)
[ 1627.338590] v4l2loopback: Unknown symbol __sanitizer_cov_trace_switch (err -2)
[ 1627.338882] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp4 (err -2)
[ 1627.338999] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp8 (err -2)
[ 1627.339201] v4l2loopback: Unknown symbol __sanitizer_cov_trace_pc (err -2)
[ 1627.347087] v4l2loopback: Unknown symbol __sanitizer_cov_trace_cmp4 (err -2)
[ 1637.969323] v4l2loopback: Unknown symbol __sanitizer_cov_trace_switch (err -2)
[ 1637.969766] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp4 (err -2)
[ 1637.969891] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp8 (err -2)
[ 1637.970099] v4l2loopback: Unknown symbol __sanitizer_cov_trace_pc (err -2)
[ 1637.970443] v4l2loopback: Unknown symbol __sanitizer_cov_trace_cmp4 (err -2)
[ 1836.945197] v4l2loopback: Unknown symbol __sanitizer_cov_trace_switch (err -2)
[ 1836.945491] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp4 (err -2)
[ 1836.945612] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp8 (err -2)
[ 1836.945816] v4l2loopback: Unknown symbol __sanitizer_cov_trace_pc (err -2)
[ 1836.946162] v4l2loopback: Unknown symbol __sanitizer_cov_trace_cmp4 (err -2)
[ 2379.881733] v4l2loopback: Unknown symbol __sanitizer_cov_trace_switch (err -2)
[ 2379.882023] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp4 (err -2)
[ 2379.882142] v4l2loopback: Unknown symbol __sanitizer_cov_trace_const_cmp8 (err -2)
[ 2379.882346] v4l2loopback: Unknown symbol __sanitizer_cov_trace_pc (err -2)
[ 2379.882690] v4l2loopback: Unknown symbol __sanitizer_cov_trace_cmp4 (err -2)
```
  * kernel version: _____

         Linux raspberrypi 4.19.60+ #1247 Thu Jul 25 14:28:41 BST 2019 armv6l GNU/Linux

  * Distribution (+version): _____

          No LSB modules are available.
			Distributor ID:	Raspbian
			Description:	Raspbian GNU/Linux 10 (buster)
			Release:	10
			Codename:	buster
  
#### Steps to reproduce:

  1. sudo aptitude install v4l2loopback-dkms
  2. sudo modprobe v4l2loopback devices=2
  
#### Observed Results:

  modprobe: ERROR: could not insert 'v4l2loopback': Unknown symbol in module, or unknown parameter (see dmesg)

  
This also happened when I did `make && sudo make install` on the cloned repo.

Any ideas on what is causing the error? And short of a system cleanup and reinstalling the kernel headers, any instructions on fixing it?

umlaeute, 2019-08-14:
this looks as if you are using a custom compiler that cannot be used to build kernel modules, because it differs from the compiler that was used when building the kernel (`afl-gcc`, `clang` or somesuch).

can you build *any* other kernel-module? (check with some random `-dkms` package)

Hello71, 2019-11-01:
or headers do not match. looks like CONFIG_KASAN or something was set in headers but not kernel.

wmealing, 2020-02-19:
The symbols can't be found because your build environment differs from your module build environment.  Fix that and the problem will go away.

umlaeute, 2020-02-26:
i tend to agree with the other posters that your running kernel and headers to not match, so closing this,

closed 2020-02-26 by umlaeute

celvin, 2020-05-09:
remove it:
sudo modprobe -r v4l2loopback
find "/lib/modules/$(uname -r)" | grep v4l2loopback | sudo xargs rm

then:
sudo depmod -a
make
sudo make install
sudo modprobe v4l2loopback

then you should see the device list by:
v4l2-ctl --list-devices