v4l2loopback/v4l2loopback #230: Cannot complete make install on ubuntu 19.04 (crypto?)

polerin, 2019-08-20:
### Step 2: Describe your environment

  * `v4l2loopback` version:
0.12.2

  * kernel version:
5.0.0-25-generic #26-Ubuntu SMP Thu Aug 1 12:04:58 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

  * Distribution (+version): 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.04
Release:	19.04
Codename:	disco

  
### Step 3: Describe the problem:
Unable to complete `sudo make install` fails with

#### Steps to reproduce:

  1. clone repo from github, cd into checkout
  2. make
  3. make install
  
#### Observed Results:
```
polerin@dev-systems:~/temp/v4l2loopback-master$ make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/polerin/temp/v4l2loopback-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-25-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-25-generic'
polerin@dev-systems:~/temp/v4l2loopback-master$ make install
make -C /lib/modules/`uname -r`/build M=/home/polerin/temp/v4l2loopback-master modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-25-generic'
  INSTALL /home/polerin/temp/v4l2loopback-master/v4l2loopback.ko
cp: cannot create regular file '/lib/modules/5.0.0-25-generic/extra/v4l2loopback.ko': Permission denied
make[2]: *** [scripts/Makefile.modinst:40: /home/polerin/temp/v4l2loopback-master/v4l2loopback.ko] Error 1
make[1]: *** [Makefile:1619: _emodinst_] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-25-generic'
make: *** [Makefile:47: install] Error 2
```

ls of the kernal directory

```
polerin@dev-systems:~/temp/v4l2loopback-master$ ls -la /usr/src/linux-headers-5.0.0-25-generic/
total 2012
drwxr-xr-x  8 root root    4096 Aug 14 06:00 .
drwxr-xr-x 10 root root    4096 Aug 15 06:49 ..
drwxr-xr-x  3 root root    4096 Aug 14 06:00 arch
lrwxrwxrwx  1 root root      31 Aug  1 05:04 block -> ../linux-headers-5.0.0-25/block
lrwxrwxrwx  1 root root      31 Aug  1 05:04 certs -> ../linux-headers-5.0.0-25/certs
-rw-r--r--  1 root root  224293 Aug  1 05:04 .config
-rw-r--r--  1 root root  224417 Aug  1 05:04 .config.old
lrwxrwxrwx  1 root root      32 Aug  1 05:04 crypto -> ../linux-headers-5.0.0-25/crypto
lrwxrwxrwx  1 root root      39 Aug  1 05:04 Documentation -> ../linux-headers-5.0.0-25/Documentation
lrwxrwxrwx  1 root root      33 Aug  1 05:04 drivers -> ../linux-headers-5.0.0-25/drivers
lrwxrwxrwx  1 root root      34 Aug  1 05:04 firmware -> ../linux-headers-5.0.0-25/firmware
lrwxrwxrwx  1 root root      28 Aug  1 05:04 fs -> ../linux-headers-5.0.0-25/fs
drwxr-xr-x  4 root root    4096 Aug 14 06:00 include
lrwxrwxrwx  1 root root      30 Aug  1 05:04 init -> ../linux-headers-5.0.0-25/init
lrwxrwxrwx  1 root root      29 Aug  1 05:04 ipc -> ../linux-headers-5.0.0-25/ipc
lrwxrwxrwx  1 root root      32 Aug  1 05:04 Kbuild -> ../linux-headers-5.0.0-25/Kbuild
lrwxrwxrwx  1 root root      33 Aug  1 05:04 Kconfig -> ../linux-headers-5.0.0-25/Kconfig
drwxr-xr-x  2 root root    4096 Aug 14 06:00 kernel
lrwxrwxrwx  1 root root      29 Aug  1 05:04 lib -> ../linux-headers-5.0.0-25/lib
lrwxrwxrwx  1 root root      34 Aug  1 05:04 Makefile -> ../linux-headers-5.0.0-25/Makefile
-rw-r--r--  1 root root    1021 Aug  1 05:04 .missing-syscalls.d
lrwxrwxrwx  1 root root      28 Aug  1 05:04 mm -> ../linux-headers-5.0.0-25/mm
-rw-r--r--  1 root root 1564153 Aug  1 05:04 Module.symvers
lrwxrwxrwx  1 root root      29 Aug  1 05:04 net -> ../linux-headers-5.0.0-25/net
lrwxrwxrwx  1 root root      33 Aug  1 05:04 samples -> ../linux-headers-5.0.0-25/samples
drwxr-xr-x  7 root root   12288 Aug 14 06:00 scripts
lrwxrwxrwx  1 root root      34 Aug  1 05:04 security -> ../linux-headers-5.0.0-25/security
lrwxrwxrwx  1 root root      31 Aug  1 05:04 sound -> ../linux-headers-5.0.0-25/sound
drwxr-xr-x  2 root root    4096 Aug  1 05:04 .tmp_versions
drwxr-xr-x  3 root root    4096 Aug 14 06:00 tools
lrwxrwxrwx  1 root root      32 Aug  1 05:04 ubuntu -> ../linux-headers-5.0.0-25/ubuntu
lrwxrwxrwx  1 root root      29 Aug  1 05:04 usr -> ../linux-headers-5.0.0-25/usr
lrwxrwxrwx  1 root root      30 Aug  1 05:04 virt -> ../linux-headers-5.0.0-25/virt
```

```
polerin@dev-systems:~/temp/v4l2loopback-master$ ls -la /usr/src/linux-headers-5.0.0-25-generic/crypto/
asymmetric_keys/ async_tx/        Kconfig          Makefile         
polerin@dev-systems:~/temp/v4l2loopback-master$ ls -la /usr/src/linux-headers-5.0.0-25-generic/crypto/
total 80
drwxr-xr-x  4 root root  4096 Aug 14 06:00 .
drwxr-xr-x 25 root root  4096 Aug 14 06:00 ..
drwxr-xr-x  2 root root  4096 Aug 14 06:00 asymmetric_keys
drwxr-xr-x  2 root root  4096 Aug 14 06:00 async_tx
-rw-r--r--  1 root root 55695 Mar  3 17:21 Kconfig
-rw-r--r--  1 root root  6260 Mar  3 17:21 Makefile
```

#### Expected Results:

complete make install without error

umlaeute, 2019-08-21:
to quote the [README](/umlaeute/v4l2loopback#install)

>  to install the module run "make install" (you might have to be 'root' to have all necessary permissions to install the module).

closed 2019-08-21 by umlaeute

polerin, 2019-08-21:
I forgot to sudo the example I pasted in, but I had done it on previous examples.  Log here 

```
[sudo] password for polerin: 
make -C /lib/modules/`uname -r`/build M=/home/polerin/temp/v4l2loopback-master modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-25-generic'
  INSTALL /home/polerin/temp/v4l2loopback-master/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.0.0-25-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-25-generic'
```

If you note, it's not failing on permission, it's failing on lack of a crypto component, and I can't find any documentation from either ubuntu or here about what might be the appropriate fix.

umlaeute, 2019-08-21:
>[sudo] password for polerin: 
>
>If you note, it's not failing on permission, it's failing on lack of a
>crypto component, and I can't find any documentation from either ubuntu
>or here about what might be the appropriate fix.

i note that it is not failing at all (and that you didn't check other - closed - issues in this tracker, else you would have noted that others have had this "problem"  before and that there really isn't a problem.