v4l2loopback/v4l2loopback #235: ondemandcam.c not waiting for frames to be consumed

allsey87, 2019-10-11:
### System configuration

version: v4l2loopback driver version 0.12.2 loaded
kernel version: Linux ToshibaLaptop 5.2.11-arch1-1-ARCH #1 SMP PREEMPT Thu Aug 29 08:09:36 UTC 2019 x86_64 GNU/Linux
  
### Problem
The example [ondemandcam.c](https://github.com/umlaeute/v4l2loopback/blob/v0.12.2/examples/ondemandcam.c) is producing frames as fast as possible and does not wait for the other end of the V4L2 pipe to be opened.

#### Steps to reproduce:

  1. modprobe the driver
  2. add a `fprintf(stderr, ".");` to `sendvid` in [ondemandcam.c](https://github.com/umlaeute/v4l2loopback/blob/v0.12.2/examples/ondemandcam.c)
  3. compile and link [ondemandcam.c](https://github.com/umlaeute/v4l2loopback/blob/v0.12.2/examples/ondemandcam.c) against pthread and run it
  
#### Observed Results:

The ondemandcam is printing dots to the screen at a rapid rate even though the device is not open for reading. Opening the device in ffplay shows that the code is working (cycling through 4 colors, changing every second), but reports an error when trying to set the framerate:

> [video4linux2,v4l2 @ 0x7f61b0000b80] The driver does not permit changing the time per frame
  
#### Expected Results:

1. Frames should not be produced unless a program has opened /dev/videoX for reading.
2. `ffplay` should be able to control the framerate and consume X FPS

copiltembel, 2020-11-03:
More than one year later, I'm having the same problem.

@allsey87 Did you manage to fix this?

copiltembel, 2020-11-03:
This works and it is a more elegant solution: https://github.com/umlaeute/v4l2loopback/issues/109#issuecomment-617638198