v4l2loopback/v4l2loopback #237: does not work since the last update of kernel

corinnespi, 2019-10-14:
**Environment**

    v4l2loopback version: 0.12.1 (i suppose)

    kernel version: Linux corinnes-N2x0WU 4.15.0-65-generic #74-Ubuntu SMP Tue Sep 17 17:06:04 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

    Distribution (+version): Ubuntu 18.04

 **the problem:**

v4l2loopback does not create a new virtual cam when I do
sudo modprobe v4l2loopback
There is only my normal webcam that is listed when I do v4l2-ctl --list-devices
Previously it was working.

**Expected :** 
Create a virtual cam to be able tu use is with obs studio and zoom !

umlaeute, 2019-10-14:
does it create a new `/dev/video*` device (your webcam will most likely have `/dev/video0` assigned, so *v4l2loopback* should grab the next available one, which ought to be  `/dev/video1`)

If so, do you have a producer running that feeds the device with data?

what do you mean with " Previously it was working."? which version of v4l2loopback? which kernel-version?

also, to get the version of the loaded v4l2loopback module, do something like `dmesg | grep v4l2loopback`

corinnespi, 2019-10-14:
No, nothing, there is only the device wich is my real webcam

Previosly, it could see when I did v4l2-ctl --list-devices that there was a device, and over all, when I was on zoom, i could choose this other device. 

When you mean a producer, do you mean for example OBS-studio ?

I used obs-studio, to have my video, with a windows
[image]
And then, I opened zoom and used this virtual webcam, to give for example this image. 

I hope it is clear.

corinnespi, 2019-10-14:
When I do only dmesg : 
corinnes@corinnes-N2x0WU:~$ dmesg
[    0.000000] microcode: microcode updated early to revision 0xb4, date = 2019-04-01
[    0.000000] Linux version 4.15.0-65-generic (buildd@lgw01-amd64-006) (gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)) #74-Ubuntu SMP Tue Sep 17 17:06:04 UTC 2019 (Ubuntu 4.15.0-65.74-generic 4.15.18)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-65-generic root=UUID=1f7e1b84-86f1-4d87-9de7-26eea70cbcf2 ro quiet splash acpi_os_name=Linux acpi_osi= vt.handoff=1
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000077cf6fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000077cf7000-0x0000000077cf7fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000077cf8000-0x0000000077cf8fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000077cf9000-0x00000000891b1fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000891b2000-0x000000008a01cfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008a01d000-0x000000008a65ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000008a660000-0x000000008a6eafff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008a6eb000-0x000000008b32bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008b32c000-0x000000008b3fdfff] type 20
[    0.000000] BIOS-e820: [mem 0x000000008b3fe000-0x000000008b3fefff] usable
[    0.000000] BIOS-e820: [mem 0x000000008b3ff000-0x000000008fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000046effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.60 by American Megatrends
[    0.000000] efi:  ACPI 2.0=0x8a660000  ACPI=0x8a660000  SMBIOS=0x8b28b000  SMBIOS 3.0=0x8b28a000  ESRT=0x86560418  MEMATTR=0x85e01018 
[    0.000000] secureboot: Secure boot could not be determined (mode 0)
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: TUXEDO N2x0WU                         /N2x0WU                         , BIOS 5.12 11/01/2017
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x46f000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: write-back
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 00C0000000 mask 7FC0000000 uncachable
[    0.000000]   1 base 00A0000000 mask 7FE0000000 uncachable
[    0.000000]   2 base 0090000000 mask 7FF0000000 uncachable
[    0.000000]   3 base 008C000000 mask 7FFC000000 uncachable
[    0.000000]   4 base 008B800000 mask 7FFF800000 uncachable
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] e820: last_pfn = 0x8b3ff max_arch_pfn = 0x400000000
[    0.000000] esrt: Reserving ESRT space from 0x0000000086560418 to 0x0000000086560450.
[    0.000000] Scanning 1 areas for low memory corruption
[... 1133 lines cut ...]
[20370.962068] usb 1-1.2: Product: Microsoft® Digital Media Keyboard 3000
[20370.962069] usb 1-1.2: Manufacturer: Microsoft
[20370.970787] input: Microsoft Microsoft® Digital Media Keyboard 3000 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:045E:0730.0004/input/input40
[20371.028663] microsoft 0003:045E:0730.0004: input,hidraw0: USB HID v1.11 Keyboard [Microsoft Microsoft® Digital Media Keyboard 3000] on usb-0000:00:14.0-1.2/input0
[20371.042478] input: Microsoft Microsoft® Digital Media Keyboard 3000 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:045E:0730.0005/input/input41
[... cut ...]
[20371.579880] usb 1-1.1: new high-speed USB device number 12 using xhci_hcd
[20371.680571] usb 1-1.1: New USB device found, idVendor=0e8d, idProduct=2008
[20371.680574] usb 1-1.1: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[20371.680577] usb 1-1.1: Product: U FEEL LITE
[20371.680578] usb 1-1.1: Manufacturer: WIKO
[20371.680580] usb 1-1.1: SerialNumber: ZP8HMFYP8HAMLJC6
[21586.151113] input: 8C:DE:52:AF:50:A8 as /devices/virtual/input/input43
[22026.213197] usb 1-1.1: USB disconnect, device number 12

And the second part  grep v4l2loopback
It is 
[image]

umlaeute, 2019-10-14:
you should run the following command (all in one line), that filters the output of `dmesg`, only displaying lines that contain `v4l2loopback`

~~~sh
dmesg | grep v4l2loopback
~~~

But from your unfiltered output, it seems that the module doesn't load at all.
what's the output of `sudo modprobe v4l2loopback`?

if you get errors, try running `sudo depmod` after installing the module.

corinnespi, 2019-10-14:
I did sudo depmod ands then sudo modprobe v4l2loopback
and it gave me 
modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/4.15.0-65-generic

umlaeute, 2019-10-14:
so you didn't run `make install`.

you can _test_  without installing by running the following
```sh
make modprobe
```

but to deploy, you should run

```sh
make install
```

closed 2019-10-14 by umlaeute