v4l2loopback/v4l2loopback #238: following does not work since the last update of kernel (#237)

corinnespi, 2019-10-15:
see #237 
I did previously make install and it was working until the new kernel. 
I redo it and it gave that : 
/v4l2loopback$ sudo make install
[sudo] Mot de passe de corinnes : 
make -C /lib/modules/`uname -r`/build M=/home/corinnes/v4l2loopback modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-65-generic »
  INSTALL /home/corinnes/v4l2loopback/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-65-generic
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-65-generic »

umlaeute, 2019-10-15:
- please use a meaningful title ("does not work" is not meaningful, neither is "found a bug"; "following" is meaningless when you look at the *list* of issues)

anyhow:

I am not sure whether the error you see is really an error. there have been reports of similar "issues" (e.g. #230) and **all of them** turned out to be some intermediate glitches that resolved to *warnings* in the end.

If you indeed experience an error, the last line of the output you receive from `sudo make install` should be something like that:

~~~sh
make: *** [Makefile:47: install] Error 2
~~~

If the last line does not indicate an error, most likely there is none. and you succeeded to `install` the module.
So run `modprobe` again and see whether you can load the module.

corinnespi, 2019-10-15:
After : modprobe v4l2loopback in terminal
modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/4.15.0-65-generic

umlaeute, 2019-10-15:
*sigh*, you know that `v4l2loopback` is pretty redundant in this title?
why don't you just use something like "module not found after `make install`"?
don't you think that this classifies your problem more neatly?

anyhow, please run:

~~~
sudo make install V=1; echo $?
~~~

and paste the full output you get, including the command you run (at the beginning) and the cmdline-prompt (at the end).

corinnespi, 2019-10-15:
~$ cd v4l2loopback
~/v4l2loopback$ sudo make install V=1; echo $?
make -C /lib/modules/`uname -r`/build M=/home/corinnes/v4l2loopback modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-65-generic »
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (	\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /lib/modules/4.15.0-65-generic/extra
make -f ./scripts/Makefile.modinst
  mkdir -p /lib/modules/4.15.0-65-generic/extra ; cp /home/corinnes/v4l2loopback/v4l2loopback.ko /lib/modules/4.15.0-65-generic/extra ; true /lib/modules/4.15.0-65-generic/extra/v4l2loopback.ko ; if (echo "/lib/modules/4.15.0-65-generic/extra/v4l2loopback.ko" | egrep -q "\/drivers\/staging\/") && [ -f ./drivers/staging/signature-inclusion ] && (! egrep -x "v4l2loopback.ko" ./drivers/staging/signature-inclusion) ; then echo Not signing "/lib/modules/4.15.0-65-generic/extra/v4l2loopback.ko"; else scripts/sign-file "sha512" "certs/signing_key.pem" certs/signing_key.x509 /lib/modules/4.15.0-65-generic/extra/v4l2loopback.ko || true && true /lib/modules/4.15.0-65-generic/extra/v4l2loopback.ko; fi
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  /bin/bash ./scripts/depmod.sh /sbin/depmod 4.15.0-65-generic ""
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-65-generic »
0

corinnespi, 2019-10-15:
PS : Sorry about name and so on, I am only a normal human ! A Muggle !

umlaeute, 2019-10-15:
so according to the output, both the compilation and the installation succeed and the module is copied to `/lib/modules/4.15.0-65-generic/extra/`
could you confirm that the `v4l2loopback.ko` file exists inside this directory?

*then* run `sudo depmod -a` and see what it reports.

then run `sudo modprobe v4l2loopback`

if it still cannot find the module, try running `sudo strace -e trace=%file modprobe v4l2loopback` (after installing the *strace* package), and post the output of that command.

corinnespi, 2019-10-15:
/home/corinnes/v4l2loopback/v4l2loopback.ko
when i do sudo depmod -a, nothing happens ! 
[image]

Trace : 
sudo strace -e trace=%file modprobe v4l2loopback
execve("/sbin/modprobe", ["modprobe", "v4l2loopback"], 0x7ffec243d728 /* 27 vars */) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
stat("/etc/modprobe.d", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
openat(AT_FDCWD, "/etc/modprobe.d", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
newfstatat(3, "blacklist-modem.conf", {st_mode=S_IFREG|0644, st_size=156, ...}, 0) = 0
newfstatat(3, "blacklist-oss.conf", {st_mode=S_IFREG|0644, st_size=1059, ...}, 0) = 0
newfstatat(3, "blacklist-rare-network.conf", {st_mode=S_IFREG|0644, st_size=583, ...}, 0) = 0
newfstatat(3, "blacklist.conf", {st_mode=S_IFREG|0644, st_size=1667, ...}, 0) = 0
newfstatat(3, "intel-microcode-blacklist.conf", {st_mode=S_IFREG|0644, st_size=154, ...}, 0) = 0
newfstatat(3, "blacklist-framebuffer.conf", {st_mode=S_IFREG|0644, st_size=697, ...}, 0) = 0
newfstatat(3, "blacklist-ath_pci.conf", {st_mode=S_IFREG|0644, st_size=325, ...}, 0) = 0
newfstatat(3, "alsa-base.conf", {st_mode=S_IFREG|0644, st_size=2507, ...}, 0) = 0
newfstatat(3, "dkms.conf", {st_mode=S_IFREG|0644, st_size=127, ...}, 0) = 0
newfstatat(3, "blacklist-firewire.conf", {st_mode=S_IFREG|0644, st_size=210, ...}, 0) = 0
newfstatat(3, "iwlwifi.conf", {st_mode=S_IFREG|0644, st_size=347, ...}, 0) = 0
newfstatat(3, "amd64-microcode-blacklist.conf", {st_mode=S_IFREG|0644, st_size=154, ...}, 0) = 0
stat("/run/modprobe.d", 0x7ffca67cfe30) = -1 ENOENT (No such file or directory)
stat("/lib/modprobe.d", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
openat(AT_FDCWD, "/lib/modprobe.d", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
newfstatat(3, "fbdev-blacklist.conf", {st_mode=S_IFREG|0644, st_size=390, ...}, 0) = 0
newfstatat(3, "aliases.conf", {st_mode=S_IFREG|0644, st_size=655, ...}, 0) = 0
newfstatat(3, "blacklist_linux_4.15.0-64-generic.conf", {st_mode=S_IFREG|0644, st_size=1461, ...}, 0) = 0
newfstatat(3, "systemd.conf", {st_mode=S_IFREG|0644, st_size=765, ...}, 0) = 0
newfstatat(3, "blacklist_linux_4.15.0-65-generic.conf", {st_mode=S_IFREG|0644, st_size=1461, ...}, 0) = 0
openat(AT_FDCWD, "/lib/modprobe.d/aliases.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/alsa-base.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/amd64-microcode-blacklist.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/blacklist-ath_pci.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/blacklist-firewire.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/blacklist-framebuffer.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/blacklist-modem.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/blacklist-oss.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/blacklist-rare-network.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/blacklist.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/modprobe.d/blacklist_linux_4.15.0-64-generic.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/modprobe.d/blacklist_linux_4.15.0-65-generic.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/dkms.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/modprobe.d/fbdev-blacklist.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/intel-microcode-blacklist.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/etc/modprobe.d/iwlwifi.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/modules/4.15.0-65-generic/modules.softdep", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/modprobe.d/systemd.conf", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/proc/cmdline", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/modules/4.15.0-65-generic/modules.dep.bin", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/modules/4.15.0-65-generic/modules.alias.bin", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/modules/4.15.0-65-generic/modules.symbols.bin", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/modules/4.15.0-65-generic/modules.builtin.bin", O_RDONLY|O_CLOEXEC) = 3
stat("/lib/modules/4.15.0-65-generic/kernel/drivers/media/v4l2-core/videodev.ko", {st_mode=S_IFREG|0644, st_size=291713, ...}) = 0
stat("/lib/modules/4.15.0-65-generic/kernel/drivers/media/media.ko", {st_mode=S_IFREG|0644, st_size=54793, ...}) = 0
openat(AT_FDCWD, "/sys/module/v4l2loopback/initstate", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/sys/module/v4l2loopback", 0x7ffca67cfdb0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/sys/module/media/initstate", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/sys/module/videodev/initstate", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/sys/module/v4l2loopback/initstate", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/sys/module/v4l2loopback", 0x7ffca67cfdb0) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/modules/4.15.0-65-generic/extra/v4l2loopback.ko", O_RDONLY|O_CLOEXEC) = 3
modprobe: ERROR: could not insert 'v4l2loopback': Exec format error
+++ exited with 1 +++

umlaeute, 2019-10-15:
> so according to the output, both the compilation and the installation succeed and the module is copied to `/lib/modules/4.15.0-65-generic/extra/`
> could you confirm that the `v4l2loopback.ko` file exists inside this directory?

with *this* directory i really meant the `/lib/modules/4.15.0-65-generic/extra/` directory.
(`modprobe` will not search in `/home/corinnes/v4l2loopback/`, so this doesn't matter).

anyhow, your trace clearly shows that suddenly the module can be found (see the third-last line in the trace, the one containing a successfull `openat`).
You will also notice that the error message changed from `FATAL: Module v4l2loopback not found` to `ERROR: Exec format error`.

That might have multiple reasons, but I'd like to stress that all of these reasons are due to your system setup, and not related to (or fixable by) `v4l2loopback`.

The obvious fix is to make sure that you properly re-compile:

~~~sh
make clean
make
sudo make install
sudo modprobe v4l2loopback
~~~

make sure that `make clean` succeeds first (that is: it can remove any build-leftovers from previous build attempts), before you run the rest of the commands.
If that doesn't help either, then I cannot help you anymore, and you should try to find out what happened to your system.

umlaeute, 2019-10-16:
so did that fix the issue?

corinnespi, 2019-10-16:
I was not at office today, and tomorrow it will be the same. I will look at that friday, as I don't want to do it quickly, I have to concentrate. Thanks for asking. If it is ok, I will inform you of course.

corinnespi, 2019-10-25:
I did "make clean", I have "make: *** Aucune règle pour fabriquer la cible « clean ». Arrêt.
" which means no rules to make a clean !

umlaeute, 2019-10-25:
*where* did you run `make clean`?

corinnespi, 2019-10-25:
On the terminal.
after that I used also a cleanup software from Ubuntu. Is that enough?
I am not on my computer now so I cannot tel you the exact name of the software.

umlaeute, 2019-10-25:
nah. i meant "in which directory".

if you have no rule for " clean", then i think you are running that command in the wrong directory.

corinnespi, 2019-10-25:
OK, I did not understand I have to be in the directory from v4l2loopback
Make clean : 
rm -f *~
rm -f Module.symvers Module.markers modules.order
make -C /lib/modules/`uname -r`/build M=/home/corinnes/v4l2loopback clean
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-66-generic »
  CLEAN   /home/corinnes/v4l2loopback/.tmp_versions
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-66-generic »
rm -f *~
rm -f Module.symvers Module.markers modules.order
make -C /lib/modules/`uname -r`/build M=/home/corinnes/v4l2loopback clean
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-66-generic »
  CLEAN   /home/corinnes/v4l2loopback/.tmp_versions
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-66-generic »

Make : 
make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/corinnes/v4l2loopback modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-66-generic »
  CC [M]  /home/corinnes/v4l2loopback/v4l2loopback.o
symbolmap: la: invalid section
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/corinnes/v4l2loopback/v4l2loopback.mod.o
  LD [M]  /home/corinnes/v4l2loopback/v4l2loopback.ko
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-66-generic »

sudo make install
[sudo] Mot de passe de corinnes : 
make -C /lib/modules/`uname -r`/build M=/home/corinnes/v4l2loopback modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-66-generic »
  INSTALL /home/corinnes/v4l2loopback/v4l2loopback.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-66-generic
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-66-generic »

sudo modprobe v4l2loopback
modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/4.15.0-66-generic

And it still don't work with obs and no virtual cam : 
v4l2-ctl --list-devices
Chicony USB2.0 Camera: Chicony  (usb-0000:00:14.0-6):
	/dev/video0
:(

umlaeute, 2019-10-28:
i'm about 99% sure that this problem is not related to `v4l2loopback` itself, but rather to your specific system setup (or PBKAC). i'm aware that this assessment doesn't help you much (but I also don't want to keep spending my time on a problem which i feel is not mine). so...

since you are using ubuntu, why don't you just install the `v4l2loopback` package that comes with your distribution:

~~~
sudo apt-get install v4l2loopback-dkms
~~~

this should install all the relevant packages for building kernel modules, and build and install them.
(although Ubuntu/bionic comes with `v4l2loopback-0.10.0`, so you are not getting the latest stuff)

if the above still fails to produce usable kernel modules, try installing another dkms-package (e.g. `acpi-call-dkms`, or ) and see whether this produces a usable module (e.g. `modprobe acpi_call` suceeds) - if it fails, you probably should seek advice in some general ubuntu forum.

if everything works and *only* `v4l2loopback` fails to build, come back and we might continue to hunt down the issue (no need to open a new issue then, just keep posting to this one and I might re-open it)

closed 2019-10-28 by umlaeute

BellemareMederic, 2020-11-24:
sudo depmod -a

> so according to the output, both the compilation and the installation succeed and the module is copied to `/lib/modules/4.15.0-65-generic/extra/`
> could you confirm that the `v4l2loopback.ko` file [...]

sudo depmod -a 
save my installation thanks