v4l2loopback/v4l2loopback #241: ffmpeg problems with v4l2loopback version 0.10.0

givo, 2019-10-28:
* `v4l2loopback`: 0.10.0-1ubuntu1.1

  * `kernel`: 5.0.0-32-generic

  * `Distribution`: Ubuntu 18.04.3 LTS

I am using `v4l2loopback` for streaming a RTSP camera as a virtual camera and it was working great with `ffmpeg` on a pc with kernel version 4.15.0-15 and `v4l2loopback` version 0.9.1. But on a  a pc with kernel version `5.0.0-32-generic` and `v4l2loopback` version 0.10.0 ffmpeg throws the following error:

```
[video4linux2,v4l2 @ 0x5645108c4700] ioctl(VIDIOC_G_FMT): Invalid argument
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 -- 
```

#### Steps to reproduce:

  1.`sudo apt-get install v4l2loopback`
  2. `sudo modprobe v4l2loopback video_nr=0 card_label="camera"`
  3. 
 ```
ffmpeg -y -rtsp_transport tcp -i rtsp://admin:Orca123456@192.168.1.32/Streaming/Channels/1/ -pix_fmt yuyv422 -f v4l2 /dev/video2
```
  
#### Observed Results:

```
[video4linux2,v4l2 @ 0x5645108c4700] ioctl(VIDIOC_G_FMT): Invalid argument
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 -- 
```
  
#### Expected Results:

  * No errors from ffmpeg

Maybe there a bug on the newest version of v4l2loopback (0.10.0)?

umlaeute, 2019-10-28:
the newest version of `v4l2loopback` is - as of today -  [0.12.2](https://github.com/umlaeute/v4l2loopback/releases).

please check whether your problem persists with this newer version.

(also note, that there is no `v4l2loopback` package in Debian/Ubuntu/... and the like)

givo, 2019-10-28:
Ok, installing through source fixed the problem.

Are you planning on updating the package on the official registry?

umlaeute, 2019-10-28:
registry? i think you are confusing things here.

in any case, newer releases of ubuntu come with [newer versions of v4l2loopback](https://packages.ubuntu.com/search?keywords=v4l2loopback-dkms&searchon=names&suite=all&section=all)

closed 2020-02-13 by umlaeute