v4l2loopback/v4l2loopback #245: Multiple loopbacks from single source on Raspberry Pi

danriches, 2019-11-19:
Hi,

I have a v4l2 application that works very well however I would like to use it alongside an rtsp streaming application. My question is how do I go about configuring your software to copy or clone video from /dev/video0 to /dev/video1 and /dev/video2? The applications will also need to issue v4l2-ctl commands to set the HDMI input timing and EDID data (first application). I'm running this on a Raspberry Pi using my own equivalent of the B101 board from Auvidea but this shouldn't be any cause for concern as it's just a V4L2 output.

Kind Regards,

Dan

danriches, 2019-11-19:
I can;t seem to compile this on an RPi 4B (2Gb) running latest version of Raspbian Buster:

pi@rpi4b2g:~/v4l2loopback $ make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/pi/v4l2loopback modules
make[1]: *** /lib/modules/4.19.81-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:43: v4l2loopback.ko] Error 2

Any Idea's anyone????

Thanks in advance, Dan

danriches, 2019-11-20:
I ended up running the commands under package install and it worked. I did everything there was to do on the manual source build and install but no, at least it's loaded the kernel module with 2 output devices as I wanted.

Thank you and sorry for the stupid questions, I should've tried harder first off ;)

Dan

closed 2019-11-20 by danriches

mysunnytime, 2020-09-14:
> I ended up running the commands under package install and it worked. I did everything there was to do on the manual source build and install but no, at least it's loaded the kernel module with 2 [...]

Hi! I encountered the same error. May I ask how did you install it with "command under package install"? I don't find it in readme.. Thanks!