v4l2loopback/v4l2loopback #246: Cannot use 2nd video loopback device

danriches, 2019-11-25:
Hi,

I have a raspberry pi 3B+ running latest Raspbian, when I run the following commands all works as expected:
```
v4l2-ctl -d /dev/video0 -s pal
sleep 1
sudo modprobe v4l2loopback devices=2
sleep 2
sudo v4l2compress_h264 /dev/video0 /dev/video2 &
sleep 2
sudo v4l2rtspserver -S /dev/video2,default &
```

However I can't use 6by9's (github.com/6by9/v4l2_mmal) example on the /dev/video1 device. All I get is this:

```
pi@raspberrypi:~/CaptureImage $ sudo ./v4l2_mmal --capture=90 -n 3 --encode-to=/home/pi/test.jpg /dev/video1
We're encoding to /home/pi/test.jpg
Device /dev/video1 opened.
Device `Dummy video device (0x0000)' on `platform:v4l2loopback-000' (driver 'v4l2 loopback') is a video capture (without mplanes) device.
Unable to get format: Invalid argument (22).
Current frame rate: 30/1
Unable to get format: Invalid argument (22).
3 buffers requested.
length: 0 offset: 0 timestamp type/source: unk/EoF
length: 0 offset: 0 timestamp type/source: unk/EoF
length: 0 offset: 0 timestamp type/source: unk/EoF
Unable to start streaming: Input/output error (5).
Closing dma_buf 0
Closing dma_buf 0
Closing dma_buf 0
3 buffers released.

```

What do I need to do to set the capabilities up correctly, preferably copied from /dev/video0 or do I need to feed the loopback device somehow. Sorry for the possibly daft questions as I'm struggling to get this last part working correctly...

Thanks in advance!!

Dan

danriches, 2019-11-25:
What I'm trying to do is create 2 clones of the /dev/video0 video stream (it's either 720x576 or 1920x1080). Then stream the /dev/video2 after running the H264 compressor on the feed and rtsp stream it out. This part works great, however when I try to use my existing application based upon github.com/6by9/v4l2_mmal (which does work on it's own by the way!) I get a video source format error, basically the format is also set to H264 and despite the source video0 being 720x576 I get a source video of 1920x1080 on video2.

I just can't seem to get both working at once, please please can someone point me in the right direction!?!?

Kind Regards,

Dan

danriches, 2019-11-25:
So if I try to use the v4l2copy application to copy /dev/video0 to /dev/video1 after running the 4 lines at the top of this thread it fails with:

```
pi@raspberrypi:~/CaptureImage $ sudo v4l2copy /dev/video0 /dev/video1
log level:500

[NOTICE] src/V4l2Device.cpp:124
	driver:unicam capabilities:85200001 mandatory:4000001
[NOTICE] src/V4l2Device.cpp:127
	/dev/video0 support capture
[NOTICE] src/V4l2Device.cpp:129
	/dev/video0 support read/write
[NOTICE] src/V4l2Device.cpp:130
	/dev/video0 support streaming
[NOTICE] src/V4l2Device.cpp:62
	/dev/video0:UYVY size:720x576 bufferSize:829440
[ERROR] src/V4l2Device.cpp:188
	Cannot set format:UYVY for device:/dev/video0 Device or resource busy
[ERROR] src/V4l2Device.cpp:74
	Cannot init device:/dev/video0
[NOTICE] src/V4l2MmapDevice.cpp:141
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
	Device /dev/video0
[WARN] src/v4l2copy.cpp:91
	Cannot create V4L2 capture interface for device:/dev/video0
```

So, I'm still none the wiser. 

Please can someone help me???

Kind Regards,

Dan

umlaeute, 2020-02-13:
the last error seems to indicate that `/dev/video0` is busy.
from your description, i figure that `/dev/video0` is a *real* device, and only `/dev/video1` and `/dev/video2` are loopback devices.
so the problem seems to be somewhere else.

also notice, that recent kernels might create two devices for a single (hardware) device.

raffaeler, 2021-07-01:
> also notice, that recent kernels might create two devices for a single (hardware) device.

@umlaeute could you please elaborate this?

umlaeute, 2021-07-01:
some uvc-cameras (like the built in camera of my laptop) use a secondary device (`/dev/video1` in addition to `/dev/video0`) for some meta-information.
this secondary device cannot be used to capture video from.

raffaeler, 2021-07-01:
Thank you @umlaeute, got it. I also have seen similar duplication behavior for other devices.

umlaeute, 2023-05-26:
it's been a while, so:
- does this issue still exist with current `head`?
- if so, is the issue specific to `v4l2_mmal`?

(I might be totally confused about what this bug-report is actually about: the top-post only mentions `v4l2_mmal`, but it seems the top-post does not include any reference to a `v4l2loopback` device at all. it uses an obscure `v4l2compress_h264` that nobody knows where it comes from (well, at least: *I* do not know what this is; and if I am supposed to fix the problem it would be great if we could reproduce the problem and rule out problems with 3rd party applications). the first hard reference to a loopback device is in https://github.com/umlaeute/v4l2loopback/issues/246#issuecomment-558170389)

it would be fantastic if you could clarify your problem (otherwise I'm going to close it as non-actionable)

umlaeute, 2023-05-26:
> So if I try to use the v4l2copy application to copy /dev/video0 to /dev/video1 after running the 4 lines at the top of this thread it fails with:
> ```
> $ sudo v4l2copy /dev/video0 /dev/video1
> [...]
> 	Cannot set format:UYVY for device:/dev/video0 Device or resource busy
> ```

and the "4 lines at the top if this thread" include:
> ```
> sudo v4l2compress_h264 /dev/video0 /dev/video2 &
> ```

which indicates that you are simply trying to capture from `/dev/video0` with two applications simultaneously.
this is forbidden per V4L2 specifications (apart from that: the error is with the real device, rather than the loopback device)

in any case, I'm closing this now (and will happily re-open it if things clarify)

closed 2023-05-26 by umlaeute

igorhoop, 2023-12-20:
Hi @umlaeute . 

I use **v4l2loopback** on my laptop and and your program worked fine.
But now I'm trying to run it on **embedded** computers and it doesn't work. For example, now I am trying to run on NVidia Jetson. I connected a USB webcam:

`sudo modprobe v4l2loopback video_nr=99`

Now l have 2 devices:

```
ls /dev/video*

video0      # real device
video99    # dummy device
```

Next, I direct the real stream to the virtual stream using **ffmpeg**:
`ffmpeg -f v4l2 -i /dev/video0 -f v4l2 /dev/video99`

Next, I run **VLC** and view the image from **video99**. At this step everything works.
Now I want to run a copy of VLC and view **video99** again. That is, at the same time 2 VLCs must show the image from **video99**

But I get a error:
`v4l2 stream error: **cannot start streaming: Device or resources busy**`

If I turn off the first VLC instance, then the second VLC instance starts working normally.

I don't understand why this is happening. Perhaps I did something wrong, but it worked on my laptop.

Information that is installed on NVidia Jetson:
```
OS: Ubuntu 18.04
Kernel: 4.9.140-tegra
```