v4l2loopback/v4l2loopback #247: ioctl(VIDIOC_G_FMT): invalid argument ,could not write header for output file (incorrect codec parameters ) invalid argument

jas-ts, 2019-11-27:
i am trying to convert landscape mode to portrait mode for my usb camera on ubuntu 18.04, i have kernal version 5.0.0-39- , i am getting above error whenever i run 

ffmpeg -f v4l2 -i /dev/video0 -vf transpose=1 -f v4l2 /dev/video4

can anybody help what i am doing wrong here

GonzaloSP, 2020-01-30:
I am having the same issue.
Where you able to solve it?

hugomatic, 2020-03-06:
hashtag me too!

pokebox, 2020-03-16:
I encountered the same problem on UBUNTU 18.04 LTS, but it seems to only appear on x64 platform, install` v4l2loopback-dkms `on Ubuntu 19.10 on Raspberry Pi 4, and then use ffmepg to work fine. I had suspected a problem with ffmpeg, because the default apt installed version is 3.4.6, and the Raspberry Pi is 4.1.4. I uninstalled ffmpeg installed by apt and installed it using snap. Now ffmpeg on x64 platform is also the 4.1.4 version, but it doesn't work either.
The Ubuntu 18.04 kernel version on the x64 platform is `5.3.0-40-generic # 32 ~ 18.04.1-Ubuntu`, the Ubuntu 19.10 kernel version on the Raspberry Pi 4 is `5.3.00-1018-raspi2 # 20-Ubuntu`.
In addition, I also tried to compile and install the code of v4l2loopback from git on x64 platform, but it didn't work.

pokebox, 2020-03-16:
I tested UBUNTU 19.10 on the x64 platform and found that v4l2loopback works normally. The installed version is the same as that of the Raspberry Pi's Ubuntu, both of which are 0.12.1. I tried to install the v4l2loopback-dkms_0.12.1-1_all.deb package of Ubuntu 19.10 on Ubuntu 18.04LTS separately. The error is reported. The specific compilation information is as follows:

`DKMS make.log for v4l2loopback-0.12.1 for kernel 5.3.0-40-generic (x86_64)
2020年 03月 16日 星期一 16:50:22 CST
Building v4l2-loopback driver...
make -C /lib/modules/5.3.0-40-generic/build M=/var/lib/dkms/v4l2loopback/0.12.1/build modules
make[1]: 进入目录“/usr/src/linux-headers-5.3.0-40-generic”
  CC [M]  /var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.c: In function ‘vidioc_qbuf’:
/var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.c:1509:4: error: implicit declaration of function ‘v4l2_get_timestamp’; did you mean ‘v4l2_get_subdevdata’? [-Werror=implicit-function-declaration]
    v4l2_get_timestamp(&b->buffer.timestamp);
    ^~~~~~~~~~~~~~~~~~
    v4l2_get_subdevdata
cc1: some warnings being treated as errors
scripts/Makefile.build:288: recipe for target '/var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.o' failed
make[2]: *** [/var/lib/dkms/v4l2loopback/0.12.1/build/v4l2loopback.o] Error 1
Makefile:1655: recipe for target '_module_/var/lib/dkms/v4l2loopback/0.12.1/build' failed
make[1]: *** [_module_/var/lib/dkms/v4l2loopback/0.12.1/build] Error 2
make[1]: 离开目录“/usr/src/linux-headers-5.3.0-40-generic”
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2`

mozai, 2020-03-18:
When I try using gst-launch-1.0 instead, I get "Device '/dev/video0' is not a output device."

umlaeute, 2020-03-18:
`/dev/video0` or `/dev/video20`?
could it be that `/dev/video0` is a real hardware-device (not created by the *v4l2loopback* module)?

mozai, 2020-03-18:
Yup.  `v4l2-ctl --list-devices` informs me /dev/video{0,1} are used by a (single) hardware device.  /dev/video2 is the "Dummy video device (0x0000)" and I can write to that third device without ioctl(VIDIOC_G_FMT)

umlaeute, 2020-03-18:
btw @mozai did you actually hijack this thread? please don't.
ithink i'm going to delete the last 4 comments (including this one)

rayitopy, 2020-03-19:
same issue here
Compiling from source in Ubuntu 18.04

rayitopy, 2020-03-20:
Started working after removing the version installed by apt then installing from source and restarting ¯\_(ツ)_/¯

sijanec, 2020-04-15:
Same issue on `Linux 5.3.0-46-generic #38~18.04.1-Ubuntu SMP Tue Mar 31 04:17:56 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`  with `Intel(R) Core(TM)2 Duo CPU     P8700  @ 2.53GHz`

pedrodiamel, 2020-04-20:
Ubuntu 18.04
Linux 5.3.0-46-generic

#remove apt package
sudo modprobe -r v4l2loopback
sudo apt remove v4l2loopback-dkms

#install aux
sudo apt-get install linux-generic
sudo apt install dkms

#install v4l2loopback from the repository
https://github.com/umlaeute/v4l2loopback.git
cd v4l2loopback
make 

#instal mod
sudo cp -R . /usr/src/v4l2loopback-1.1
sudo dkms add -m v4l2loopback -v 1.1
sudo dkms build -m v4l2loopback -v 1.1
sudo dkms install -m v4l2loopback -v 1.1
sudo reboot

umlaeute, 2020-04-20:
@pedrodiamel this is most certainly the wrong way to inject a different version of `v4l2loopback` into dkms. a better approach is to fetch a newer version of `v4l2loopback-dkms` package, e.g.

~~~sh
wget http://deb.debian.org/debian/pool/main/v/v4l2loopback/v4l2loopback-dkms_0.12.4-1_all.deb
sudo dpkg -i v4l2loopback-dkms_0.12.4-1_all.deb
~~~

however, how does this fix the problem at hand?
if you want to say that the problem is "fixed in the current versions of the `v4l2loopback` module", why don't you just say so?

should this be closed then? @jas-ts et al...?

pittma, 2020-05-04:
I _think_ I'm seeing the same problem here...

```
gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0
```

```
ffmpeg version 4.1.4-1build2 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.2.1-4ubuntu1)
  configuration: --prefix=/usr --extra-version=1build2 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.
[mjpeg @ 0x5653f54d2ac0] Format mjpeg detected only with low score of 12, misdetection possible!
Input #0, mjpeg, from 'pipe:':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj422p(pc, bt470bg/unknown/unknown), 1056x704, 25 tbr, 1200k tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> rawvideo (native))
[swscaler @ 0x5653f5626a00] deprecated pixel format used, make sure you did set range correctly
[video4linux2,v4l2 @ 0x5653f54d87c0] ioctl(VIDIOC_G_FMT): Invalid argument
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 -- 
Conversion failed!
```

```
dpitt at p-thunk in ~ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 19.10
Release:        19.10
Codename:       eoan
dpitt at p-thunk in ~ uname -a
Linux p-thunk 5.3.0-46-generic #38-Ubuntu SMP Fri Mar 27 17:37:05 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

And I built v4l2loopback from the 0.12.5 tag.

pittma, 2020-05-04:
argh, no, I'm sorry—I had the wrong device. Disregard!

AryToNeX, 2020-05-25:
I am building this project from the master branch, and I have the same issue. What's going on here?
Also, this happens
```
$ v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
```

IvanUAH, 2020-07-30:
Did some one have solution for this issue?

sijanec, 2020-07-31:
not a solution but I can recommend you akvcam (
https://github.com/webcamoid/akvcam) for virtual camera emulation.

xB-2048, 2020-08-26:
For those who are wondering what went wrong here. You might be using a wrong /dev/video* device

`ls -l /dev/video*
crw-rw----+ 1 root video 81, 0 Aug 24 10:19 /dev/video0
crw-rw----+ 1 root video 81, 1 Aug 24 10:19 /dev/video1
crw-rw----+ 1 root video 81, 2 Aug 26 22:26 /dev/video2`

Now update your command to use the other video devices and it will work. In my case, it worked with /dev/video2

`gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video2`

xB-2048, 2020-08-26:
<pre><code>tell application "Foo"
    beep
end tell
</code></pre>

nils-wisiol, 2020-09-08:
@xB-2048's solution worked for me! Thanks!

moraesjeremias, 2020-09-22:
@xB-2048 thanks a lot for the solution! Totally worked!

BramAlkema, 2020-09-22:
Solved for a specific case. 
I had an existing v4l2loopback instance, OBS virtual Cam. Trying to modprobe another v4l2loopback device, an android IP cam stream, on top of an existing instance won't work (Duh!). (FFMpeg results in ioctl(VIDIOC_G_FMT)

Followed [how-do-i-automatically-run-modprobe-v4l2loopback-on-boot](https://askubuntu.com/questions/1245212/how-do-i-automatically-run-modprobe-v4l2loopback-on-boot) Added **both** boot time devices to /etc/modprobe.d/v4l2loopback.conf
`options v4l2loopback devices=2 video_nr=4,10 card_label="MiA3","OBS-Cam" exclusive_caps=1,1`

Checking... 
```
$v4l2-ctl --list-devices
MiA3" (platform:v4l2loopback-000):
	/dev/video4

"OBS-Cam (platform:v4l2loopback-001):
	/dev/video10
```

Runs... 
`ffmpeg -i http://192.168.2.2:8080/video -fflags nobuffer  -vf format=yuv420p  -r 30 -f v4l2 /dev/video4`
without ioctl faults.

umlaeute, 2020-09-23:
> Trying to modprobe another v4l2loopback device,

ah well, this is not possible.
having a module loaded in the kernel is a binary thing (very similar to being pregnant): either the module *is* loaded or it is *not*: you cannot *load more* of the same module (or another instance of it).
this is a fundamental difference to user-space applications (and not an issue of `v4l2loopback` itself)

BramAlkema, 2020-09-23:
@umlaeute Thanks. And yes, "Duh!" means "obviously" and "no shit sherlock". No, it's got nothing to do with v4l2loopback. 
Lesson here was, don't just copy/paste scripts without checking you already have the module loaded for another device.

mark0n, 2020-10-26:
I also ran into this issue (on Ubuntu 20.10). Turns out I my output device was already in use for another camera. Pointing to an unused output device worked smoothly. Not sure if this solves the problem of the OP but it might help others searching for a solution.

M-Abozaid, 2020-11-02:
For me the commands: 
`
modprobe v4l2loopback
ffmpeg -f x11grab -r 15 -s 1280x720 -i :0.0+0,0 -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video1
`
were not working on Fish.

They're working fine on bash, however.

peterquiel, 2020-11-11:
> For those who are wondering what went wrong here. You might be using a wrong /dev/video* device
> 
> ```
> ls -l /dev/video*
> crw-rw----+ 1 root video 81, 0 Aug 24 10:19 /dev/video0
> crw-rw----+ 1 root video 81, 1 Aug 24 10:19 /dev/video1
> crw-rw----+ 1 root video 81, 2 Aug 26 22:26 /dev/video2
> ```
> ...

Worked for me as well.. thank you so much!

umlaeute, 2020-11-11:
given there's no activity from the OP and everybody else here seems to be happy with the proposed solution, i think i can close this.

closed 2020-11-11 by umlaeute

elect86, 2021-09-08:
I'm keep running into this

When I boot my PC, then it works fine, this is my iter:
```bash
elect@5800x:~$ ps aux | grep gphoto
elect       7954  0.0  0.0 253336 16420 ?        Ssl  set07   0:00 /usr/libexec/gvfs-gphoto2-volume-monitor
elect      84406  0.3  0.0 511784 13592 ?        Sl   11:09   0:00 /usr/libexec/gvfsd-gphoto2 --spawner :1.3 /org/gtk/gvfs/exec_spaw/4
elect      84449  0.0  0.0   9500  2524 pts/1    S+   11:09   0:00 grep --color=auto gphoto
```
and I kill the first two process, and then simply:

```
elect@5800x:~$ sudo kill 7954
[sudo] password for elect: 
elect@5800x:~$ sudo kill 84406
elect@5800x:~$ sudo modprobe v4l2loopback exclusive_caps=1 max_buffers=2
elect@5800x:~$ gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0 
```

But the next time I want to use my Canon again, then I have to reboot, otherwise I'll get 
```bash
[video4linux2,v4l2 @ 0x560d95322e80] ioctl(VIDIOC_G_FMT): Invalid argument
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 -- 
Conversion failed!
```

julioxx22, 2022-02-03:
@elect86 I have the same issue, I'm using video0 and it works fine the first time and after I finish the gphoto2 process, the issue is the following:

`[mjpeg_cuvid @ 0x55cec046ab40] ignoring invalid SAR: 0/0
[video4linux2,v4l2 @ 0x55cec046b300] ioctl(VIDIOC_G_FMT): Invalid argument
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 -- `

eitijupaenoithoowohd, 2022-02-08:
Same issue

msmeissn, 2022-02-11:
seems only a one time use now for some reason, but the problem is likely more on the v4l2loopback device side, not gphoto2 :(

elect86, 2022-02-11:
Any idea how we might investigate further or trying to solve it?

AdamsNimbus, 2022-03-09:
Same issue as above - works once per boot... Still investigating.

givo, 2023-01-31:
I am now facing this issue after I've upgraded my Ubuntu 20.04 kernel to from 5.11 to 5.15.0-58-generic. It's happening to me when I stop an ffmpeg process that writes to a v4l2loopback device and start it again.
On kernel 5.11 I it was working smoothly.

This is pretty much the command I am using to write to a v4l2loopback device:
```
ffmpeg -r 25 -rtsp_transport tcp -i rtsp://192.168.2.15:1935/live/day_center -pix_fmt yuyv422 -f v4l2 /dev/video0
```

this is the error I am getting:
```
[video4linux2,v4l2 @ 0x55e50c309d80] ioctl(VIDIOC_G_FMT): Invalid argument
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 --
```

I also using the latest v4l2loopback release - 0.12.7

schplorg, 2023-06-15:
For anyone with the same issue, do check if the video device is already occupied. My camera created video4 out of the blue and broke my whole setup.

Nischal-Vadiraj, 2023-08-19:
According to the info mentioned [here](/umlaeute/v4l2loopback#changing-options)

First unload the module with rmmod 
```
sudo rmmod v4l2loopback
```

Then load it again using
```
sudo modprobe v4l2loopback video_nr=5 card_label="cam corder" exclusive_caps=1
```
This creates a virtual output device `/dev/video5` and it has the name`"cam corder"`as label on it 

Use it according to your needs
```
ffmpeg -stream_loop 10 -re -i cam.mp4 -f v4l2 -vcodec rawvideo -pix_fmt yuv420p /dev/video5
```

now in this use case , open the camera/webcam and you should see the video playing in loop

> Replace `cam.mp4` with any video 
> Replace `10` with preferred loop count
>`/dev/video5` is the output device

LeVuMinhHuy, 2023-11-22:
> Same issue as above - works once per boot...
> 
> Small fix to avoid reboot:
> 
> ```
> sudo modprobe -r v4l2loopback
> sudo modprobe v4l2loopback exclusive_caps=1 max_buffers=2
> ```
> 
> Then rerun your ffmpeg stream.

you are my hero !

spidy0x0, 2023-12-20:
in my case, guvcview was holding it hostage by being in a sleep state. had to kill those processes first

ImadSaddik, 2025-06-06:
I was facing the same issue when running the following command

```bash
gphoto2 --stdout --capture-movie | ffmpeg -f mjpeg -i - -vf scale=1920:1080 -vcodec rawvideo -pix_fmt yuv420p -f v4l2 /dev/video0
```

Then, I tried to list what is inside `/dev/video/`

```bash
ls /dev/video*
```

And I found that I have other options:

```bash
/dev/video0  /dev/video1  /dev/video2
```

So, I tried `/dev/video1` instead of `/dev/video0`, that did not work but `/dev/video2` worked. In my case, changing the `/dev/video0` to `/dev/video2` solved the issue.

```bash
gphoto2 --stdout --capture-movie | ffmpeg -f mjpeg -i - -vf scale=1920:1080 -vcodec rawvideo -pix_fmt yuv420p -f v4l2 /dev/video0
```

umlaeute, 2025-06-06:
> So, I tried /dev/video1 instead of /dev/video0, that did not work but /dev/video2 worked. In my case, changing the /dev/video0 to /dev/video2 solved the issue.

that basically means that you needed to find the loopback device (rather than trying to write images to, e.g. your webcam).

the canonical ways to check which loopback devices you have are:
- run `v4l2loopback-ctl list` (preferred)
- check the content of `/sys/devices/virtual/video4linux/` (usable even without `v4l2loopback-ctl`

ImadSaddik, 2025-06-06:
@umlaeute Thanks for the info