v4l2loopback/v4l2loopback #248: Kernel module fails to load with kernel 5.4.1

Friz64, 2019-12-03:
* `v4l2loopback` version: aur/v4l2loopback-dkms-git 0.12.2.r0.gd3b198e-1
  * kernel version: 5.4.1-zen
  * Distribution (+version): Arch Linux latest as of 2019-12-03 13:31:30 UTC  

After having upgraded to kernel 5.4.1 `v4l2loopback` fails to load.
  
#### Observed Results:

```
[    3.108679] v4l2loopback: loading out-of-tree module taints kernel.
[    3.108704] v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
[    3.109053] ------------[ cut here ]------------
[    3.109059] WARNING: CPU: 4 PID: 414 at drivers/media/v4l2-core/v4l2-dev.c:863 __video_register_device+0x6cd/0xfe0 [videodev]
[    3.109060] Modules linked in: v4l2loopback(OE+) videodev mc crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 hid_generic usbhid hid sd_mod serio_raw atkbd libps2 ahci libahci libata xhci_pci crc32c_intel scsi_mod xhci_hcd i8042 serio
[    3.109069] CPU: 4 PID: 414 Comm: systemd-modules Tainted: G           OE     5.4.1-zen2-1-zen #1
[    3.109070] Hardware name: BIOSTAR Group B350GT3/B350GT3, BIOS 5.13 08/07/2018
[    3.109074] RIP: 0010:__video_register_device+0x6cd/0xfe0 [videodev]
[    3.109075] Code: fe ff 48 89 83 80 00 00 00 48 85 c0 0f 84 51 04 00 00 8b 53 2c 85 d2 0f 85 21 04 00 00 f0 80 8b e8 04 00 00 01 45 31 ff eb 08 <0f> 0b 41 bf ea ff ff ff 48 8b 44 24 18 65 48 33 04 25 28 00 00 00
[    3.109076] RSP: 0018:ffffa4a44066fbd8 EFLAGS: 00010246
[    3.109077] RAX: ffff9db8d1a86000 RBX: ffff9db8d0f1a800 RCX: 0000000000000001
[    3.109078] RDX: 00000000ffffffff RSI: 0000000000000000 RDI: 0000000000000000
[    3.109079] RBP: ffffffffc034f140 R08: ffffffffc034f140 R09: ffff9db8d0eb0100
[    3.109080] R10: 000000000098f903 R11: ffffffffc034bdb8 R12: 0000000000000001
[    3.109080] R13: 00000000ffffffff R14: ffff9db8d1a86000 R15: ffff9db8d0f1a800
[    3.109081] FS:  00007f6683eac840(0000) GS:ffff9db8d6d00000(0000) knlGS:0000000000000000
[    3.109082] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    3.109083] CR2: 0000555ce4a9d000 CR3: 0000000210afa000 CR4: 00000000003406e0
[    3.109084] Call Trace:
[    3.109089]  init_module+0x6c9/0x1000 [v4l2loopback]
[    3.109090]  ? 0xffffffffc0353000
[    3.109092]  do_one_initcall+0x59/0x234
[    3.109095]  do_init_module+0x5c/0x260
[    3.109097]  load_module+0x2816/0x2a20
[    3.109101]  ? __do_sys_init_module+0x169/0x190
[    3.109102]  __do_sys_init_module+0x169/0x190
[    3.109105]  do_syscall_64+0x4e/0x140
[    3.109107]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[    3.109108] RIP: 0033:0x7f6684cd9d2e
[    3.109109] Code: 48 8b 0d 55 01 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 22 01 0c 00 f7 d8 64 89 01 48
[    3.109110] RSP: 002b:00007ffe7d7678d8 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[    3.109111] RAX: ffffffffffffffda RBX: 0000555ce41e5680 RCX: 00007f6684cd9d2e
[    3.109111] RDX: 00007f668492884d RSI: 0000000000016448 RDI: 0000555ce4a883f0
[    3.109112] RBP: 0000555ce4a883f0 R08: 0000555ce41e401a R09: 0000000000000000
[    3.109113] R10: 0000555ce41e4010 R11: 0000000000000246 R12: 00007f668492884d
[    3.109114] R13: 0000000000000000 R14: 0000555ce41e55f0 R15: 0000555ce41e5680
[    3.109116] ---[ end trace fc942c73f6752cdb ]---
[    3.109116] v4l2loopback: failed video_register_device()
[    3.109117] ------------[ cut here ]------------
```
  
#### Expected Results:

Module loads as expected just like with the old kernel 5.3.13-zen

Friz64, 2019-12-05:
Could you take a look at this @umlaeute?

ribalda, 2019-12-06:
Just send a pull request that I believe it solves the issue: https://github.com/umlaeute/v4l2loopback/pull/249

davidlj95, 2019-12-06:
Thanks @ribalda I'll review and try it today!

davidlj95, 2019-12-06:
#249 worked for me

closed 2019-12-06 by umlaeute