v4l2loopback/v4l2loopback #250: can't work：when i put it inkernel and kconfig=y ，chrome can‘t see the video*

qiurui144, 2020-01-17:
* `v4l2loopback` version:   0.12.3（code from github）

  * kernel version:4.4.131/4.4.154

when i put it inkernel and kconfig=y ，chrome can‘t see the video*

umlaeute, 2020-01-17:
you need to provide much information

what does the kernel say when loading/accessing the device?
what are you using as producer?
does it work with other applications?
did you read the [related issues](https://github.com/umlaeute/v4l2loopback/issues?q=is%3Aissue+chrome), and how does your problem relate to them?

i'm closing this for now. but will re-open once there's a significant amount of data to triage the bug.

closed 2020-01-17 by umlaeute

qiurui144, 2020-01-18:
in kernel，all is ok.
i compare mine to the ko  modprobed from module,the info can see has been loaded, the video* is there.
the info from the udevadm is ok too.
/sys/class/... is also ok...
but chrome can not find the video source
can i send the web demo to you?

qiurui144, 2020-01-18:
next day ,i can send the dmesg and the info to you .

umlaeute, 2020-01-18:
i still don't understand your actual problem.
by now i think that you are trying to build the module into the kernel, as opposed to using it as a kernel-module.
is this correct?

does it work if you are using the `v4l2loopback` kernel-module?

> in kernel，all is ok.

how do you now? what is the actual `v4l2loopback`-related output?

- the info from the udevadm is ok too.

how do you know? what *is* the actual info?

> /sys/class/... is also ok...

how do you know? what does it say?

> but chrome can not find the video source

so what about other applications (`firefox`, `GStreamer`, `ffmpeg`, `v4l2-ctl`, ...)?
do they work ok?

qiurui144, 2020-01-18:
no , i want to push the code in kernel,and set its config=y not as a module

qiurui144, 2020-01-18:
v4l2-ctl is ok,
in firefox i don't have the demo
ffmpeg will be tested tomorrow.

umlaeute, 2020-01-18:
>no, 

i asked 8 questions.
which one did you answer with "no"?

> i want to push the code in kernel,

i dont understand what this means.

> and set its config=y not as a module

what do you mean with "it"? the code? the kernel?
what does this mean?

i'm having serious problems understanding what you want to say.

qiurui144, 2020-01-19:
lanxus@lubuntu:~$ uname -a
Linux lubuntu 4.4.154 #22 SMP Fri Jan 17 14:32:48 CST 2020 aarch64 aarch64 aarch64 GNU/Linux

lanxus@lubuntu:/dev$ udevadm info --name=/dev/video0
P: /devices/virtual/video4linux/video0
N: video0
E: COLORD_DEVICE=1
E: COLORD_KIND=camera
E: DEVNAME=/dev/video0
E: DEVPATH=/devices/virtual/video4linux/video0
E: ID_V4L_CAPABILITIES=:capture:video_output:
E: ID_V4L_PRODUCT=Dummy video device (0x0000)
E: ID_V4L_VERSION=2
E: MAJOR=81
E: MINOR=0
E: SUBSYSTEM=video4linux
E: TAGS=:uaccess:seat:
E: USEC_INITIALIZED=6249312

lanxus@lubuntu:/dev$ udevadm info -a -p /sys/class/video4linux/video0
Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.
 looking at device '/devices/virtual/video4linux/video0':
    KERNEL=="video0"
    SUBSYSTEM=="video4linux"
    DRIVER==""
    ATTR{buffers}=="8"
    ATTR{dev_debug}=="0"
    ATTR{format}==""
    ATTR{index}=="0"
    ATTR{max_openers}=="10"
    ATTR{name}=="Dummy video device (0x0000)"

lanxus@lubuntu:/dev$ v4l2-ctl --list-devices
Dummy video device (0x0000) (platform:v4l2loopback-000):
	/dev/video0

[web_demo.tar.gz](https://github.com/umlaeute/v4l2loopback/files/4082121/web_demo.tar.gz)

qiurui144, 2020-01-19:
lanxus@lubuntu:~/Downloads$ ffmpeg -re -i lv_0_20200104204905.mp4 -f v4l2 /dev/video0
ffmpeg version 2.8.15-0ubuntu0.16.04.1 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.10) 20160609
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'lv_0_20200104204905.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
  Duration: 00:06:00.68, start: 0.046440, bitrate: 1221 kb/s
    Stream #0:0(und): Video: h264 (Baseline) (avc1 / 0x31637661), yuv420p(tv, smpte170m/bt470bg/smpte170m), 1440x1088, 1090 kb/s, 30 fps, 30 tbr, 1000k tbn, 2000k tbc (default)
    Metadata:
      handler_name    : VideoHandler
    Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 128 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
Output #0, v4l2, to '/dev/video0':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf56.40.101
    Stream #0:0(und): Video: rawvideo (I420 / 0x30323449), yuv420p, 1440x1088, q=2-31, 200 kb/s, 30 fps, 30 tbn, 30 tbc (default)
    Metadata:
      handler_name    : VideoHandler
      encoder         : Lavc56.60.100 rawvideo
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> rawvideo (native))
Press [q] to stop, [?] for help
frame=   11 fps=0.0 q=-0.0 size=N/A time=00:00:00.36 bitrate=N/A dup=1 drop=0   frame=   26 fps= 26 q=-0.0 size=N/A time=00:00:00.86 bitrate=N/A dup=1 drop=0   frame=   41 fps= 27 q=-0.0 size=N/A time=00:00:01.36 bitrate=N/A dup=1 drop=0   frame=   57 fps= 28 q=-0.0 size=N/A time=00:00:01.89 bitrate=N/A dup=1 drop=0   frame=   72 fps= 28 q=-0.0 size=N/A time=00:00:02.39 bitrate=N/A dup=1 drop=0   frame=   87 fps= 29 q=-0.0 size=N/A time=00:00:02.89 bitrate=N/A dup=1 drop=0   frame=  102 fps= 29 q=-0.0 size=N/A time=00:00:03.39 bitrate=N/A dup=1 drop=0   frame=  117 fps= 29 q=-0.0 size=N/A time=00:00:03.89 bitrate=N/A dup=1 drop=0   frame=  132 fps= 29 q=-0.0 size=N/A time=00:00:04.39 bitrate=N/A dup=1 drop=0   frame=  148 fps= 29 q=-0.0 size=N/A time=00:00:04.93 bitrate=N/A dup=1 drop=0   frame=  163 fps= 29 q=-0.0 size=N/A time=00:00:05.43 bitrate=N/A dup=1 drop=0   frame=  178 fps= 29 q=-0.0 size=N/A time=00:00:05.93 bitrate=N/A dup=1 drop=0   frame=  193 fps= 29 q=-0.0 size=N/A time=00:00:06.43 bitrate=N/A dup=1 drop=0   frame=  208 fps= 29 q=-0.0 size=N/A time=00:00:06.93 bitrate=N/A dup=1 drop=0   frame=  223 fps= 29 q=-0.0 size=N/A time=00:00:07.43 bitrate=N/A dup=1 drop=0   frame=  239 fps= 30 q=-0.0 size=N/A time=00:00:07.96 bitrate=N/A dup=1 drop=0   frame=  254 fps= 30 q=-0.0 size=N/A time=00:00:08.46 bitrate=N/A dup=1 drop=0   frame=  269 fps= 30 q=-0.0 size=N/A time=00:00:08.96 bitrate=N/A dup=1 drop=0   frame=  284 fps= 30 q=-0.0 size=N/A time=00:00:09.46 bitrate=N/A dup=1 drop=0   frame=  299 fps= 30 q=-0.0 size=N/A time=00:00:09.96 bitrate=N/A dup=1 drop=0   frame=  314 fps= 30 q=-0.0 size=N/A time=00:00:10.46 bitrate=N/A dup=1 drop=0   frame=  329 fps= 30 q=-0.0 size=N/A time=00:00:10.96 bitrate=N/A dup=1 drop=0   frame=  344 fps= 30 q=-0.0 size=N/A time=00:00:11.46 bitrate=N/A dup=1 drop=0   frame=  360 fps= 30 q=-0.0 size=N/A time=00:00:11.99 bitrate=N/A dup=1 drop=0   frame=  375 fps= 30 q=-0.0 size=N/A time=00:00:12.49 bitrate=N/A dup=1 drop=0   frame=  377 fps= 30 q=-0.0 Lsize=N/A time=00:00:12.56 bitrate=N/A dup=1 drop=0    
//i press q to quit
video:865215kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown

qiurui144, 2020-01-19:
i push the output log to the github .

>no,

i asked 8 questions.
which one did you answer with "no"?

> i want to push the code in kernel,

i dont understand what this means.

> and set its config=y not as a module

what do you mean with "it"? the code? the kernel?
what does this mean?

i'm having serious problems understanding what you want to say.

umlaeute, 2020-01-19:
thanks.
now please give us a *step-by-step* guide how to reproduce the problem on a fresh system.

umlaeute, 2020-01-19:
also note, that afaics the solution mentioned in https://github.com/umlaeute/v4l2loopback/issues/193 (setting `exclusive_caps=1` for the relevant device) fixes the issue.

make sure to show in your *step-by-step* guide, how you set the `exclusive_caps`.

qiurui144, 2020-01-20:
i don't treat it as a module
i push the .c and .h into the driver folder,change kconfig and makefile for the v4l2loopback,then change the defconfig to Y.
the .o is built and the v4l2loopback has been loaded

umlaeute, 2020-01-20:
please provide step-by -step instructions, documenting every single step in a way, that enables people who don't know anything about kernels, modules, compilers,... to reproduce your problem. like:
1. 
download the kernel with

apt-get install linux-source-1.2.3

2. 
get the v4l2loopback sources

git clone https://github.com/umlaeute/v4l2loopback

3. 
copy the v4l2loopback sources to the kernel-tree

cp v4l2loopback/v4l2loopback.c /usr/src/linux-1.2.3/...

4. 
and so on...

qiurui144, 2020-01-31:
1. https://github.com/rockchip-linux/manifests
download by repo for V2.0.7 SDK
2.put v4l2loopback.c & v4l2loopback_formats.h into kernel/drivers/video/
add the config into kernel/drivers/video/kconfig and kernel/drivers/video/Makefile(the normal info is ok)
3../build.sh kernel (this will be make boot (include kernel and resource-devicetree)),be careful about the config

because this is base on rockchip board , if you need more info ,email me

i'm so sorry ,because epidemic happens in china, i can't go to my company for my code ,sorry.

and so on...

qiurui144, 2020-01-31:
1. https://github.com/rockchip-linux/manifests
download by repo for V2.0.7 SDK
2.put v4l2loopback.c & v4l2loopback_formats.h into kernel/drivers/video/
add the config into kernel/drivers/video/kconfig and kernel/drivers/video/Makefile(the normal info is ok)
3../build.sh kernel (this will be make boot (include kernel and resource-devicetree)),be careful about the config

because this is base on rockchip board , if you need more info ,email me

qiurui144, 2020-02-05:
anyone here?