v4l2loopback/v4l2loopback #251: WebRTC videotestsrc freezes with gst-launch-1.0 but not with gst-launch-0.10

wehlutyk, 2020-01-30:
First of all, thanks for the great module! Lots of fun stuff can be done with this. Now I've been all over SO and the issues here, to no avail with this.

### Environment

* `v4l2loopback` version: 0.12.3
* kernel version: `Linux keon 4.19.99-1-lts #1 SMP Mon, 27 Jan 2020 21:30:28 +0000 x86_64 GNU/Linux`
* Distribution (+version):
    ```
    LSB Version:	1.4
    Distributor ID:	Arch
    Description:	Arch Linux
    Release:	rolling
    Codename:	n/a
    ```
  
### Problem

Ultimately, I want to feed GStreamer's `videotestsrc` to a WebRTC conference (`videotestsrc` for now, and ultimately an incoming RTMP stream).

To test using `videotestsrc` as a webcam in the browser, I run: 
```
sudo modprobe v4l2loopback exclusive_caps=1
gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video2  # with or without identity drop-allocation=1, if there is an error
```

Then I open the following webpage:
```html
<!doctype html>
<html>
  <head>
    <title>WebRTC video test</title>
    <meta charset='utf-8'>
  </head>
  <body>
    <div class="contentarea">
      <video id="video">Video stream not available.</video>
    </div>
    <script>
      navigator.mediaDevices.getUserMedia({video: true})
      .then(function(mediaStream) {
        var video = document.querySelector('video');
        video.srcObject = mediaStream;
        video.onloadedmetadata = function(e) {
          video.play();
        };
      })
      .catch(function(err) { console.log(err.name + ": " + err.message); });
    </script>
  </body>
</html>
```

In the browser, the v4l2loopback device appears as an option for webcams, so I select it.

#### Observed Results

A single video frame is shown (the test pattern) after which the video freezes. I expect the video to run.

If instead I use `gst-launch-0.10 videotestsrc ! v4l2sink device=/dev/video2 ` (so GStreamer 0.10), the video plays in the browser.

### Relevant logs
In case the caps used are useful:

```
gst-launch-1.0 -v videotestsrc ! v4l2sink device=/dev/video2
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1
/GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0.GstPad:sink: caps = video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, framerate=(fraction)30/1, multiview-mode=(string)mono, interlace-mode=(string)progressive, colorimetry=(string)2:4:7:1, pixel-aspect-ratio=(fraction)1/1
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```

```
$ gst-launch-0.10 -v videotestsrc ! v4l2sink device=/dev/video2
Setting pipeline to PAUSED ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw-yuv, format=(fourcc)YUY2, width=(int)320, height=(int)240, interlaced=(boolean)false, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, color-matrix=(string)sdtv, chroma-site=(string)mpeg2
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: queue-size = 8
/GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0.GstPad:sink: caps = video/x-raw-yuv, format=(fourcc)YUY2, width=(int)320, height=(int)240, interlaced=(boolean)false, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, color-matrix=(string)sdtv, chroma-site=(string)mpeg2
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```

umlaeute, 2020-01-30:
did you try setting `max_buffers=2`? (i'm thinking of making this the default)

umlaeute, 2020-01-30:
which browsers did you try?
while i get occasional blocks with firefox, chromium (chrome) doesn't seem to have this problem.

wehlutyk, 2020-01-30:
So
```
$ sudo modprobe v4l2loopback exclusive_caps=1 max_buffers=2
$ gst-launch-1.0 -v videotestsrc ! v4l2sink device=/dev/video2
```
Same behaviour.

I tested Firefox, and the device doesn't show up in Chromium (looks like the Chromium bug [linked to in #78](https://github.com/umlaeute/v4l2loopback/issues/78#issuecomment-492721762)).

Surprisingly, it does show up in [jitsi-meet](https://jitsi.org/jitsi-meet/) which is an Electron app, and the video preview shows a running `videotestsrc` in the source-selection modal dialog, but after validating the app goes back to the physical webcam source (after sometimes showing a single frame from `videotestsrc`).

umlaeute, 2020-01-30:
just to be sure: you *did* unload the `v4l2loopback` module before running `modprobe` again?

wehlutyk, 2020-01-30:
Yes I did

wehlutyk, 2020-01-30:
I should add that in both cases VLC can read the capture device and the video plays (so it's only when reading it in Firefox that I see this behaviour). Could VLC help me debug this?

Wallacy, 2020-02-07:
> did you try setting `max_buffers=2`? (i'm thinking of making this the default)

Not related to this thread but max_buffers=2 some some issues on my pipeline. Also, will be nice to make MAX_DEVICES configurable too.

aanarchy, 2020-02-11:
I also have this issue

> did you try setting `max_buffers=2`? (i'm thinking of making this the default)

I also loaded the module with max_buffers=2 to no avail. 
The specific command I ran is this: "sudo modprobe v4l2loopback exclusive_caps=1 devices=1 max_buffers=2"
The device works perfectly in Brave (Chromium derivative) but freezes in Firefox.

stephematician, 2025-02-02:
@wehlutyk - I made a quick attempt to replicate this; but I'm not a web developer. I used nginx to serve your example with the following settings:

```
server {
        listen 5000 default_server;
        listen [::]:5000 default_server;

        root /var/www/home;

        server_name _;

        location / {
                # First attempt to serve request as file, then
                # as directory, then fall back to displaying a 404.
                try_files $uri $uri/ =404;
        }

}
```

The contents of `/var/www/home` include `index.html` as per your example.

In Firefox - I had to right-click the resouce and select <kbd>Play</kbd> then the video plays. This worked with either gstreamer-1.0 or gstreamer-0.10

stephematician, 2025-02-03:
Video plays automatically by changing `.onloadedmetadata` to `.oncanplay`:

```html
    <script>
      navigator.mediaDevices.getUserMedia({video: true})
      .then(function(mediaStream) {
        var video = document.querySelector('video');
        video.srcObject = mediaStream;
        video.oncanplay = function(e) {
          video.play();
        };
      })
      .catch(function(err) { console.log(err.name + ": " + err.message); });
    </script>
```

wehlutyk, 2025-02-04:
I'm sorry I've been out of this, if it's okay for you we can keep it open if someone else can test, but I'm otherwise fine with closing as I can't help currently. Thanks!

stephematician, 2025-02-04:
Update: with #611 this worked in __GStreamer__ 1.24.2 (Ubuntu 24.04), 1.8.3 (Ubuntu 16.04), and 0.10.36 (Ubuntu 16.04). So I'm pretty sure that the PR fixes it, but it'll take some patience before it gets merged as it's a big one.

closed 2025-02-06 by umlaeute