v4l2loopback/v4l2loopback #255: 5.5.4-xanmod3: modprobe "Bad address"

ZyanKLee, 2020-02-16:
First of all: Thanks for your great work on this fabolous project.

Now, on to less merry topics: ;-)

### Step 2: Describe your environment

  * `v4l2loopback` version: `v4l2loopback-dkms 0.12.2-1`
  * kernel version: `5.5.4-xanmod3 #1.200215`
  * Distribution (+version): 

          Distributor ID:	Ubuntu
          Description:	Ubuntu 19.10
          Release:	19.10
          Codename:	eoan

### Step 3: Describe the problem:

#### Steps to reproduce:

  1. install xanmod kernel on ubuntu eoan
  2. reboot
  3. install v4l2loopback-dkms 
  
#### Observed Results:

  * modprobe and insmod complain about "Bad address"
  * module is not loaded
  * trace in dmesg (see below)
  
#### Expected Results:

  * I expected the module to be loaded and the v4l2 loopback device node to be available after that.
  
#### DMESG Trace:

  ```
[  363.331171] ------------[ cut here ]------------
[  363.331191] WARNING: CPU: 5 PID: 561 at drivers/media/v4l2-core/v4l2-dev.c:885 __video_register_device+0x9c6/0x11b0 [videodev]
[  363.331192] Modules linked in: v4l2loopback(OE+) veth xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c bpfilter br_netfilter bridge stp llc aufs overlay binfmt_misc nls_iso8859_1 nvidia_uvm(OE) nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm snd_hda_codec_hdmi snd_hda_codec_realtek irqbypass crct10dif_pclmul crc32_pclmul snd_hda_codec_generic ledtrig_audio ghash_clmulni_intel snd_usb_audio snd_hda_intel drm_kms_helper snd_intel_dspcfg snd_usbmidi_lib snd_hda_codec snd_seq_midi uvcvideo drm snd_seq_midi_event aesni_intel videobuf2_vmalloc snd_rawmidi crypto_simd ipmi_si cryptd videobuf2_memops videobuf2_v4l2 xpad videobuf2_common glue_helper videodev ipmi_devintf input_leds joydev mc ff_memless snd_hda_core intel_cstate mxm_wmi ipmi_msghandler snd_seq snd_hwdep
[  363.331240]  intel_rapl_perf fb_sys_fops syscopyarea sysfillrect sysimgblt snd_pcm snd_seq_device snd_timer ioatdma mei_me snd mei soundcore dca mac_hid parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_lg_g15 hid_steam hid_generic usbhid hid isci ahci e1000e libsas i2c_i801 libahci lpc_ich scsi_transport_sas wmi
[  363.331259] CPU: 5 PID: 561 Comm: modprobe Tainted: P        W  OE     5.5.4-xanmod3 #1.200215
[  363.331260] Hardware name: ASUSTeK COMPUTER INC. Z9PE-D8 WS/Z9PE-D8 WS, BIOS 5802 06/10/2015
[  363.331268] RIP: 0010:__video_register_device+0x9c6/0x11b0 [videodev]
[  363.331271] Code: 12 00 48 89 83 80 00 00 00 48 85 c0 0f 84 0c 02 00 00 8b 53 2c 85 d2 0f 85 dc 01 00 00 f0 80 8b 08 05 00 00 01 45 31 ff eb 08 <0f> 0b 41 bf ea ff ff ff 48 8b 44 24 20 65 48 33 04 25 28 00 00 00
[  363.331273] RSP: 0018:ffffbc4288b0fbe0 EFLAGS: 00010246
[  363.331275] RAX: ffff9d0499530000 RBX: ffff9d04e7627000 RCX: 0000000000000000
[  363.331276] RDX: 00000000ffffffff RSI: 0000000000000000 RDI: ffff9d04e7627000
[  363.331277] RBP: ffffffffc0b91180 R08: ffffffffc0b91180 R09: ffff9d04995300a0
[  363.331278] R10: ffffffffc0b8ddc4 R11: ffffffffc04b78e0 R12: 0000000000000001
[  363.331280] R13: 00000000ffffffff R14: ffff9d0499530310 R15: ffff9d0499530078
[  363.331282] FS:  00007fafb2f92240(0000) GS:ffff9d04efb40000(0000) knlGS:0000000000000000
[  363.331283] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  363.331285] CR2: 00007ffcb5b0ce48 CR3: 00000002e22b2005 CR4: 00000000000606e0
[  363.331286] Call Trace:
[  363.331298]  init_module+0x6db/0x1000 [v4l2loopback]
[  363.331302]  ? 0xffffffffc0b2c000
[  363.331308]  do_one_initcall+0x46/0x230
[  363.331313]  ? free_unref_page_commit+0x95/0x110
[  363.331318]  ? kmem_cache_alloc_trace+0x17b/0x220
[  363.331321]  ? do_init_module+0x23/0x230
[  363.331324]  do_init_module+0x5c/0x230
[  363.331327]  load_module+0x28bf/0x2b30
[  363.331333]  ? ima_post_read_file+0xf0/0x100
[  363.331338]  ? __do_sys_finit_module+0xaa/0x110
[  363.331340]  __do_sys_finit_module+0xaa/0x110
[  363.331345]  do_syscall_64+0x5f/0x220
[  363.331351]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  363.331353] RIP: 0033:0x7fafb30d794d
[  363.331355] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 13 e5 0c 00 f7 d8 64 89 01 48
[  363.331357] RSP: 002b:00007ffcb5b0ff38 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[... 110 lines cut ...]
[  363.331867]  free_devices+0x9b/0x10c [v4l2loopback]
[  363.331870]  init_module+0x70a/0x1000 [v4l2loopback]
[  363.331872]  ? 0xffffffffc0b2c000
[  363.331875]  do_one_initcall+0x46/0x230
[  363.331877]  ? free_unref_page_commit+0x95/0x110
[  363.331880]  ? kmem_cache_alloc_trace+0x17b/0x220
[  363.331882]  ? do_init_module+0x23/0x230
[  363.331884]  do_init_module+0x5c/0x230
[  363.331886]  load_module+0x28bf/0x2b30
[  363.331890]  ? ima_post_read_file+0xf0/0x100
[  363.331894]  ? __do_sys_finit_module+0xaa/0x110
[  363.331896]  __do_sys_finit_module+0xaa/0x110
[  363.331901]  do_syscall_64+0x5f/0x220
[  363.331903]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  363.331905] RIP: 0033:0x7fafb30d794d
[  363.331906] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 13 e5 0c 00 f7 d8 64 89 01 48
[  363.331907] RSP: 002b:00007ffcb5b0ff38 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  363.331909] RAX: ffffffffffffffda RBX: 000055a195704ea0 RCX: 00007fafb30d794d
[  363.331910] RDX: 0000000000000000 RSI: 000055a19468e3f0 RDI: 0000000000000003
[  363.331911] RBP: 000055a19468e3f0 R08: 0000000000000000 R09: 000055a195704ea0
[  363.331912] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[  363.331913] R13: 000055a195704690 R14: 0000000000040000 R15: 000055a195704ea0
[  363.331916] ---[ end trace 01f3f46f710a6c43 ]---

  ```

ElDavoo, 2020-03-27:
Same issue with a gentoo linux install with 5.5.13 kernel and  v4l2loopback 0.12.1.1_pre

pdvrieze, 2020-03-31:
I used the straight from git version on Gentoo (with kernel 5.5.8). Would this be a matter of having some missing configuration options? If so which (btw. the fork from droidcam does work and load)

ElDavoo, 2020-04-02:
Strange, as I solved by using the latest commit from master (always using the same, last gcc version (9.3.0)), while you solved updating gcc (from which version to which?)

umlaeute, 2020-04-02:
@ElDavoo i'm pretty sure that updating the module-sources had nothing to do with fixing the "Bad address" error.

1seven, 2020-04-09:
Hi! I have the same issue using Gentoo Linux with the 5.4.28 kernel (with 4.19.97 everything was working), gcc (Gentoo 9.2.0-r2 p3) 9.2.0 and v4l2loopback 0.12.1.1_pre.

@ElDavoo Cloning the repository directly, _make_, _make install_ didn't fix the problem for me. Did you probably change something else as well?

marsark, 2020-04-09:
Same problem here. Also Gentoo Linux, kernel 5.4.28. gcc 9.2.0. Module build by portage ebuild fails and latest sources from Git fails also.

1seven, 2020-04-10:
@marsark so your module doesn't even build or did you also mean modprobe-ing  fails with "Bad address"?

marsark, 2020-04-10:
In my case, the module builds, but it fails to modprobe to kernel with "Bad Address."

Ulrar, 2020-04-13:
Version 9999 (from repo) seems to solve that issue on gentoo

umlaeute, 2020-04-13:
@Ulrar what version`9999`? there is no such version of `v4l2loopback`, so which *repo* do you refer to?

Ulrar, 2020-04-13:
I mean gentoo's version 9999, that's how gentoo references the git version. It just means it's built from whatever is the latest commit on master at the time

1seven, 2020-04-14:
Using version 9999 from the official gentoo repo works for me as well :)

I just wonder why the maintainer of the ebuild chose to use the git protocol instead of https, because after emerging I receive the following message:

* Messages for package media-video/v4l2loopback-9999:

* git-r3: git protocol is completely unsecure and may render the ebuild
* easily susceptible to MITM attacks (even if used only as fallback). Please
* use https instead.
* [URI: git://github.com/umlaeute/v4l2loopback.git]

cloning via https should work perfectly fine as well I'd assume.
However, this might not be an issue to discuss here, but rather with the maintainer of that ebuild.

ZyanKLee, 2020-05-03:
I can confirm: with current git master branch I can load the module into my current kernel (`5.6.8-xanmod1`)

umlaeute, 2020-05-04:
thanks for reporting back and confirming that it now works. closing this.

i seriously doubt that the current git `master` of v4l2loopback (which is d3f15e3ce3388d5b4a99c2240aac3c06bc39dda8) has anything to do with solving the "Bad address" problem.

if the module compiles correctly, then it should be properly usable by the kernel it was compiled against.
however, if you manage to force a module onto a different kernel than it was compiled for, then bad things can happen.

i'm really suspecting that the problem lies with the kernel provider (e.g. *xanmod*) failing to provide kernel headers that match the kernel image.

@Ulrar wrote:
> I just wonder why the maintainer of the ebuild [...]
> [...]
> this might not be an issue to discuss here, but rather with the maintainer of that ebuild.

indeed.
please ask them directly and don't use *this* tracker for this. we already have a quite high signal/noise ratio, and such comments don't help at all.

closed 2020-05-04 by umlaeute

titanofold, 2020-06-01:
> thanks for reporting back and confirming that it now works. closing this.
> 
> i seriously doubt that the current git `master` of v4l2loopback (which is [d3f15e3](https://github.com/umlaeute/v4l2loopback/commit/d3f15e3ce3388d5b4a99c2240aac3c06bc39dda8)) has anything to do with solving the "Bad address" problem.
> 
> if the module compiles correctly, then it should be properly usable by the kernel it was compiled against.
> however, if you manage to force a module onto a different kernel than it was compiled for, then bad things can happen.
> 
> i'm really suspecting that the problem lies with the kernel provider (e.g. _xanmod_) failing to provide kernel headers that match the kernel image.

The most current, versioned release in the Gentoo repo is 0.12.1.1_pre.

No matter how many times I recompiled the kernel (5.4.38) or v4l2loopback (always after compiling the kernel), I got "Bad address".

Using the current git via Gentoo's live ebuild (the 9999 version) is what fixed it for me.

Literally, that's the only change to my system between bad address and working.

umlaeute, 2020-06-01:
well, yes: this ticket is probably a duplicate of #248 (and eventually #257)