v4l2loopback/v4l2loopback #260: Module fails to load

Xartos, 2020-03-13:
### Describe your environment

  * `v4l2loopback` version: master, v0.12.3, v0.12.0

  * kernel version: 
```
$ uname -a
Linux my-computer-name 4.15.0-1067-oem #77-Ubuntu SMP Mon Jan 20 09:23:07 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

  * Distribution (+version):
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 19.10
Release:	19.10
Codename:	eoan
```
  * Compiler: `gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008`
  
### Describe the problem:

After the module is built successfully and I try to load the module I get:
```
$ sudo modprobe v4l2loopback 
modprobe: ERROR: could not insert 'v4l2loopback': Exec format error
```
And the message that the kernel prints is:
```
[ 1667.728397] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000002619fd62, val ffffffffc0cc929d
```

#### Steps to reproduce:

  1. `make`
  2. `sudo make install`
  3. `sudo depmod -a`
  4. `sudo modprobe v4l2loopback`
  
#### Observed Results:

```
$ sudo modprobe v4l2loopback 
modprobe: ERROR: could not insert 'v4l2loopback': Exec format error
```

```
$ dmesg
.
.
.
[ 1667.728397] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 000000002619fd62, val ffffffffc0cc929d
```
  
#### Expected Results:

Expected the module to load correctly

#### Notes:

The same exact problem occurs if I download the `v4l2loopback-dkms` package. And secure boot is disabled.

umlaeute, 2020-03-13:
while i acknowledge your problem, this is not fixable on the v4l2loopback side.

the issue is either 
- with your kernel/kernel-headers installation (as in: they don't match), or 
- with your kbuild toolchain installation (to build kernel-modules you must use the same compiler that was used for building the kernel).

see also related issues like https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/84

closed 2020-03-13 by umlaeute

Xartos, 2020-03-13:
Thanks, the link help me fix the problem!
I built it with `make CC=/usr/bin/gcc-8` and now it works