v4l2loopback/v4l2loopback #261: gst-launch-1.0 v4l2src stops after start

nikicat, 2020-03-15:
# Environment

  * `v4l2loopback` version: 0.12.3
  * kernel version: 5.4.22-1-lts
  * Distribution:
    LSB Version:	1.4
    Distributor ID:	Arch
    Description:	Arch Linux
    Release:	rolling
    Codename:	n/a

### The problem
`gst-launch-1.0 v4l2src device=/dev/video0 ! xvimagesink` doesn't work with v4l2loopback

#### Steps to reproduce:

  1. `modprobe v4l2loopback`
  2. `gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video0`
  3. `gst-launch-1.0 v4l2src device=/dev/video0 ! xvimagesink`
  
#### Observed Results:

  `gst-launch-1.0 v4l2src device=/dev/video0 ! xvimagesink` shows window with a test source for a moment and then immidiately quits.
It's output:
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Failed to allocate a buffer
Additional debug info:
../gst-plugins-good/sys/v4l2/gstv4l2src.c(1001): gst_v4l2src_create (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0
Execution ended after 0:00:00.051914320
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

VLC works fine with the device
Chrome dosn't see the device
Firefox sees the device but shows only one still frame from it.
#### Expected Results:

  * gst-launch-1.0 v4l2src shows captured video
  * Chrome sees the device
  * Chrome and Firefox capture video form the device

umlaeute, 2020-03-15:
afaics, this is a duplicate of #251 and probably #193

umlaeute, 2020-03-15:
ah, and the `gst-launch-1.0 v4l2src device=/dev/video0 ! xvimagesink` issue can easily fixed by only allowing two (2) buffers:

~~~sh
# modprobe v4l2loopback max_buffers=2
~~~

this should probably be the default.

closed 2020-03-15 by umlaeute

nikicat, 2020-03-15:
Thanks for the answers! `max_buffers=2 exclusive_caps=1` helped with gst-launch and Chrome, but Firefox still shows single frame (0 fps). Am I missing any more options?

umlaeute, 2020-03-16:
no. but it's still a dupe of #251