v4l2loopback/v4l2loopback #263: v4l2loopback build failed since v5.6-rc1

liuyd96, 2020-03-19:
### Testing environment

  * `v4l2loopback` version:  0.12.3

  * kernel version: 5.6.0-rc1

  * Distribution (+version):  Debian GNU/Linux 9 (stretch)
  
### Problem: Just build failed

> root@lkp-hsw-ep4 /lkp/benchmarks/v4l2loopback# make
> Building v4l2-loopback driver...
> make -C /lib/modules/`uname -r`/build M=/lkp/benchmarks/v4l2loopback modules
> make[1]: Entering directory '/tmp/linux'
> 
>   WARNING: Symbol version dump ./Module.symvers
>            is missing; modules will have no dependencies and modversions.
> 
>   CC [M]  /lkp/benchmarks/v4l2loopback/v4l2loopback.o
> /lkp/benchmarks/v4l2loopback/v4l2loopback.c: In function 'vidioc_qbuf':
> /lkp/benchmarks/v4l2loopback/v4l2loopback.c:1523:24: error: passing argument 1 of 'v4l2l_get_timestamp' from incompatible pointer type [-Werror=incompatible-pointer-types]
>     v4l2l_get_timestamp(&b->buffer.timestamp);
>                         ^
> /lkp/benchmarks/v4l2loopback/v4l2loopback.c:137:20: note: expected 'struct timeval *' but argument is of type 'struct __kernel_v4l2_timeval *'
>  static inline void v4l2l_get_timestamp(struct timeval *tv) {
>                     ^~~~~~~~~~~~~~~~~~~
> /lkp/benchmarks/v4l2loopback/v4l2loopback.c: In function 'v4l2_loopback_write':
> /lkp/benchmarks/v4l2loopback/v4l2loopback.c:1950:22: error: passing argument 1 of 'v4l2l_get_timestamp' from incompatible pointer type [-Werror=incompatible-pointer-types]
>   v4l2l_get_timestamp(&b->timestamp);
>                       ^
> /lkp/benchmarks/v4l2loopback/v4l2loopback.c:137:20: note: expected 'struct timeval *' but argument is of type 'struct __kernel_v4l2_timeval *'
>  static inline void v4l2l_get_timestamp(struct timeval *tv) {
>                     ^~~~~~~~~~~~~~~~~~~
> /lkp/benchmarks/v4l2loopback/v4l2loopback.c: In function 'init_buffers':
> /lkp/benchmarks/v4l2loopback/v4l2loopback.c:2055:23: error: passing argument 1 of 'v4l2l_get_timestamp' from incompatible pointer type [-Werror=incompatible-pointer-types]
>    v4l2l_get_timestamp(&b->timestamp);
>                        ^
> /lkp/benchmarks/v4l2loopback/v4l2loopback.c:137:20: note: expected 'struct timeval *' but argument is of type 'struct __kernel_v4l2_timeval *'
>  static inline void v4l2l_get_timestamp(struct timeval *tv) {
>                     ^~~~~~~~~~~~~~~~~~~
> cc1: some warnings being treated as errors
> scripts/Makefile.build:267: recipe for target '/lkp/benchmarks/v4l2loopback/v4l2loopback.o' failed
> make[2]: *** [/lkp/benchmarks/v4l2loopback/v4l2loopback.o] Error 1
> Makefile:1681: recipe for target '/lkp/benchmarks/v4l2loopback' failed
> make[1]: *** [/lkp/benchmarks/v4l2loopback] Error 2
> make[1]: Leaving directory '/tmp/linux'
> Makefile:42: recipe for target 'v4l2loopback.ko' failed
> make: *** [v4l2loopback.ko] Error 2
   
Build failed since v5.6-rc1. I have tried gcc-6 and gcc-7.

umlaeute, 2020-03-19:
Closed by #252

umlaeute, 2020-03-19:
there's no release yet, but you can get the current git-version which contains the fix.

closed 2020-03-19 by umlaeute