v4l2loopback/v4l2loopback #264: Fake ffmpeg webcam has bad resolution

tuxflo, 2020-03-19:
Now with Corona around I try to use v4l2loopback to share my screen with coworkers through different online tools like Zoom, Jitsi, MS Teams or any other WebRTC based tool. However the resolution on the receiving clients is too bad to see the screen content.
The resolution of my screen is 1920x1080.

### Environment

  * `v4l2loopback` version: 0.12.3-2
```
sudo dmesg  | grep -i v4l2loopback
[ 1853.691003] Modules linked in: v4l2loopback(OE) ...
```

  * kernel version: 5.5.8-1-MANJARO 

  * Distribution: manjaro (KDE edition), 19.02

#### Steps to reproduce:

  1. create a v4l2loopback device using ` sudo modprobe v4l2loopback  devices=1 video_nr=5 exclusive_caps=1`
  2. Stream desktop with ffmpeg: ` ffmpeg -f x11grab -s 1920x1080 -i :0.0+0,0 -vf format=pix_fmts=yuv420p -f v4l2 /dev/video5`
  3. Use the "Dummy webcam" in a video chat application like Jitsi or Zoom
  4. Test the application with a partner or second computer

  
#### Observed Results:

Resolution is not good enough to see anything (see picture below), although on the own screen everything seems fine.

[image]
  
#### Expected Results:
Image quality is the same as provided by ffmpeg (1920x1080) or at least not that bad on the second PC (or real partner computer).

umlaeute, 2020-03-20:
`v4l2loopback` doesn't do any resampling or conversion: it passes the input to it's output.

so it's either your producer (`ffmpeg`) or your consumer (`jitsi`, `zoom`,...) that makes the quality go bad.

closed 2020-03-20 by umlaeute

jremi, 2020-06-25:
@tuxflo did you figure out how to make the quality better?

tuxflo, 2020-06-25:
No I didn't. I found out that multiple screen sharing support was added in one of the recent Chromium updates, so I didn't have to lookout for workarounds like this.