v4l2loopback/v4l2loopback #267: v4l2: Inappropriate ioctl for device

ahtabrizi, 2020-03-26:
###  Describe your environment

  * `v4l2loopback` version: _____

          sudo dmesg  | grep -i v4l2loopback
          [  385.078050] v4l2loopback: loading out-of-tree module taints kernel.
          [  385.078145] v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
          [  385.079220] v4l2loopback driver version 0.9.1 loaded

  * kernel version: _____

          uname -a
         Linux redhood 4.15.0-91-generic #92~16.04.1-Ubuntu SMP Fri Feb 28 14:57:22 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

  * Distribution (+version): _____

          lsb_release -a
          No LSB modules are available.
          Distributor ID: Ubuntu
          Description:    Ubuntu 16.04.6 LTS
          Release:        16.04
          Codename:       xenial

  
###  Describe the problem:
I'm trying to use a virtual device by v4l2loopback as a camera using `dmabuf` functionality. I feedthis virtual device from my laptops camera `/dev/video0`. I'm doing the following steps.

#### Steps to reproduce:

  1. run this command to link USB camera to virtual device `/dev/video1`

           gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! v4l2sink device=/dev/video1

  2.  C code to initialize the v4l2 device as `dmabuf`, [here](https://gist.github.com/ahtabrizi/8aad8fe96e0183a2a3adf21d6d379299)

#### Observed Results:
It throws an error when it tries to get the `fd` to the buffers by running `ioctl(fd, VIDIOC_EXPBUF, &expbuf)`. the error is 

           Inappropriate ioctl for device

#### What I have tried
- Simple gstreamer consumer works fine from commad line
- Simple `OpenCV` python program to read this video device
It seems that device is working currectly with normal camera modes, but fails when trying to export file descriptors of the buffers 
- I even used `io-mode` property of `v4l2src` and `v4l2sink`:
```
# io-mode only for v4l2src
gst-launch-1.0 v4l2src device=/dev/video0 io-mode=dmabuf ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! v4l2sink device=/dev/video1
```
works but running the C code outputs same error, `Inappropriate ioctl for device`.
```
# io-mode only for v4l2sink
gst-launch-1.0 v4l2src device=/dev/video0 io-mode=dmabuf ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! v4l2sink device=/dev/video1

# io-mode for both v4l2src and v4l2sink
gst-launch-1.0 v4l2src device=/dev/video0 io-mode=dmabuf ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! v4l2sink device=/dev/video1 io-mode=dmabuf
```
both commands outputs:
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Failed to allocated required memory.
Additional debug info:
gstv4l2sink.c(621): gst_v4l2sink_show_frame (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Buffer pool activation failed
Execution ended after 0:00:01.341391891
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
- I even added `videoconvert`, `tee` and `queue` commands to gstreamer commands but errors remaind the same.

Any tips to get this working?!

umlaeute, 2020-03-26:
please try the [latest release](https://github.com/umlaeute/v4l2loopback/releases) of `v4l2loopback` (at time of writing this is [v0.12.3](https://github.com/umlaeute/v4l2loopback/releases/v0.12.3)).  if that doesn't fix the problem, try the current *git master*

ahtabrizi, 2020-03-26:
@umlaeute tested with both of them. Same problem

umlaeute, 2020-03-26:
ah well, on re-evaluating:

- the test-code you posted is a bit bogus; it exits and prints errors, even if `ioctl(VIDIOC_EXPBUF)` returns success (`0`)
- i don't think anybody has written code to explicitely support `VIDIOC_EXPBUF`. no promise has been made to support it
  - if you want it to be supported, patches are welcome :-)

closed 2020-03-26 by umlaeute

ahtabrizi, 2020-03-27:
Thanks for your fast reply.
well, test-code is just the initialization part where the error occurs. I thought that it was redundant to include EGL stuff which I show the frames with. By the way, it works fin on `dev/device0` which is my laptop's webcam.
Anyway, apparently there is no support for `VIDIOC_EXPBUF` in `v4l2loopback`. I don't know if I will have the time for it, but just for future reference where should I begin to do the patch?

Thanks

mohamedAdhamc, 2024-09-04:
Any chance the .vidioc_expbuf() callback would ever be implemented in v4l2loopback ? I'm facing the same problem as well.