v4l2loopback/v4l2loopback #269: Installation errors on 5.5.11 kernel via aptitude

Micker003, 2020-03-30:
I tried to install via Aptitude on deepin 15.11 with the kernel upgraded to 5.5.11-050511-lowlatency. Installing gives this feedback:
`0 packages upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 32.6 kB of archives. After unpacking 109 kB will be used.
Get: 1 http://packages.deepin.com/deepin lion/main amd64 v4l2loopback-dkms all 0.10.0-1 [32.6 kB]
Fetched 32.6 kB in 0s (66.4 kB/s)      
Selecting previously unselected package v4l2loopback-dkms.
(Reading database ... 274492 files and directories currently installed.)
Preparing to unpack .../v4l2loopback-dkms_0.10.0-1_all.deb ...
Unpacking v4l2loopback-dkms (0.10.0-1) ...
Setting up v4l2loopback-dkms (0.10.0-1) ...
Loading new v4l2loopback-0.10.0 DKMS files...
Building for 5.5.11-050511-lowlatency
Building initial module for 5.5.11-050511-lowlatency
Error! Bad return status for module build on kernel: 5.5.11-050511-lowlatency (x86_64)
Consult /var/lib/dkms/v4l2loopback/0.10.0/build/make.log for more information.
`

The log file tells this:
`DKMS make.log for v4l2loopback-0.10.0 for kernel 5.5.11-050511-lowlatency (x86_64)
Mon Mar 30 09:44:13 CEST 2020
Building v4l2-loopback driver...
make -C /lib/modules/5.5.11-050511-lowlatency/build M=/var/lib/dkms/v4l2loopback/0.10.0/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.5.11-050511-lowlatency'
  CC [M]  /var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.c: In function â€˜vidioc_qbufâ€™:
/var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.c:1518:4: error: implicit declaration of function â€˜do_gettimeofdayâ€™ [-Werror=implicit-function-declaration]
    do_gettimeofday(&b->buffer.timestamp);
    ^~~~~~~~~~~~~~~
/var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.c: In function â€˜v4l2_loopback_initâ€™:
/var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.c:2206:2: error: implicit declaration of function â€˜setup_timerâ€™ [-Werror=implicit-function-declaration]
  setup_timer(&dev->sustain_timer, sustain_timer_clb, nr);
  ^~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:265: recipe for target '/var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.o' failed
make[2]: *** [/var/lib/dkms/v4l2loopback/0.10.0/build/v4l2loopback.o] Error 1
Makefile:1694: recipe for target '/var/lib/dkms/v4l2loopback/0.10.0/build' failed
make[1]: *** [/var/lib/dkms/v4l2loopback/0.10.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.5.11-050511-lowlatency'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2
`
[This bug report](https://github.com/umlaeute/v4l2loopback/issues/248) seems to describe the same issue, but since it has already been merged into the project this fix should already have been applied?

umlaeute, 2020-03-30:
this is a packaging problem, not a problem with `v4l2loopback` per se.

your distro obviously features the latest and greatest kernel, but at the same time ships with a 3 year old kernel module.

Debian has packages for `v0.12.3` (which includes the fix mentioned) starting with [bullseye](https://packages.debian.org/bullseye/v4l2loopback-dkms).

closed 2020-03-30 by umlaeute

lain0, 2020-03-30:
under ubuntu 19.10 vs kernel 5.5.11-generic it installs but i got error modprobe: ERROR: could not insert 'v4l2loopback': Bad address

umlaeute, 2020-03-30:
@lain0 unfortunately your problem is totally unrelated to *this* issue (apart from having problems with `v4l2loopback` and the 5.5.11-kernel).

there's already a bug-report about "Bad address" problems #255 , which is more likely to match.

lain0, 2020-03-30:
@umlaeute oh you'r right, sorry