v4l2loopback/v4l2loopback #270: "info: v4l2-controls: found control User Controls but it is readonly" flooding and bogging down in OBS

drmichaeljgruber, 2020-03-30:
* v4l2loopback version: v0.12.3-6-ge4f2c3c
  * kernel version: 5.5.11-200.fc31.x86_64 #1 SMP Mon Mar 23 17:32:43 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
  * Distribution (+version): Fedora 31
  * obs-studio: 25.0.1-1

#### Steps to reproduce:

  1. `sudo modprobe v4l2loopback exclusive_caps=1`
  2. `gphoto2 --stdout --capture-movie | gst-launch-1.0 fdsrc ! decodebin3 name=dec ! queue ! videoconvert ! v4l2sink device=/dev/video3`
  3. In obs, open the v4l device properties.
  
#### Observed Results:

  * Camera live recording shows up in the preview frame and the movie frame (pipeline works).
  * Terminal is flooded with `info: v4l2-controls: found control User Controls but it is readonly`
  * obs cpu usage goes above 100%, obs freezes.
  
#### Expected Results:

  * Expected working device properties dialog.
  
#### Relevant Code:

 See above. If I kill the pipeline the terminal is still flooded. Only killing obs helps. If I restart obs (properties dialog closed) with the pipeline running the camera works (amazingly in fact, it's a DSLR). Opening the info dialog now gives the same problem as above.

Interestingly: If I kill the pipeline and obs, rmmod and reload the module (without restarting the pipeline), then start obs, then open the info dialog, then the same problem occurs. I have no clue how this could be, and whether maybe gstreamer or obs are to blame here. `journalctl` shows nothing particular during these info floods.

Is there a simple "info query" which could maybe reproduce the obs issue without using obs?

umlaeute, 2020-03-31:
this seems to be an OBS issue.

the `v4l2loopback` device does not allow to modify any of the properties
- the loopback device just passes the frames as they are received from the producer; it doesn't do anything to these frames
- there's no way to signal a change-request from the consumer to the producer

so i guess the only way forward is to teach OBS that there might be devices that only have `readonly` controls.

closed 2020-03-31 by umlaeute

drmichaeljgruber, 2020-03-31:
> this seems to be an OBS issue.
>
> the `v4l2loopback` device does not allow to modify any of the properties
>
> * the loopback device just passes the frames as they are received from the producer; it [...]

While this all makes perfect sense I'm not sure what "User" Controls are if they are read only. Maybe they should be advertised?

Some more info: The same problem arises when I use `v4l2loopback` together with the [setup for IP wecam](https://github.com/bluezio/ipwebcam-gst). I am not 100% sure whether this appeared before the obs update from v24 to v25 - I discarded this solution early on because of the 4s delay. I came back to `v4l2loopback` because I wanted to use my DSLR - no noticable delay, but the described problem.

On the other hand, all is well with the [droidcam linux client](https://www.dev47apps.com/droidcam/linuxx/) (also on [github](https://github.com/aramg/droidcam/tree/master/linux)) which appears to use a fork of your module. I cannot use that module to feed the gphoto pipe into, though:
```
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Device '/dev/video0' is not a output device.
Additional debug info:
v4l2_calls.c(636): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Capabilities: 0x85200001
Setting pipeline to NULL ...
Freeing pipeline ...
```
But maybe a diff of the module sources gives a clue how to avoid the obs issues (even though they might be obs' fault)? Maybe it's as simple as loading your module with different caps or such?

drmichaeljgruber, 2020-03-31:
Okay, seems this is known in and hopefully fixed in this [obs commit](https://github.com/obsproject/obs-studio/commit/7b71a3b33d278a95f254fcf0a2fb4db48664b34f). I'll let you know otherwise ;-)