v4l2loopback/v4l2loopback #273: Discord app fails to display video feed

LouisTakePILLz, 2020-04-01:
I'm trying to use [Mon2Cam](https://github.com/ShayBox/Mon2Cam) to get around the [Chromium X screen enumeration bug](https://bugs.chromium.org/p/chromium/issues/detail?id=396091).

At first I couldn't even get the device to show up on either Discord or Chromium (Firefox picked it up though). When loading the kmod with `exclusive_caps=1`, both Discord and Chromium would list the device.

Tested in Chrome DevTools (ctrl+shift+i; works in the Discord app):
```js
navigator.mediaDevices
  .enumerateDevices()
  .then((a) =>
    a.filter((x) => x.kind === 'videoinput').forEach((x) => console.log(x))
  )
```

On Chromium, the camera displays just fine (tested with https://webcamtests.com/check).
Discord, however...
[image]
The camera stays black and prints this to stdout (when clicking the *Test Video* button):
```
[001:461] [24617] (webrtcvideocapturer.cc:179): Failed to find usable formats for id: platform:v4l2loopback-000
```

## Environment

v4l2loopback version: 0.12.3
kernel version: `Linux peelz-pc 5.4.28 #1-NixOS SMP Wed Mar 25 07:26:00 UTC 2020 x86_64 GNU/Linux`
distribution:
```
Distributor ID:	NixOS
Description:	NixOS 19.09.2318.ae48415e8b5 (Loris)
Release:	19.09.2318.ae48415e8b5
Codename:	loris
```
NixOS config:
<details>
  <summary>configuration.nix</summary>

```nix
{ lib, config, pkgs, ... }:

{
  # ...

  # Set kernel version
  boot.kernelPackages = let
    kernel = pkgs.linux_latest;

    linuxPackages_base = pkgs.linuxPackagesFor kernel;

    linuxPackages = linuxPackages_base.extend (lib.const (super: {
      # NixOS 19.09: v4l2loopback 0.12.0 doesn't compile for Linux 5.x
      v4l2loopback = super.v4l2loopback.overrideAttrs (oldAttrs: rec {
        version = "0.12.3";
        name = "v4l2loopback-${version}-${super.kernel.version}";
        src = pkgs.fetchFromGitHub {
          owner = "umlaeute";
          repo = "v4l2loopback";
          rev = "v${version}";
          sha256 = "01wahmrh4iw27cfmypik6frapq14vn7m9shmj5g7cr1apz2523aq";
        };
      });
    }));
  in linuxPackages;

  # Kernel modules
  boot.extraModulePackages = with config.boot.kernelPackages; [
    v4l2loopback
  ];

  # ...
}
```
</details>

## Tested versions

Chromium 80.0.3987.149 (works with `exclusive_caps=1`)
Firefox 74.0 (works)
Discord app 0.0.10 (doesn't work; camera stays black)

### Discord app dependencies (accessible via `process.versions` in the DevTools)
chrome: 78.0.3904.130
electron: 7.1.11
node: 12.8.1
v8: 7.8.279.23-electron.0

umlaeute, 2020-04-01:
have you tried filing a bug with discord?

closed 2020-04-01 by umlaeute

LouisTakePILLz, 2020-04-01:
Discord doesn't accept bug reports for the Linux client anymore :/

Is this an issue with Chromium/Electron or v4l2loopback?

LouisTakePILLz, 2020-04-01:
I did some more DevTools testing in Discord and I got the video feed working. It seems to be a Discord bug.

umlaeute, 2020-04-01:
thanks for confirming.

FranciscoDumont, 2020-07-05:
I am having the same issue, how did you solve it?

LouisTakePILLz, 2020-07-06:
I discovered that Discord doesn't like "non-standard" aspect ratios. I use obs-v4l2sink to stream to the v4l2loopback device, so I changed the OBS canvas resolution to 1920x1080 and that fixed the issue.