v4l2loopback/v4l2loopback #275: Fails to build with 4.19.113-1-MANJARO

pres-luca, 2020-04-06:
* `v4l2loopback` version: 0.12.3

  * kernel version: 4.19.112-1-MANJARO

  * Distribution (+version): Manjaro Release 19.0.2
  
When I tried to build using `make` I first encountered the following error:
`make[1]: *** /lib/modules/4.19.112-1-MANJARO/build: No such file or directory.`
Which I could quickly fix by substituting `uname -r` with a hardcoded `4.19.113-1-MANJARO` in the Makefile.
But then I got the next problem:
`make[1]: *** No rule to make target 'modules'.  Stop.`
I am not used to working with make so I am not really sure what this means. Just in case that this provides some more info that you need to understand what happens, here is the full output of running `make`:
```Building v4l2-loopback driver...
make -C /lib/modules/4.19.113-1-MANJARO/build M=/home/lucav/tmp/loop/v4l2loopback modules
make[1]: Entering directory '/usr/lib/modules/4.19.113-1-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.19.113-1-MANJARO/build'
make: *** [Makefile:43: v4l2loopback.ko] Error 2
```
Kudos for creating such a nice bug reporting template btw, this really makes it easier to explain what happened for newbies like me. I'll be happy to answer your questions if anything still remained unclear though. Thanks for your time.

pres-luca, 2020-04-07:
I just upgraded to `5.4.28-1-MANJARO` and neither of the problems I encountered are reproducable. Everything works flawlessly, thanks for creating such a nice program

closed 2020-04-07 by pres-luca

Goblin80, 2020-12-01:
Probably you were missing kernel headers:

`sudo pacman -S linux54-headers`