v4l2loopback/v4l2loopback #277: Fails to build with Linux Kernel 5.6

Srinivasa314, 2020-04-07:
Building with kernel 5.6 fails with these errors
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:137:47: warning: ‘struct timeval’ declared inside parameter list will not be visible outside of this definition or declaration
  137 | static inline void v4l2l_get_timestamp(struct timeval *tv) {
      |                                               ^~~~~~~
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c: In function ‘v4l2l_get_timestamp’:
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:147:4: error: dereferencing pointer to incomplete type ‘struct timeval’
  147 |  tv->tv_sec = (time_t)ts.tv_sec;
      |    ^~
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:147:16: error: ‘time_t’ undeclared (first use in this function); did you mean ‘ktime_t’?
  147 |  tv->tv_sec = (time_t)ts.tv_sec;
      |                ^~~~~~
      |                ktime_t
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:147:16: note: each undeclared identifier is reported only once for each function it appears in
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:147:23: error: expected ‘;’ before ‘ts’
  147 |  tv->tv_sec = (time_t)ts.tv_sec;
      |                       ^~
      |                       ;
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c: In function ‘vidioc_qbuf’:
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:1523:24: error: passing argument 1 of ‘v4l2l_get_timestamp’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1523 |    v4l2l_get_timestamp(&b->buffer.timestamp);
      |                        ^~~~~~~~~~~~~~~~~~~~
      |                        |
      |                        struct __kernel_v4l2_timeval *
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:137:56: note: expected ‘struct timeval *’ but argument is of type ‘struct __kernel_v4l2_timeval *’
  137 | static inline void v4l2l_get_timestamp(struct timeval *tv) {
      |                                        ~~~~~~~~~~~~~~~~^~
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c: In function ‘v4l2_loopback_write’:
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:1950:22: error: passing argument 1 of ‘v4l2l_get_timestamp’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1950 |  v4l2l_get_timestamp(&b->timestamp);
      |                      ^~~~~~~~~~~~~
      |                      |
      |                      struct __kernel_v4l2_timeval *
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:137:56: note: expected ‘struct timeval *’ but argument is of type ‘struct __kernel_v4l2_timeval *’
  137 | static inline void v4l2l_get_timestamp(struct timeval *tv) {
      |                                        ~~~~~~~~~~~~~~~~^~
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c: In function ‘init_buffers’:
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:2055:23: error: passing argument 1 of ‘v4l2l_get_timestamp’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 2055 |   v4l2l_get_timestamp(&b->timestamp);
      |                       ^~~~~~~~~~~~~
      |                       |
      |                       struct __kernel_v4l2_timeval *
v4l2loopback-dkms/src/v4l2loopback-0.12.3/v4l2loopback.c:137:56: note: expected ‘struct timeval *’ but argument is of type ‘struct __kernel_v4l2_timeval *’
  137 | static inline void v4l2l_get_timestamp(struct timeval *tv) {
      |                                        ~~~~~~~~~~~~~~~~^~

umlaeute, 2020-04-07:
fixed in git.

closed 2020-04-07 by umlaeute

maximbaz, 2020-04-08:
Would you please consider cutting a new release?