v4l2loopback/v4l2loopback #278: Cannot create new /dev/video* device

AFanthomme, 2020-04-09:
#### System information:

 * `v4l2loopback` version: v4l2loopback driver version 0.12.3

  * kernel version: 4.15.0-96-generic #97-Ubuntu SMP Wed Apr 1 03:25:46 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

  * Distribution (+version): Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

#### Steps to reproduce and observed Results:

* List existing video devices: `ls -ltrh /dev/video*`
> crw-rw----+ 1 root video 81, 0 avril  9 10:57 /dev/video0

I think this device was created using the command below, as my desktop has not had any video device connected to it, ever.

* try creating a new device:
`sudo modprobe v4l2loopback devices=1 card_label="OBS cam" exclusive_caps=1`

I have tried without any option, adding the video_number=3 option, with or without sudo, no change here.

* List existing video devices again, and find same output as before: `ls -ltrh /dev/video*`
> crw-rw----+ 1 root video 81, 0 avril  9 10:57 /dev/video0

#### Expected Results:

I would have expected to get a new video device, however this is not the case.
The objective of this is to setup a virtual webcam using obs and obs-virtual-sink (https://github.com/CatxFish/obs-v4l2sink) for use in jitsi, if that is of any relevance.

Additionally, tests using vlc `vlc v4l2:///dev/video0` and webrtc (https://test.webrtc.org/) both fail (webrtc finds no input device, while vlc returns "cannot open device '/dev/video0': Operation not permitted" even with my user part of the video group) so I guess the existing created video device does not work.
  
Thanks in advance for any help on clearing this issue.

derneue, 2020-04-09:
Can you try to unload your module
```
sudo rmmod v4l2loopback
```
To check if the device /dev/video0 disappears? If not, something in your computer is a capture/sink device.

AFanthomme, 2020-04-09:
I can confirm, using this command to unload v4l2loopback does make the existing device disappear:
> ls: cannot access '/dev/video*': No such file or directory 

Additionally, now that I unload, I can create a new device with any number I want (eg 3) or 3 devices at once, which is already an improvement. I cannot create a new one after the first one (or ones) already exist, but that can be circumvened easily.

Still, the problem that it cannot be opened from vlc nor read from webrtc remains

AFanthomme, 2020-04-09:
After reboot and upgrade of all packages, everything seems to work perfectly fine, so I have no idea what caused the problem in the first place.

Best regards

closed 2020-04-09 by AFanthomme