v4l2loopback/v4l2loopback #283: Switching producers crashes ffmpeg reading from the device

tinloaf, 2020-04-14:
### Step 2: Describe your environment

  * `v4l2loopback` version: 0.12.4 (compiled from git commit 9bd555319eb74abcbd88f2311fc7ce87b508090c )
  * kernel version: 4.15.0-96-generic (current Ubuntu Bionic's default)
  * Distribution (+version): Ubuntu 18.04.4 LTS

  
### Step 3: Describe the problem:

I have created a loopback device (`/dev/video2`). I start an `ffmpeg` stream streaming a static picture into this device (see exact commands below), and then start a second `ffmpeg` recording from this device to a file. Now I close the `ffmpeg` that streams the static picture and instead start a third `ffmpeg` that streams from my webcam (`/dev/video1`) into the loopback device.

The recording `ffmpeg` instance now crashes with:

```
[video4linux2,v4l2 @ 0x55a97b5e58c0] Dequeued v4l2 buffer contains 385024 bytes, but 384000 were expected. Flags: 0x00000001.
/dev/video2: Invalid data found when processing input
```

#### Steps to reproduce:

  1. Create a loopback device as `/dev/video2`, have a webcam as `/dev/video1`, have some PNG at `standby.png`.
  2. Start the static picture stream: `ffmpeg -loglevel warning -loop 1 -f image2 -i standby.png -r 15 -vcodec rawvideo -pix_fmt yuv420p -f v4l2 /dev/video2`
  3. Start recording from the device: `ffmpeg -loglevel warning -i /dev/video2 /tmp/out.mpeg`
  4. Kill the static picture stream with `SIGINT` (i.e., press CTRL-C)
  5. Start streaming the webcam to the loopback device: `ffmpeg -i /dev/video1 -f v4l2 /dev/video2`

#### Observed Results:

The crash mentioned previously. The complete output of the crashed `ffmpeg` instance is:
```
> ffmpeg -loglevel warning -i /dev/video2 /tmp/out.mpeg
File '/tmp/out.mpeg' already exists. Overwrite ? [y/N] y
[mpeg @ 0x55a97b5e8d60] VBV buffer size not set, using default size of 130KB
If you want the mpeg file to be compliant to some specification
Like DVD, VCD or others, make sure you set the correct buffer size
Past duration 0.698097 too large

<<< Switch of producers happens here >>>

[video4linux2,v4l2 @ 0x55a97b5e58c0] Dequeued v4l2 buffer contains 385024 bytes, but 384000 were expected. Flags: 0x00000001.
/dev/video2: Invalid data found when processing input
```
  
#### Expected Results:

No crash. :)
  
#### Related issues

#190 sees the same error, however it is caused by gstreamer input. Also, the fix by @ochilan in https://github.com/umlaeute/v4l2loopback/issues/190#issuecomment-428313097 does not solve this problem, so I assume this to be a separate bug.

tinloaf, 2020-04-14:
Experimenting around I found out that this does not crash if I set both producers to the same resolution and pixel format. If I change the static picture streaming command to:
```ffmpeg -loglevel warning -loop 1 -f image2 -i standby.png -r 15 -vcodec rawvideo -pix_fmt yuv420p -s 1024x786 -f v4l2 /dev/video2
```

and the webcam-stream to
```
ffmpeg -i /dev/video1 -pix_fmt yuv420p -s 1024x768 -f v4l2 /dev/video2
```

it does not crash. I assume that setting resolution and pixel format equally means that each picture in the stream has the same size. Does that somehow help with the buffer sizes?

umlaeute, 2020-04-14:
hmm, if `ffmpeg` crashes, then it's most likely a bug in `ffmpeg`. why do you report it here?

apart from that: if a consumer is attached to a device, the format should be locked (so a new producer will not be able to change the format). so probably there is *also* an issue with `v4l2loopback`

tinloaf, 2020-04-15:
> hmm, if `ffmpeg` crashes, then it's most likely a bug in `ffmpeg`. why do you report it here?

Because in the mentioned issue #190 you said that this error message indicates a bug in `v4l2loopback`. I really do not know that much about v4l, but I was assuming that a v4l device consumer may make certain assumptions about the data read from the device, and that maybe `v4l2loopback` violates these rules?

> apart from that: if a consumer is attached to a device, the format should be locked (so a new producer will not be able to change the format). so probably there is _also_ an issue with `v4l2loopback`

So the consumer may indeed assume that resolution and pixel format do not change? What is supposed to happen then if the producer changes to one with a different format? The producer should see an error?

umlaeute, 2020-04-15:
>>  hmm, if ffmpeg crashes, then it's most likely a bug in ffmpeg. why do you report it here?
> Because in the mentioned issue #190 you said [...]

doh. :wink: 

anyhow, an application should never crash in such circumstances. so there clearly *is* an `ffmpeg` error.

> The producer should see an error?

yes.

umlaeute, 2023-04-27:
> Experimenting around I found out that this does not crash if I set both producers to the same resolution and pixel format.

well yes. you cannot change the resolution while a consumer is attached.

as of 6e3d0a77736d7a035fa11639be49933f2a849a32 (in current `main`), this is explicitly forbidden to the producer.

closed 2023-04-27 by umlaeute