v4l2loopback/v4l2loopback #286: Build failed with lowlatency kernel 5.3.0-46 in ubuntu

tiborkiss, 2020-04-15:
I have been used v4l2loopback for a while in generic version of the kernel in Ubuntu 18.04.
When I switched to low latency kernel (the same version!), I encountered the following build problem:

`tibor@tikotk:~/w/github/v4l2loopback$ make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/tibor/w/github/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-46-lowlatency'
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "_cond_resched" [/home/tibor/w/github/v4l2loopback/v4l2loopback.ko] undefined!
scripts/Makefile.modpost:92: recipe for target '__modpost' failed
make[2]: *** [__modpost] Error 1
Makefile:1658: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-46-lowlatency'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2
`

I am using the latest master git repository for building.

bwanaaa, 2020-04-18:
I am using 5.3.0-42-generic
and get this

```
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/stefan/Downloads/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-42-generic'
  CC [M]  /home/stefan/Downloads/v4l2loopback/v4l2loopback.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/stefan/Downloads/v4l2loopback/v4l2loopback.mod.o
  LD [M]  /home/stefan/Downloads/v4l2loopback/v4l2loopback.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-42-generic'

```

with the new v4l2loopback.mod file in the directory

umlaeute, 2020-04-22:
do you have matching kernel-headers for your lowlatency kernel?

montvid, 2020-05-15:
try http://ftp.de.debian.org/debian/pool/main/v/v4l2loopback/v4l2loopback-dkms_0.12.5-1_all.deb works with 5.4 amd64 rt kernel.

umlaeute, 2020-07-01:
you probably need to run `make clean` before...

wandetri, 2020-09-04:
> try http://ftp.de.debian.org/debian/pool/main/v/v4l2loopback/v4l2loopback-dkms_0.12.5-1_all.deb works with 5.4 amd64 rt kernel on debian.

thank you!, it works with 5.8.5 on Ubuntu 20.04

clementdevos, 2020-11-03:
> try http://ftp.de.debian.org/debian/pool/main/v/v4l2loopback/v4l2loopback-dkms_0.12.5-1_all.deb works with 5.4 amd64 rt kernel on debian.

Worked like a charm for me too  for iriunwebcam.

umlaeute, 2020-11-03:
clsoing due to inactivity from the OP

closed 2020-11-03 by umlaeute